espressif/esp-idf #8050: mdns is not working on wifi-ap (hotspot) mode

ESP3235, 2021-12-10:
esp-idf v4.3.1
module : ESP32-S2
IDE : eclipse

Hi to all.
I want to run mDns in hotspot mode , But not work.

david-cermak, 2021-12-10:
Hi @ESP3235 

Could you please share more details? What exactly does not work? Querying, discovery, services, hostnames, which tools are you using for discoveries, publishing services, sending queries?

This scenario (mdns on AP) is a very basic use case which must work and is being continuously tested. The only trouble I could think of is that the mdns library probes and advertizes after the AP started, so whenever a station joins this AP, the advertizing packets of the published services might be already sent out. The services should be still discoverable.

ESP3235, 2021-12-10:
mDns is working when ESP32 connected to wifi network , but when working in hotspot mode , http server just work with IP 192.168.4.1

closed 2021-12-11 by ESP3235