espressif/esp-idf #8052: W5500 IP Address: 0.0.0.0

kbssa, 2021-12-10:
I'am trying to make a W5500 ethernet module work wihtout success.

The code compiles but the IP on the serial monitor is always (IP Address: 0.0.0.0).

This the full code:

```
#include <Arduino.h>
#include <WiFi.h>
#include <ETH.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include "sdkconfig.h"
#include "driver/spi_master.h"

AsyncWebServer server(80);

const char* PARAM_MESSAGE = "message";

void notFound(AsyncWebServerRequest* request) {
	request->send(404, "text/plain", "Not found");
}

bool setupW5500() {
	WiFi.begin();

	tcpip_adapter_set_default_eth_handlers();

	// Initialize TCP/IP network interface (should be called only once in application)
	ESP_ERROR_CHECK(esp_netif_init());
	esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
	esp_netif_t* eth_netif = esp_netif_new(&cfg);
	// Set default handlers to process TCP/IP stuffs
	ESP_ERROR_CHECK(esp_eth_set_default_handlers(eth_netif));

	esp_eth_mac_t* eth_mac = NULL;
	esp_eth_phy_t* eth_phy = NULL;

	gpio_install_isr_service(0);

	spi_bus_config_t buscfg = {
	  .mosi_io_num = 23,
	  .miso_io_num = 19,
	  .sclk_io_num = 18,
	  .quadwp_io_num = -1,
	  .quadhd_io_num = -1,
	};
	ESP_ERROR_CHECK(spi_bus_initialize(SPI3_HOST, &buscfg, 1));

	spi_device_handle_t spi_handle = NULL;
	spi_device_interface_config_t devcfg = {
		.command_bits = 16, // Actually it's the address phase in W5500 SPI frame
		.address_bits = 8,  // Actually it's the control phase in W5500 SPI frame
		.mode = 0,
		.clock_speed_hz = 12 * 1000 * 1000,
		.spics_io_num = 5,
		.queue_size = 20
	};
	ESP_ERROR_CHECK(spi_bus_add_device(SPI3_HOST, &devcfg, &spi_handle));
	/* w5500 ethernet driver is based on spi driver */
	eth_w5500_config_t w5500_config = ETH_W5500_DEFAULT_CONFIG(spi_handle);
	w5500_config.int_gpio_num = 25;

	eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
	eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
	phy_config.reset_gpio_num = 26;

	eth_mac = esp_eth_mac_new_w5500(&w5500_config, &mac_config);
	if (eth_mac == NULL) {
		log_e("esp_eth_mac_new_esp32 failed");
		Serial.println("esp_eth_mac_new_esp32 failed");
		return false;
	}

	eth_phy = esp_eth_phy_new_w5500(&phy_config);
	if (eth_phy == NULL) {
		log_e("esp_eth_phy_new failed");
		Serial.println("esp_eth_phy_new failed");
		return false;
	}

	esp_eth_config_t eth_config = ETH_DEFAULT_CONFIG(eth_mac, eth_phy);
	esp_eth_handle_t eth_handle = NULL;
	ESP_ERROR_CHECK(esp_eth_driver_install(&eth_config, &eth_handle));

	uint8_t macArr[] = { 0x02, 0x00, 0x00, 0x12, 0x34, 0x56 };
	ESP_ERROR_CHECK(esp_eth_ioctl(eth_handle, ETH_CMD_S_MAC_ADDR, macArr));

	/* attach Ethernet driver to TCP/IP stack */
	ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
	/* start Ethernet driver state machine */
	ESP_ERROR_CHECK(esp_eth_start(eth_handle));

	return true;
}

void setupWebserver() {
	server.on("/", HTTP_GET, [](AsyncWebServerRequest* request) {
		request->send(200, "text/plain", "Hello, world");
		});

	// Send a GET request to <IP>/get?message=<message>
	server.on("/get", HTTP_GET, [](AsyncWebServerRequest* request) {
		String message;
		if (request->hasParam(PARAM_MESSAGE)) {
			message = request->getParam(PARAM_MESSAGE)->value();
		}
		else {
			message = "No message sent";
		}
		request->send(200, "text/plain", "Hello, GET: " + message);
		});

	// Send a POST request to <IP>/post with a form field message set to <message>
	server.on("/post", HTTP_POST, [](AsyncWebServerRequest* request) {
		String message;
		if (request->hasParam(PARAM_MESSAGE, true)) {
			message = request->getParam(PARAM_MESSAGE, true)->value();
		}
		else {
			message = "No message sent";
		}
		request->send(200, "text/plain", "Hello, POST: " + message);
		});

	server.onNotFound(notFound);
	server.begin();

	Serial.print("IP Address: ");
	Serial.println(ETH.localIP());
}

void setup() {
	Serial.begin(115200);
	setupW5500();    //comment this line to get RFID working
	setupWebserver(); //comment this line to get RFID working
}

void loop() {

}
```
If anyone with much more knowhow than me with the IDF could point what is wrong in the code I would appreciated it.

david-cermak, 2021-12-16:
@kbssa Could you please try to run the same using IDF directly? You can simply start with [this example](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/basic), choose `w5500` device and see if you can get a valid IP address? This would help to narrow down the problem.

Alvin1Zhang, 2022-03-08:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details/updates.

closed 2022-03-08 by Alvin1Zhang