espressif/esp-idf #8053: ESP32 can never connect to WiFi, with identical settings/firmware to other working ESP32

MacDue, 2021-12-10:
## Environment

- Development Kit:  Heltec WiFi Kit
- Kit version: v2
- Module or chip used:  ESP32
- IDF version:
    // v4.4-beta1-177-ge104dd7f27
    // Note: I just tried v4.4 to see if it was fixed -- identical results on v4.3 release
- Build System:         CMake
- Compiler version:
   xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:   Linux
- Using an IDE: VSCode IDF extension 
- Power Supply:        USB

## Problem Description

I have two [Heltec WiFi Kit 32](https://heltec.org/project/wifi-kit-32/) ESP32 development boards, one works great on WiFi and can reliably and consistently connect, the other is pretty much unusable. Until today it would be able to connect to WiFi after around 5-10 attempts, where on each attempt it would connect to the AP (`WIFI_EVENT_STA_CONNECTED`), then a few seconds later get a (`WIFI_EVENT_STA_DISCONNECTED`) with reason 15 (`WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT`) -- until finally was able to connect. Though after connecting it would sometimes never get assigned an IP address and have to be reset, though if you were lucky it would get one.

Since today, however, it can no longer connect at all and instead will never connect and just get (`WIFI_REASON_4WAY_HANDSHAKE_TIMEOUT`) every time.

Running the same code on my other dev board works just fine & it can connect every time. 

I've been trying to understand that could be the cause of this, and if it's some software bug, that somehow does not show up on my working board (with the identical.bin flashed), or if this ESP is just broken. However, it seems like similarly poor performance is seen on other colleagues dev kits, so it seems unlikely they're all broken.

I've attempted to compare the packets between my working ESP and broken ESP, though I'm not knowledgeable enough about the low-level details of WiFi to see what exactly is going on.

### Expected Behavior

Both dev boards will connect to WiFi reliably when running the identical firmware & configuration. 

### Actual Behavior

One of the dev boards works fine, the other can't connect / barely works.

Note: it seems like the distance to the AP has no effect on performance here.

### Source code

The main implementation of the WiFI class is here: 
https://gist.github.com/MacDue/eb232f9ba5e2e468da3c76e596e4d973
I don't believe it's doing anything that unusual

## Debug Logs

Broken ESP:
```
I (00:00:06.588) WiFi: Attempting to connect to WiFi AP: <My AP>
D (7148) wifi:Start wifi connect
D (7148) wifi:connect status 0 -> 0
D (7148) wifi:connect chan=0
D (7158) wifi:first chan=1
D (7158) wifi:connect status 0 -> 1
D (7158) wifi:filter: set rx policy=3
D (7158) wifi:clear scan ap list
D (7168) wifi:start scan: type=0x50f, priority=2, cb=0x40133760, arg=0x0, ss_state=0x1, time=5483838, index=0
0x40133760: cnx_start_handoff_cb at ??:?

D (7178) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
D (7298) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (7298) wifi:perform scan: ss_state=0x9, chan<2,0>, dur<0,120>
D (7418) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (7428) wifi:perform scan: ss_state=0x9, chan<3,0>, dur<0,120>
D (7548) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (7548) wifi:perform scan: ss_state=0x9, chan<4,0>, dur<0,120>
D (7668) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (7668) wifi:perform scan: ss_state=0x9, chan<5,0>, dur<0,120>
D (7788) wifi:scan end: arg=0x0, status=0, ss_state=0x3
D (7788) wifi:perform scan: ss_state=0x9, chan<6,0>, dur<0,120>
D (7908) wifi:scan end: arg=0x0, status=0, ss_state=0x3
[... 416 lines cut ...]

I (8669) wifi:connected with GCHQ Network 2.4 GHz, aid = 10, channel 8, BW20, bssid = 84:a4:23:10:ff:7c
I (8679) wifi:security: WPA2-PSK, phy: bgn, rssi: -62
D (8679) wifi:remove all except 84:a4:23:10:ff:7c from rc list
D (8689) wifi:clear blacklist
D (00:00:08.132) nvs: nvs_set sta.chan 1 8
D (00:00:08.145) nvs: nvs_set_blob sta.apinfo 700
D (8709) wifi:filter: set rx policy=7
I (8709) wifi:pm start, type: 1

D (8709) wifi:Send sta connected event
D (8709) wifi:connect status 1 -> 5
D (8719) wifi:obss scan is disabled
D (8719) wifi:start obss scan: obss scan is stopped
I (8729) wifi:AP's beacon interval = 102400 us, DTIM period = 2
D (8729) wifi:set max rate: from <rate=130, phy=3, sig=0> to <rate=144, phy=3 sig=0>
D (8739) wifi:sig_b=0, sig_g=0, sig_n=0, max_b=22, max_g=108, max_n=144
D (8739) wifi:update trc
D (00:00:08.187) event: running post WIFI_EVENT:4 with handler 0x400e8b00 and context 0x3ffb8de4 on loop 0x3ffafdd8
0x400e8b00: comms::network::Wifi::wifi_event_callback(void*, char const*, int, void*) at /home/macdue/git/ICCPD_Comms_Prototype/build/../components/comms_core/src/comms_core/network/Wifi.cpp:245

I (00:00:08.196) WiFi: Connected to WiFi AP: <My AP>
```
Packets captured as seen by AP:
[image]

MacDue, 2021-12-11:
It appears like on the bad ESP32, the `esp_wifi_internal_tx()` function just does not send the EAPOL packets (despite returning OK) -- can't really tell more than that.

jack0c, 2022-01-07:
Some suggestions:
1. erase the NVS, just to make the two boards have same setting
2. do full calibration if it will not break your development
3. If the issue still existence, I think it is the hardware problem of the bad board

closed 2022-03-05 by MacDue