espressif/esp-idf #8054: Testing Secure Boot v2 with virtual efuse

QuentinFarizon, 2021-12-10:
## Environment

- Development Kit: ESP32-S2-DevKitM-1
- Module or chip used:  ESP32-S2-MINI-1
- IDF version v4.4-beta1
- Build System: idf.py
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply:  USB

## Problem Description

I'm trying to test out Secure boot V2 without bricking my devkit.
For that, I'm using "Virtual eFuse" (with both CONFIG_EFUSE_VIRTUAL and CONFIG_EFUSE_VIRTUAL_KEEP_IN_FLASH)
I've tried on the unmodfied (apart from config) example peripherals/rmt/led_strip

### Expected Behavior

I have configured Secure Boot v2 from menuconfig, generated pem key file, and flashed the bootloader.
I do see `W (19) boot.esp32s2: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!`

I expect to see some efused burned (virtually) at first boot, and being able to revert them when I want to rollback.
I expect an error when trying to flash a bootloader again.

### Actual Behavior

I didn't find any documentation on how to revert efuses, si I don't know how to do it ?

I can't run the command `idf.py erase-flash`, which is expected I guess ? ("A fatal error occurred: ESP32-S2 ROM does not support function erase_flash").

When I run `espefuse.py summary`, no eFuse are changed. In particular :
SECURE_BOOT_EN (BLOCK0)                            Enables secure boot                                = False R/W (0b0)
ENABLE_SECURITY_DOWNLOAD (BLOCK0)                  Enables secure UART download mode (read/write flas = False R/W (0b0)

### Steps to reproduce

`ìdf.py bootloader-flash`

### Code to reproduce this issue

Here is the configuration :
```
#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_SUPPORTS_RSA=y
CONFIG_SECURE_TARGET_HAS_SECURE_ROM_DL_MODE=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V2_ENABLED=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="secure_boot_signing_key.pem"
# CONFIG_SECURE_BOOT_ENABLE_AGGRESSIVE_KEY_REVOKE is not set
# CONFIG_SECURE_BOOT_INSECURE is not set
# CONFIG_SECURE_FLASH_ENC_ENABLED is not set
# CONFIG_SECURE_DISABLE_ROM_DL_MODE is not set
CONFIG_SECURE_ENABLE_SECURE_ROM_DL_MODE=y
# CONFIG_SECURE_INSECURE_ALLOW_DL_MODE is not set
# end of Security features
```

KonstantinKondrashov, 2021-12-13:
Hi @QuentinFarizon!

> For that, I'm using "Virtual eFuse" (with both CONFIG_EFUSE_VIRTUAL and CONFIG_EFUSE_VIRTUAL_KEEP_IN_FLASH)

The eFuses are copied to `efuse_em` partition and all operations are redirected to this flash partition instead of real eFuses.
You can erase this partition to reset virt eFuses there.

> I expect to see some efused burned (virtually) at first boot, and being able to revert them when I want to rollback.
I expect an error when trying to flash a bootloader again.

What error did you expect when trying to flash a bootloader again? I do not see why it should be.

> I didn't find any documentation on how to revert efuses, si I don't know how to do it ?

Just erase the  `efuse_em` partition.

To erase flash you can try `esptool.py erase_flash`.

QuentinFarizon, 2021-12-13:
OK, indeed `esptool.py erase_flash` works, but not `idf.py erase_flash`.

QuentinFarizon, 2021-12-14:
My point remains : why don't I see Boot Secure enabled when disaplying eFuse values with `espefuse summary` ?

mahavirj, 2021-12-14:
@QuentinFarizon 

> I expect to see some efused burned (virtually) at first boot, and being able to revert them when I want to rollback.

EFuse memory is one-time programmable region. Each eFuse is a one-bit field which can be programmed to 1 after which it cannot be reverted back to 0. Virtual eFuse is something that can help for unit testing scenarios, where all eFuses operations are re-directed to virtual storage like RAM/FLASH. No real eFuses will be programmed on device with this option. Hence, this config can not help to test end-to-end security features as ROM code still relies on actual eFuse contents.

> I'm trying to test out Secure boot V2 without bricking my devkit.

For this, you may check our Qemu fork with ESP32 support, please refer to documentation at https://github.com/espressif/qemu/wiki. Please note that you will have to use ESP32 ECO3 revision, as that supports "Secure Boot v2" scheme, more on this at https://github.com/espressif/qemu/wiki#emulating-esp32-eco3

Hope this helps.

chegewara, 2021-12-14:
I have learned a lot from this issue, and im glad, because i always been scared to start playing with encrypted binaries.
Ive been able to merge 2 esp-idf examples to do some test app:
https://github.com/espressif/esp-idf/tree/master/examples/system/efuse
https://github.com/espressif/esp-idf/tree/master/examples/security/flash_encryption

There is one problem i should mention about, because i didnt find any info about it (or i missed it):
- after building and flashing app i have to use separate command to flash bootloader `idf.py bootloader-flash -p /dev/ttyUSB` or esp32 will boot like with empty flash

Here is output with success test:
1. first boot which will perform binary encryption and restart
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff00c0,len:15072
load:0x40078000,len:25176
load:0x40080400,len:3808
0x40080400: _init at ??:?

entry 0x40080698
W (91) boot.esp32: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
I (46) boot: ESP-IDF v5.0-dev-676-g5c33570524-dirty 2nd stage bootloader
I (46) boot: compile time 16:14:46
I (47) boot: chip revision: 3
I (51) boot.esp32: SPI Speed      : 40MHz
I (56) boot.esp32: SPI Mode       : DIO
I (60) boot.esp32: SPI Flash Size : 4MB
I (65) boot: Enabling RNG early entropy source...
I (70) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (81) boot:  0 nvs              WiFi data        01 02 00011000 00006000
I (89) boot:  1 nvs_key          NVS keys         01 04 00017000 00001000
I (96) boot:  2 storage          Unknown data     01 ff 00018000 00001000
I (103) boot:  3 phy_init         RF data          01 01 00019000 00001000
I (111) boot:  4 ota_0            OTA app          00 10 00020000 00100000
I (119) boot:  5 ota_1            OTA app          00 11 00120000 00100000
W (126) efuse: Loading virtual efuse blocks from real efuses
EFUSE_BLKx:
0) 0x00000000 0x1c838ae4 0x00a11052 0x0000a200 0x00001334 0x00100000 0x00000004 
1) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 
2) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 
3) 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 

I (185) boot:  6 efuse_em         efuse            01 05 00220000 00002000
I (193) boot: End of partition table
I (198) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=08e54h ( 36436) map
I (219) esp_image: segment 1: paddr=00028e7c vaddr=3ffb0000 size=02354h (  9044) load
I (223) esp_image: segment 2: paddr=0002b1d8 vaddr=40080000 size=04e40h ( 20032) load
I (233) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=191a4h (102820) map
I (271) esp_image: segment 4: paddr=000491cc vaddr=40084e40 size=068f4h ( 26868) load
I (282) esp_image: segment 5: paddr=0004fac8 vaddr=50000000 size=00010h (    16) load
I (282) esp_image: segment 6: paddr=0004fae0 vaddr=00000000 size=004f0h (  1264) 
I (288) esp_image: Verifying image signature...
I (293) secure_boot_v2: Secure boot V2 is not enabled yet and eFuse digest keys are not set
I (302) secure_boot_v2: Verifying with RSA-PSS...
I (311) secure_boot_v2: Signature verified successfully!
I (319) boot: Loaded app from partition at offset 0x20000
I (319) secure_boot_v2: enabling secure boot v2...
I (325) efuse: Batch mode of writing fields is enabled
I (330) esp_image: segment 0: paddr=00001020 vaddr=3fff00c0 size=03ae0h ( 15072) 
I (344) esp_image: segment 1: paddr=00004b08 vaddr=40078000 size=06258h ( 25176) 
I (356) esp_image: segment 2: paddr=0000ad68 vaddr=40080400 size=00ee0h (  3808) 
I (358) esp_image: Verifying image signature...
I (361) secure_boot_v2: Secure boot V2 is not enabled yet and eFuse digest keys are not set
I (370) secure_boot_v2: Verifying with RSA-PSS...
I (379) secure_boot_v2: Signature verified successfully!
I (381) secure_boot_v2: Secure boot digests absent, generating..
I (404) secure_boot_v2: Digests successfully calculated, 1 valid signatures (image offset 0x1000)
I (404) secure_boot_v2: 1 signature block(s) found appended to the bootloader.
[... 146 lines cut ...]
I (555) nvs: NVS key partition empty, generating keys
I (595) nvs: NVS partition "nvs" is encrypted.
This is esp32 chip with 2 CPU core(s), WiFi/BT/BLE, silicon revision 3, 4MB external flash
FLASH_CRYPT_CNT eFuse value is 1
Flash encryption feature is enabled in DEVELOPMENT mode
Erasing partition "storage" (0x1000 bytes)
Writing data with esp_partition_write:
I (625) example: 0x3ffb5ec0   00 01 02 03 04 05 06 07  08 09 0a 0b 0c 0d 0e 0f  |................|
I (625) example: 0x3ffb5ed0   10 11 12 13 14 15 16 17  18 19 1a 1b 1c 1d 1e 1f  |................|
Reading with esp_partition_read:
I (645) example: 0x3ffb5ea0   00 01 02 03 04 05 06 07  08 09 0a 0b 0c 0d 0e 0f  |................|
I (655) example: 0x3ffb5eb0   10 11 12 13 14 15 16 17  18 19 1a 1b 1c 1d 1e 1f  |................|
Reading with spi_flash_read:
I (665) example: 0x3ffb5ea0   00 01 02 03 04 05 06 07  08 09 0a 0b 0c 0d 0e 0f  |................|
I (675) example: 0x3ffb5eb0   10 11 12 13 14 15 16 17  18 19 1a 1b 1c 1d 1e 1f  |................|
I (685) example: read efuse fields
I (685) example: 1. read MAC address: 10:52:1c:83:8a:e4
I (695) example: 2. read secure_version: 0
I (695) example: 3. read ABS_DONE_1 field: 1
```

So, thank you @QuentinFarizon for raising issue and @mahavirj for support

I also should mention that i had to change few options in menuconfig to make it all works
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7712198/sdkconfig.txt)

mahavirj, 2021-12-17:
@chegewara 

Thank you for sharing additional information. It is quite helpful!

>  after building and flashing app i have to use separate command to flash bootloader idf.py bootloader-flash -p /dev/ttyUSB or esp32 will boot like with empty flash

If secure boot is enabled then bootloader has to be flashed separately. This is documented in note in the section at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html#how-to-enable-secure-boot-v2

mahavirj, 2021-12-22:
Closing. Please feel to re-open if there are any further comments or concerns here.

closed 2021-12-22 by mahavirj