espressif/esp-idf #8061: [USB host] failed to run on S3, works fine on S2

chegewara, 2021-12-11:
Here is simple usb host app i am using to test plug in/out
https://gist.github.com/chegewara/8f497fe8d8cf6b4597ea74f601307ff4

This app is working perfectly fine on S2, but it fails on S3
```
I (29) boot: ESP-IDF v5.0-dev-676-g5c33570524-dirty 2nd stage bootloader
I (29) boot: compile time 13:25:17
I (29) boot: chip revision: 0
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
W (52) bootloader_random: RNG for ESP32-S3 not currently supported
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=096c8h ( 38600) map
I (112) esp_image: segment 1: paddr=000196f0 vaddr=3fc90fe0 size=027b8h ( 10168) load
I (116) esp_image: segment 2: paddr=0001beb0 vaddr=40374000 size=04168h ( 16744) load
I (126) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1cf78h (118648) map
I (152) esp_image: segment 4: paddr=0003cfa0 vaddr=40378168 size=08e6ch ( 36460) load
I (160) esp_image: segment 5: paddr=00045e14 vaddr=50000000 size=00010h (    16) load
I (166) boot: Loaded app from partition at offset 0x10000
I (166) boot: Disabling RNG early entropy source...
W (169) bootloader_random: RNG for ESP32-S3 not currently supported
I (187) cpu_start: Pro cpu up.
I (188) cpu_start: Starting app cpu, entry point is 0x40375104
0x40375104: call_start_cpu1 at /home/chegewara/esp/master/components/esp_system/port/cpu_start.c:164

I (166) cpu_start: App cpu up.
I (202) cpu_start: Pro cpu start user code
I (202) cpu_start: cpu freq: 160000000 Hz
I (202) cpu_start: Application information:
I (205) cpu_start: Project name:     usb-host-test
I (210) cpu_start: App version:      1
I (215) cpu_start: Compile time:     Dec 11 2021 13:25:10
I (221) cpu_start: ELF file SHA256:  1fc60f079b2b978b...
I (227) cpu_start: ESP-IDF:          v5.0-dev-676-g5c33570524-dirty
I (234) heap_init: Initializing. RAM available for dynamic allocation:
I (241) heap_init: At 3FC941A0 len 0004BE60 (303 KiB): D/IRAM
I (247) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (254) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (260) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (267) spi_flash: detected chip: generic
I (271) spi_flash: flash io: dio
W (275) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (289) sleep: Configure to isolate all GPIO pins in sleep state
I (295) sleep: Enable automatic switching of GPIO sleep configuration
I (303) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (383) : install status: 0
I (383) : client register status: 0
usb_host_lib_handle_events event_flags: 0
E (693) HUB: Short device descriptor transfer failed, USB_TRANSFER_STATUS_ERROR

assert failed: enum_stage_cleanup_failed hub.c:453 (dequeued_enum_urb == p_hub_driver_obj->single_thread.enum_urb)

Backtrace:0x40375a62:0x3fcf41100x40378f85:0x3fcf4130 0x4037f4fd:0x3fcf4150 0x4200a264:0x3fcf4270 0x4200a39b:0x3fcf4290 0x4200a6e2:0x3fcf42b0 0x42006120:0x3fcf42d0 0x420059e3:0x3fcf42f0 0x4201cda9:0x3fcf4330 0x4037bc4d:0x3fcf4350 
0x40375a62: panic_abort at /home/chegewara/esp/master/components/esp_system/panic.c:403

0x40378f85: esp_system_abort at /home/chegewara/esp/master/components/esp_system/esp_system.c:123

0x4037f4fd: __assert_func at /home/chegewara/esp/master/components/newlib/assert.c:85

0x4200a264: enum_stage_cleanup_failed at /home/chegewara/esp/master/components/usb/hub.c:453 (discriminator 1)

0x4200a39b: enum_handle_events at /home/chegewara/esp/master/components/usb/hub.c:515

0x4200a6e2: hub_process at /home/chegewara/esp/master/components/usb/hub.c:652

0x42006120: usb_host_lib_handle_events at /home/chegewara/esp/master/components/usb/usb_host.c:490

0x420059e3: app_main at /home/chegewara/demos/tests/usb-host-test/build/../main/usb-host-test.c:47 (discriminator 15)

0x4201cda9: main_task at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/portable/port_common.c:127 (discriminator 2)

0x4037bc4d: vPortTaskWrapper at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:130
```

Im not sure what may be the problem, is it chip revision 0 or maybe pins are not configured properly:
```
I (289) sleep: Configure to isolate all GPIO pins in sleep state
I (295) sleep: Enable automatic switching of GPIO sleep configuration
```
esp-idf commit: 5c33570524118873f7bd32490c7a0442fede4bf8
board: esp32 S3 wroom 1, N8R2

Thanks

chegewara, 2021-12-11:
Looks like its pins misconfig. Adding prevent crashing, but didnt solve problem completely:
```
void app_main(void)
{
    gpio_reset_pin(GPIO_NUM_19);
```

Dazza0, 2021-12-13:
@chegewara It's probably something to do the difference in how the USB PHY is setup on the [S2](https://github.com/espressif/esp-idf/blob/e104dd7f27d2e73ab0e9b614dd7b9295099069bf/components/hal/esp32s2/include/hal/usb_phy_ll.h#L23) vs the [S3](https://github.com/espressif/esp-idf/blob/e104dd7f27d2e73ab0e9b614dd7b9295099069bf/components/hal/esp32s3/include/hal/usb_phy_ll.h#L25) (there's an internal MR pending review already). There are an extra couple of bits to set on the S3 in order to switch the USB PHY from being connected to the USB Serial JTAG Controller to the USB OTG Controller.

There's a USB PHY driver that should already abstract away these differences, but that USB Host stack hasn't been updated to use the USB PHY driver (will be part of the MR). In the mean time, you can try adding modifying `usbh_ll_internal_phy_conf()` to the following:

```c
#include "soc/rtc_cntl_struct.h"
static inline void usbh_ll_internal_phy_conf(usb_wrap_dev_t *hw)
{
    //Enable internal PHY
    hw->otg_conf.pad_enable = 1;
    hw->otg_conf.phy_sel = 0;
    // phy_sel is controlled by the following register value
    RTCCNTL.usb_conf.sw_hw_usb_phy_sel = 1;
    // phy_sel=sw_usb_phy_sel=1, USB_OTG is connected with internal PHY
    RTCCNTL.usb_conf.sw_usb_phy_sel = 1;
    //Set pulldowns on D+ and D-
    hw->otg_conf.pad_pull_override = 1;
    hw->otg_conf.dp_pulldown = 1;
    hw->otg_conf.dm_pulldown = 1;
}
```

chegewara, 2021-12-13:
Thanks @Dazza0 
I thought it may be related to JTAG, as it is something extra on S3
Also thanks for that snippet, will check it, and im glad there is MR already, no rush i am just testing it on S3, no really needed yet

chegewara, 2021-12-13:
Ok, that seems to solove problem, even if USB host seems to be just a bit less stable on S3, but dont know why yet.

Thanks

foxabilo, 2021-12-13:
Just to add my results, when I update usbh_ll.h to include the RTCCNTL configuration as suggested by @Dazza0 the code does indeed function, but as @chegewara mentioned it is somewhat unstable. I can have a completely successful OTG Device insertion and removal like this 
```
create async task
I (7522) USB_HOST_CLIENT_EVENT_NEW_DEV: client event: 0, address: 1
I (7522) : device speed: USB_SPEED_FULL, device address: 1, max ep_ctrl size: 64, config: 1
config: 0[1], interface: 0[1], intf class: 255
E (29332) USBH: Device 1 gone
I (29332) USB_HOST_CLIENT_EVENT_DEV_GONE: client event: 1
W (29332) : DEVICE gone event
I (29362) : install status: 0
I (29362) : client register status: 0
I (33692) USB_HOST_CLIENT_EVENT_NEW_DEV: client event: 0, address: 1
I (33692) : device speed: USB_SPEED_FULL, device address: 1, max ep_ctrl size: 64, config: 1
config: 0[1], interface: 0[1], intf class: 255
E (36722) USBH: Device 1 gone
I (36722) USB_HOST_CLIENT_EVENT_DEV_GONE: client event: 1
W (36722) : DEVICE gone event
I (36752) : install status: 0
I (36752) : client register status: 0
```
or I can have Root port reset and reboot fails like this 
```
create async task
I (2023) USB_HOST_CLIENT_EVENT_NEW_DEV: client event: 0, address: 1
I (2023) : device speed: USB_SPEED_FULL, device address: 1, max ep_ctrl size: 64, config: 1
config: 0[1], interface: 0[1], intf class: 255
E (4843) USBH: Device 1 gone
I (4843) USB_HOST_CLIENT_EVENT_DEV_GONE: client event: 1
W (4843) : DEVICE gone event
I (4873) : install status: 0
I (4873) : client register status: 0
E (6533) HUB: Root port reset failed
E (6783) HUB: Root port reset failed
I (7203) USB_HOST_CLIENT_EVENT_NEW_DEV: client event: 0, address: 1
I (7203) : device speed: USB_SPEED_FULL, device address: 1, max ep_ctrl size: 64, config: 1
config: 0[1], interface: 0[1], intf class: 255
E (9903) USBH: Device 1 gone
I (9903) USB_HOST_CLIENT_EVENT_DEV_GONE: client event: 1
W (9903) : DEVICE gone event
I (9933) : install status: 0
I (9933) : client register status: 0
I (12253) USB_HOST_CLIENT_EVENT_NEW_DEV: client event: 0, address: 1
I (12253) : device speed: USB_SPEED_FULL, device address: 1, max ep_ctrl size: 64, config: 1
config: 0[1], interface: 0[1], intf class: 255
E (15603) USBH: Device 1 gone
I (15603) USB_HOST_CLIENT_EVENT_DEV_GONE: client event: 1
W (15603) : DEVICE gone event
I (15633) : install status: 0
I (15633) : client register status: 0
E (17413) HUB: Root port reset failed

abort() was called at PC 0x4200b600 on core 0
0x4200b600: root_port_handle_events at C:/Users/spenc/esp/esp-idf/components/usb/hub.c:222

Backtrace:0x40375aba:0x3fce15a00x4037907d:0x3fce15c0 0x4037f43a:0x3fce15e0 0x4200b600:0x3fce1650 0x4200c0bd:0x3fce1670 0x420071c3:0x3fce1690 0x42005ef0:0x3fce16b0 0x4037bd0d:0x3fce16e0
0x40375aba: panic_abort at C:/Users/spenc/esp/esp-idf/components/esp_system/panic.c:402

0x4037907d: esp_system_abort at C:/Users/spenc/esp/esp-idf/components/esp_system/esp_system.c:121

0x4037f43a: abort at C:/Users/spenc/esp/esp-idf/components/newlib/abort.c:46

0x4200b600: root_port_handle_events at C:/Users/spenc/esp/esp-idf/components/usb/hub.c:222

0x4200c0bd: hub_process at C:/Users/spenc/esp/esp-idf/components/usb/hub.c:649

0x420071c3: usb_host_lib_handle_events at C:/Users/spenc/esp/esp-idf/components/usb/usb_host.c:490

0x42005ef0: client_async_seq_task(void*) at C:/Users/spenc/esp/esp32-usb-host/components/usb_host/host/usb_host.cpp:40

0x4037bd0d: vPortTaskWrapper at C:/Users/spenc/esp/esp-idf/components/freertos/port/xtensa/port.c:130

ELF file SHA256: e9a22024db2aa6f2

Rebooting...
ESP-ROM:esp32s3-20210327

chipweinberger, 2021-12-21:
I'm hitting this issue as well. #8122

closed 2022-01-06 by espressif-bot

vigyanabikshu1, 2022-06-13:
> Just to add my results, when I update usbh_ll.h to include the RTCCNTL configuration as suggested by @Dazza0 the code does indeed function, but as @chegewara mentioned it is somewhat unstable. I can [...]

I received the same error due to board mistake. Problem was D+ short with ground, resolved by clearing  short circuit.