espressif/esp-idf #8063: wifi_prov_mgr_start_provisioning crashes when service_key is set

Lisa999, 2021-12-11:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- IDF version: v4.3.1

## Problem Description

Example: https://github.com/espressif/esp-idf/tree/5c33570/examples/provisioning/wifi_prov_mgr 

Is not working when service_key is set

### Expected Behavior

Different payload output

### Actual Behavior

Crash

### Steps to reproduce

1. Use SoftAP scheme (.scheme = wifi_prov_scheme_softap)
2. Change service_key
3. Call wifi_prov_mgr_start_provisioning

### Code to reproduce this issue

Change
"const char *service_key = NULL;   "
To
"const char *service_key = "1234";  "

laukik-hase, 2021-12-13:
Hello, @Lisa999

Thanks for the issue report. I was able to reproduce this on the tag [`v4.3.1`](https://github.com/espressif/esp-idf/tree/v4.3.1).

In the `wifi_prov_mgr` example, `service_key` is the password for the ESP32 SoftAP, which should be at least 8 characters. In your case, the service key is `1234` (only 4 characters) which results in a crash.

Could you please attach the logs of the crash from your side or acknowledge if they are the same as mine. Following are the crash logs from my testing:

```bash
I (704) app: Starting provisioning
I (704) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (804) wifi:mode : sta (8c:aa:b5:b5:a9:f4)
I (804) wifi:enable tsf
I (814) wifi:mode : sta (8c:aa:b5:b5:a9:f4) + softAP (8c:aa:b5:b5:a9:f5)
I (814) wifi:Total power save buffer number: 16
I (814) wifi:Init max length of beacon: 752/752
I (824) wifi:Init max length of beacon: 752/752
W (824) wifi:password less than 8

E (824) wifi_prov_scheme_softap: Failed to set Wi-Fi config : 12299
E (834) wifi_prov_scheme_softap: Failed to start Wi-Fi AP
E (944) wifi_prov_mgr: Failed to start service
ESP_ERROR_CHECK failed: esp_err_t 0x300b (ESP_ERR_WIFI_PASSWORD) at 0x40087c70
0x40087c70: _esp_error_check_failed at /home/laukik/esp/esp-idf/components/esp_common/src/esp_err.c:41

file: "../main/app_main.c" line 294
func: app_main
expression: wifi_prov_mgr_start_provisioning(security, pop, service_name, service_key)

abort() was called at PC 0x40087c73 on core 0
0x40087c73: _esp_error_check_failed at /home/laukik/esp/esp-idf/components/esp_common/src/esp_err.c:42

Backtrace:0x40088bd7:0x3ffb2470 0x400893d5:0x3ffb2490 0x4008ffbe:0x3ffb24b0 0x40087c73:0x3ffb2520 0x400d7dd1:0x3ffb2540 0x40157819:0x3ffb2690 0x4008c8a9:0x3ffb26b0
0x40088bd7: panic_abort at /home/laukik/esp/esp-idf/components/esp_system/panic.c:368

0x400893d5: esp_system_abort at /home/laukik/esp/esp-idf/components/esp_system/system_api.c:112

0x4008ffbe: abort at /home/laukik/esp/esp-idf/components/newlib/abort.c:46

0x40087c73: _esp_error_check_failed at /home/laukik/esp/esp-idf/components/esp_common/src/esp_err.c:42

0x400d7dd1: app_main at /home/laukik/esp/esp-idf/examples/provisioning/wifi_prov_mgr/build/../main/app_main.c:294 (discriminator 1)

0x40157819: main_task at /home/laukik/esp/esp-idf/components/freertos/port/port_common.c:133 (discriminator 2)

0x4008c8a9: vPortTaskWrapper at /home/laukik/esp/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: 4c1fa400e0cc24bd

```

From the above logs, the error code `12299` corresponds to [`ESP_ERR_WIFI_PASSWORD`](https://github.com/espressif/esp-idf/blob/5c33570524118873f7bd32490c7a0442fede4bf8/components/esp_wifi/include/esp_wifi.h#L74) signifying an invalid password.

Lisa999, 2021-12-13:
Hi @laukik-hase,

I wasn't aware that the password is bound to limits, that's not explained anywhere. I'll try if 8 characters is working for me and put the crash log here.
Thanks for the help!

Lisa999, 2021-12-23:
I've tested it. The 8 characters is working just fine. So it's just not working when < 8 characters. There should be a check on that.

closed 2021-12-27 by espressif-bot