espressif/esp-idf #8065: GDB debugging commands do the wrong thing

szekelyisz, 2021-12-12:
## Environment

- Development Kit:      ESP32-Ethernet-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): 1.1
- Module or chip used:  ESP32-WROVER-B
- IDF version (run ``git describe --tags`` to find it): v4.4-beta1-177-ge104dd7f27
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: Eclipse
- Power Supply:         USB

## Problem Description

Stepping commands seem to be messed up. After a breakpoint is hit, "Step over" does "Step into" for example. This is not an Eclipse plugin problem as it also happens if I use the debugger console to give direct commands to gdb.

### Expected Behavior

Stepping functions to work as expected.

### Actual Behavior

"Step over" does "Step into" and other weird stuff like seemingly corrupting the stack.

### Steps to reproduce

1. Compile the `protocols/http_server/simple` example.
2. Start a JTAG debugging session.
3. Set a breakpoint at [this line](https://github.com/espressif/esp-idf/blob/e104dd7f27d2e73ab0e9b614dd7b9295099069bf/examples/protocols/http_server/simple/main/main.c#L156)
4. Perform an HTTP request that should cause the breakpoint to be hit.
5. Try stepping around. See below for a full session transcript.

### Code to reproduce this issue

[Here](https://github.com/espressif/esp-idf/tree/release/v4.4/examples/protocols/http_server/simple)

## Debug Logs

```
GNU gdb (crosstool-NG esp-2021r2) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Note: automatically using hardware breakpoints for read-only addresses.
[New Thread 1073460088]
[New Thread 1073458200]
[New Thread 1073448924]
[New Thread 1073453372]
[New Thread 1073413676]
[New Thread 1073411944]
[Switching to Thread 1073457852]

Thread 1 hit Temporary breakpoint 2, app_main () at ../main/main.c:394
394	{
[New Thread 1073510944]
[New Thread 1073466700]
[New Thread 1073479988]
[New Thread 1073470760]
[Switching to Thread 1073510944]
[... 170 lines cut ...]
#6  0x400f6181 in httpd_process_session (session=0x3ffc63f4, context=0x3ffc793c) at /home/szabi/esp/esp-idf/components/esp_http_server/src/httpd_main.c:178
        ctx = 0x3ffc793c
        fd = <optimized out>
        __func__ = <optimized out>
#7  0x40143572 in httpd_sess_enum (hd=<optimized out>, enum_function=0x400f6130 <httpd_process_session>, context=0x3ffc793c) at /home/szabi/esp/esp-idf/components/esp_http_server/src/httpd_sess.c:50
        current = 0x3ffc63f4
        end = 0x3ffc6844
#8  0x400f63be in httpd_server (hd=0x3ffc5f18) at /home/szabi/esp/esp-idf/components/esp_http_server/src/httpd_main.c:228
        read_set = {fds_bits = {0, 33554432}}
        tmp_max_fd = 57
        maxfd = <optimized out>
        __func__ = "httpd_server"
        active_cnt = <optimized out>
        context = {fdset = 0x3ffc7930, hd = 0x3ffc5f18}
#9  0x400f6422 in httpd_thread (arg=0x3ffc5f18) at /home/szabi/esp/esp-idf/components/esp_http_server/src/httpd_main.c:250
        ret = <optimized out>
        hd = 0x3ffc5f18
        __func__ = <optimized out>
#10 0x4008b4b4 in vPortTaskWrapper (pxCode=0x400f6414 <httpd_thread>, pvParameters=0x3ffc5f18) at /home/szabi/esp/esp-idf/components/freertos/port/xtensa/port.c:130
        pcTaskName = <optimized out>
```

## Other items if possible

- [simple.elf.gz](https://github.com/espressif/esp-idf/files/7698040/simple.elf.gz)

MPC-BlackBox, 2022-02-03:
I have had the same problem. I found that this is caused by **esp-2021r2-8.4.0** If you reinstall esp-idf version 4.3.1 and install the tools,
git clone -b v4.3.1 --recursive https://github.com/espressif/esp-idf.git esp-idf-v4.3.1
cd esp-idf-v4.3.1/tools
python idf_tools.py install

this will install the old GBD into your ~/.espressif directory
.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gdb

In eclipse set teh GDB client to this old GDB version.

igrr, 2022-02-08:
@szekelyisz @MPC-BlackBox Thank you for the issue report. We were able to confirm the issue. It is a regression introduced while fixing another bug (interrupted backtraces in "noreturn" functions, such as "abort".)
We will make a new release to fix the issue.

gerekon, 2022-02-08:
Sorry, closed it accidentally instead of its duplicate in OpenOCD repo.

closed 2022-02-23 by espressif-bot

szekelyisz, 2022-10-20:
I'm still experiencing this problem with toolchain version 2022r1.

szekelyisz, 2022-10-23:
@igrr Can you reopen this issue please?

Lapshin, 2022-10-23:
@szekelyisz , could you show the version? Please run `xtensa-esp32-elf-gdb --version`

szekelyisz, 2022-10-23:
> @szekelyisz , could you show the version? Please run `xtensa-esp32-elf-gdb --version`

```
$ xtensa-esp32-elf-gdb --version
GNU gdb (esp-gdb) 11.2_20220823
```

Lapshin, 2022-10-23:
@szekelyisz , this version must contain the fix...

Could you please describe steps to reproduce?

szekelyisz, 2022-10-23:
> @szekelyisz , this version must contain the fix...
> 
> Could you please describe steps to reproduce?

Thanks for dealing with this @Lapshin . Please see the original report for detailed instructions to reproduce.

Lapshin, 2022-10-26:
@szekelyisz , sorry for the long delay!

The issue was fixed on GDB side but did not fix on OpenOCD side.

Today it should be released. [The tag](https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20221026) for it is already created but artifacts have not been uploaded yet.

After the release, IDF will be updated with the newest OpenOCD in a few days.

szekelyisz, 2022-10-26:
Thank you @Lapshin. I'll wait for the release and test it.

gerekon, 2022-10-28:
@szekelyisz Release is ready https://github.com/espressif/openocd-esp32/releases/tag/v0.11.0-esp32-20221026

Lapshin, 2022-11-12:
@szekelyisz does it work for your case now?

szekelyisz, 2022-11-22:
Sorry for the long delay @Lapshin. It seems to work fine. Thanks for your help.