espressif/esp-idf #8067: 【ESP32-C3/AUX_CHAIN_IND 】 the max length of advertising data cannot be up to 1650 octets

xiaolongba, 2021-12-12:
Hi Espressif Experts,

regarding AUX_CHAIN_IND PDUs, I have mentioned it here #7626 before, as @Sushant-Espressif bro said, as below：

>All the extended advertising and extended scan features are supported including chained data (AUX_CHAIN_IND packets) and Max extended advertising data length of 1650 is supported. the details as this link: https://github.com/espressif/esp-idf/issues/7626#issuecomment-939421240

However, recently I tried to implement this new feature with the latest master branch, it showed that the error: 7 after `esp_ble_gap_ext_adv_start` API calling, which means `ESP_BT_STATUS_PARM_INVALID`. Obviously, the length of advertising data is greater than 254 bytes. it works well when I changed the advertising data length from 1650 to 200. so my question is as below:

1. is it possible to implement the extended chained adv with 1650 length?
2. if the current SDK can support this feature, do I need to do the extra settings? example project will be better if it is possibel.

Sushant-Espressif, 2021-12-14:
Hello @xiaolongba 
Following example can be used to verify extended adv chained data of length 1650: esp-idf/examples/bluetooth/bluedroid/ble_50/ble50_security_server

_But_, you need to take care of following things:
1. When connectable advertising is used, BLE spec does not allow chained data. It means that you can send data only upto 254 bytes (including extended adv header). Default ext adv type in ble50_security_server example set in `ext_adv_params_2M` is ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE. You can change it to `ESP_BLE_GAP_SET_EXT_ADV_PROP_NONCONN_NONSCANNABLE_UNDIRECTED` or `ESP_BLE_GAP_SET_EXT_ADV_PROP_SCANNABLE` and use 1650 of ext adv data.
2. For setting ext adv data of length 1650, you can update array `ext_adv_raw_data` to size 1650 and assign custom data to it.

Note the following APIs used in this example related to extended adv:
```
esp_ble_gap_ext_adv_set_params()
esp_ble_gap_config_ext_adv_data_raw()
esp_ble_gap_ext_adv_start()
```

xiaolongba, 2021-12-14:
@Sushant-Espressif Bro

big thanks. I confirmed it works smoothly after changing the ext_adv type from `ESP_BLE_GAP_SET_EXT_ADV_PROP_CONNECTABLE `to `ESP_BLE_GAP_SET_EXT_ADV_PROP_NONCONN_NONSCANNABLE_UNDIRECTED `or `ESP_BLE_GAP_SET_EXT_ADV_PROP_SCANNABLE `

> 1. When connectable advertising is used, BLE spec does not allow chained data. It means that you can send data only upto 254 bytes (including extended adv header).

as you said above, could your kindly show me which part on the BLE spec highlighted this? I tried to figure out the answer from BLE spec before reporting this issue, but only the following info:

[image]

xiaolongba, 2021-12-14:
@Sushant-Espressif Bro,

pls ignore the doubt above, I found the answer as below:
[image]

closed 2021-12-14 by xiaolongba