espressif/esp-idf #8069: esp32-c3: Can't set cpu clock to 40Mhz

Rokachy, 2021-12-12:
when I configure the CPU clock to 40Mhz it keep the old value at the sdkconfig:
-------------------------------------
SDK4.4beta1 
#
# ESP32C3-Specific
#
CONFIG_ESP32C3_DEFAULT_CPU_FREQ_80=y
# CONFIG_ESP32C3_DEFAULT_CPU_FREQ_160 is not set
CONFIG_ESP32C3_DEFAULT_CPU_FREQ_MHZ=80
# CONFIG_ESP32C3_REV_MIN_0 is not set
# CONFIG_ESP32C3_REV_MIN_1 is not set
# CONFIG_ESP32C3_REV_MIN_2 is not set
CONFIG_ESP32C3_REV_MIN_3=y
CONFIG_ESP32C3_REV_MIN=3

this is the bootup:
--------------------
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1724
load:0x403ce000,len:0x930
load:0x403d0000,len:0x2d68
SHA-256 comparison failed:
Calculated: 1bf0d71afbeb8a510f52b19a6f3a961cbeff88cdc8cbe9e2b7bb8c53ddaa8030
Expected: 0690e3f326c7cb4cf04145a67b0e3d57762fdc693f6ce94b433256b0ec964b01
Attempting to boot anyway...
entry 0x403ce000
I (48) boot: ESP-IDF v4.4-beta1-dirty 2nd stage bootloader
I (49) boot: compile time 17:45:59
I (49) boot: chip revision: 3
I (51) boot.esp32c3: SPI Speed      : 80MHz
I (56) boot.esp32c3: SPI Mode       : DIO
I (61) boot.esp32c3: SPI Flash Size : 4MB
I (65) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (74) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (89) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (97) boot:  2 factory          factory app      00 00 00010000 00100000
I (104) boot: End of partition table
I (108) esp_image: segment 0: paddr=00010020 vaddr=3c080020 size=147d8h ( 83928) map
I (130) esp_image: segment 1: paddr=00024800 vaddr=3fc90800 size=02c94h ( 11412) load
I (132) esp_image: segment 2: paddr=0002749c vaddr=40380000 size=08b7ch ( 35708) load
I (142) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=77480h (488576) map
I (220) esp_image: segment 4: paddr=000a74a8 vaddr=40388b7c size=07b54h ( 31572) load
I (227) esp_image: segment 5: paddr=000af004 vaddr=50000000 size=00004h (     4) load
I (227) esp_image: segment 6: paddr=000af010 vaddr=50000008 size=00018h (    24) load
I (240) boot: Loaded app from partition at offset 0x10000
I (241) boot: Disabling RNG early entropy source...
I (258) cpu_start: Pro cpu up.
I (266) cpu_start: Pro cpu start user code
I (266) cpu_start: cpu freq: 80000000
I (266) cpu_start: Application information:
I (269) cpu_start: Project name:     deep_sleep
I (274) cpu_start: App version:      1
I (278) cpu_start: Compile time:     Dec  9 2021 17:58:24
I (284) cpu_start: ELF file SHA256:  61c17d85c726b8e4...
I (290) cpu_start: ESP-IDF:          v4.4-beta1-dirty
I (296) heap_init: Initializing. RAM available for dynamic allocation:
I (303) heap_init: At 3FC97C80 len 00028380 (160 KiB): DRAM
I (310) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (316) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (323) spi_flash: detected chip: generic
I (327) spi_flash: flash io: dio
I (332) sleep: Configure to isolate all GPIO pins in sleep state
I (338) sleep: Enable automatic switching of GPIO sleep configuration
I (345) cpu_start: Starting scheduler.

negativekelvin, 2021-12-12:
40mhz is not a valid option for production silicon

Rokachy, 2021-12-14:
is there is an option to run CPU slower than 80Mhz?

songruo, 2021-12-14:
Hi Rokachy,

May I ask what's the reason that you want to set CPU clock to 40MHz, and what's the application for it?

You could switch CPU clock source from PLL_CLK to XTAL_CLK, then CPU will be running at 40MHz. This can be done by calling `rtc_clk_cpu_freq_set_xtal()`. Note that using this API will  power off the PLL, it may affect some peripherals that you are using. If you don't want to turn off PLL, you could try `rtc_clk_cpu_freq_to_xtal(40, 1) `. Overall, we probably won't recommend you set CPU clock source to XTAL_CLK, since APB clock frequency will also be changed. You can check ESP32C3 TRM Chapter 6.2 for more details.

Rokachy, 2021-12-14:
Hi,

The product need to send bluetooth non-connectable messages every x minutes.
The product is battery powered and current consumption is a critical
parameter for us.
We are evaluating if we can use the ESP32-C3 for this.
The plan is to put the device at deep sleep mode and wakeup it every
x-minutes.
I facing 2 issues to meet our power consumption target:
   1. During the time device is active, to reduce power,I would like to run
CPU at minimum and keep BT functionality.
   2.  power up time from sleep is too long, few hundreds of msecs till it
get to main function .

Thank you,
Yehuda

xiongweichao, 2021-12-17:
@Rokachy 

 1. During the time device is active, to reduce power,I would like to run CPU at minimum and keep BT functionality.
 -  Bluetooth is not able to work under 40MHz frequency on ESP32-C3.  If you want to reduce power consumption, you can enable light sleep and DFS.

2.  power up time from sleep is too long, few hundreds of msecs till it get to main function .
 - I don't understand what it means, can you provide more details.

Thanks

Alvin1Zhang, 2022-03-03:
@Rokachy Thanks for reporting, would you please help share if any updaters for the issue? Thanks.

Rokachy, 2022-03-03:
The current consumption doesn't meet our requirements. This issue can be
closed.

closed 2022-03-08 by Alvin1Zhang