espressif/esp-idf #8071: Can't change BluFi device name

htomer5, 2021-12-12:
Hello ! 

**Is your feature request related to a problem? Please describe.**

We use BluFi  to config WiFi settings via Bluetooth, but I can't set custom device name because it is defined inside esp_idf. 

**Describe the solution you'd like**

I opened a pull request with my solution. see #8070.

Thanks.

xiewenxiang, 2021-12-13:
@htomer5 

If we put the blufi name in menuconfig, does it  meet your needs?

htomer5, 2021-12-13:
> 

Yes, it'll be perfect. I just need a way to change the device name in my project.

Thanks for all your help!

asypost, 2022-01-10:
@xiewenxiang I think it's better to make **esp_bt_dev_set_device_name** work

rtheil-growlink, 2022-04-15:
It would be much better if the value were not hard-coded as a config option. If a person has 5 of these devices they are trying to set up, how are they to know which one is which when the value is hard coded?

I agree with @asypost that setting the name with esp_bt_dev_set_device_name is a better way to make this happen. This way the app can match a serial number labeled on the device.

asypost, 2022-06-22:
Is there any progress since six month passed away? @xiewenxiang

rtheil-growlink, 2022-06-22:
Since Espressif has not improved this process, we established a way to get around it. We use the custom data callback of blufi. The mobile device sends a command and we respond to that command with the name/serial of the device. Then we're able to properly configure and display that name/serial to the end user.

esp_blufi_cb_event_t has ESP_BLUFI_EVENT_RECV_CUSTOM_DATA

asypost, 2022-06-23:
> Since Espressif has not improved this process, we established a way to get around it. We use the custom data callback of blufi. The mobile device sends a command and we respond to that command with [...]

I found a way to change the device name of the blufi device when bluedroid is used.
You can change you device name via `esp_blufi_register_callbacks(&esp_blufi_callbacks);`
The code below shows how you should do:
`static esp_blufi_callbacks_t esp_blufi_callbacks = {
    .event_cb = esp_blufi_event_handle,
    .negotiate_data_handler = blufi_dh_negotiate_data_handler,
    .encrypt_func = blufi_aes_encrypt,
    .decrypt_func = blufi_aes_decrypt,
    .checksum_func = blufi_crc_checksum};
static void esp_blufi_event_handle(esp_blufi_cb_event_t event, esp_blufi_cb_param_t *param)
{
    /* actually, should post to blufi_task handle the procedure,
     * now, as a example, we do it more simply */
    switch (event)
    {
    case ESP_BLUFI_EVENT_INIT_FINISH:
        ESP_LOGI(TAG, "BLUFI init finish\n");

        esp_ble_gap_set_device_name(BLUFI_DEV_NAME);
        esp_ble_gap_config_adv_data(&blufi_adv_data);
        break;
    case ESP_BLUFI_EVENT_DEINIT_FINISH:
        ESP_LOGI(TAG, "BLUFI deinit finish\n");
        break;
    case ESP_BLUFI_EVENT_BLE_CONNECT:
        ESP_LOGI(TAG, "BLUFI ble connect\n");
        ble_is_connected = true;
        esp_blufi_adv_stop();
        blufi_security_init();
        break;
    case ESP_BLUFI_EVENT_BLE_DISCONNECT:
        ESP_LOGI(TAG, "BLUFI ble disconnect\n");
        ble_is_connected = false;
        blufi_security_deinit();
        **esp_ble_gap_set_device_name(BLUFI_DEV_NAME);
        esp_ble_gap_config_adv_data(&blufi_adv_data);**
        break;
    case ESP_BLUFI_EVENT_SET_WIFI_OPMODE:
        ESP_LOGI(TAG, "BLUFI Set WIFI opmode %d\n", param->wifi_mode.op_mode);
        ESP_ERROR_CHECK(esp_wifi_set_mode(param->wifi_mode.op_mode));
        break;
    case ESP_BLUFI_EVENT_REQ_CONNECT_TO_AP:
        ESP_LOGI(TAG, "BLUFI requset wifi connect to AP\n");
        /* there is no wifi callback when the device has already connected to this wifi
        so disconnect wifi before connection.
        */
        esp_wifi_disconnect();
        esp_wifi_connect();
        break;
    case ESP_BLUFI_EVENT_REQ_DISCONNECT_FROM_AP:
        ESP_LOGI(TAG, "BLUFI requset wifi disconnect from AP\n");
        esp_wifi_disconnect();
        break;
    case ESP_BLUFI_EVENT_REPORT_ERROR:
        ESP_LOGE(TAG, "BLUFI report error, error code %d\n", param->report_error.state);
        esp_blufi_send_error_info(param->report_error.state);
        break;
    case ESP_BLUFI_EVENT_GET_WIFI_STATUS:
    {
        wifi_mode_t mode;
        esp_blufi_extra_info_t info;

        esp_wifi_get_mode(&mode);

        if (gl_sta_connected)
        {
            memset(&info, 0, sizeof(esp_blufi_extra_info_t));
            memcpy(info.sta_bssid, gl_sta_bssid, 6);
[... 95 lines cut ...]
    case ESP_BLUFI_EVENT_RECV_USERNAME:
        /* Not handle currently */
        break;
    case ESP_BLUFI_EVENT_RECV_CA_CERT:
        /* Not handle currently */
        break;
    case ESP_BLUFI_EVENT_RECV_CLIENT_CERT:
        /* Not handle currently */
        break;
    case ESP_BLUFI_EVENT_RECV_SERVER_CERT:
        /* Not handle currently */
        break;
    case ESP_BLUFI_EVENT_RECV_CLIENT_PRIV_KEY:
        /* Not handle currently */
        break;
        ;
    case ESP_BLUFI_EVENT_RECV_SERVER_PRIV_KEY:
        /* Not handle currently */
        break;
    default:
        break;
    }
}
`
`esp_ble_gap_set_device_name(BLUFI_DEV_NAME); esp_ble_gap_config_adv_data(&blufi_adv_data);` is the key.

jatibayuworo, 2022-12-10:
maybe the problem in setting apps use by phone.
apps name: EspBlufi
1. change name #define BLUFI_DEVICE_NAME   "BLUEFI_soBLUEz"
2. build then flash
3. open apps
4. taps icon beside title "EspBlufi" (like rectangle and arrow)
5. on "BLE device filtering" change it blank or same with the name of device name
6. restart apps

i did that and device name show in phone is same like i set in esp.

sorry if the problem mentions above not like i assume

zhp0406, 2023-11-16:
I am conducting tests on my end, and the new name is displayed after manual refresh. If there are still issues, please contact me again.

closed 2023-11-21 by Indastri

martmalo, 2024-06-19:
Cannot get it working by just changing `#define BLUFI_DEVICE_NAME "BLUEFI_soBLUEz"`
What do you mean by _manual refresh_ @zhp0406?

zhp0406, 2024-06-20:
@jatibayuworo 
Have you tried refreshing it by swiping down in your phone?

jatibayuworo, 2024-07-17:
i didn't found the problem. i only mention that it work in mine. and i tried to summarize what i've done.