espressif/esp-idf #8074: Cannot disable bluetooth - Bludroid already enabled, do disable first

Gabriel-Gardin, 2021-12-13:
## Environment

- Development Kit:      ESP32-Wrover-Kit
- Kit version (for WroverKit/PicoKit/DevKitC): v1
- Module or chip used:  ESP32-WROOM-32
- IDF version: Master
- Build System:         idf.py

### Description:
Canot disable bluetooth, calling esp_bluedroid_deinit();  results in the following LOG message: _BT_LOG: Bludroid already enabled, do disable first_. 

After investigating, i found that in /bluedroid/api/esp_bt_main.c the esp_err_t esp_bluedroid_deinit(void) function has the following lines:

`    if (bd_already_enable) {
        LOG_ERROR("Bludroid already enabled, do disable first\n");
        return ESP_ERR_INVALID_STATE;
    }` 

Since the objective of the function is to deinit bluedroid, should not be checking if BD is already disable instead?

### Expected Behavior
Be able to deinit bluedroid

### Actual Behavior
Bluedroid stay active

### Steps to reproduce

1. Start BLE
2. Try to disable it with  esp_bluedroid_deinit();

### Code to reproduce this issue
`
void ble_ibeacon_init(esp_gap_ble_cb_t callback)
{
    const esp_timer_create_args_t ibeacon_timer_args = {
        .callback = ble_ibeacon_destroy, //Cria um timer para terminar o ibeacon
        /* argument specified here will be passed to timer callback function */
        .arg = NULL,
        .name = "ibeacon_timer"
    };

    ESP_ERROR_CHECK(esp_timer_create(&ibeacon_timer_args, &ibeacon_timer));
    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    esp_bt_controller_init(&bt_cfg);
    esp_bt_controller_enable(ESP_BT_MODE_BLE);

    esp_bluedroid_init();
    esp_bluedroid_enable();
    ble_ibeacon_appRegister(callback);
}

void stop_ibeacon()
{
    esp_ble_gap_stop_advertising();
}

void ble_ibeacon_destroy()
{
    stop_ibeacon();
    esp_bluedroid_deinit();
    esp_bluedroid_disable();
}
`

## Debug Logs

```
E (11706) BT_LOG: Bludroid already enabled, do disable first
```

xiewenxiang, 2021-12-17:
Please refer to this example.
[gatt_server_service_table.zip](https://github.com/espressif/esp-idf/files/7734085/gatt_server_service_table.zip)

Gabriel-Gardin, 2021-12-23:
Thanks, I was calling the functions in the wrong order...

closed 2021-12-23 by Gabriel-Gardin