espressif/esp-idf #8075: Sleep Mode With Size Optimization Fails

jpmreece, 2021-12-13:
## Environment

- Development Kit:      none - custom board (No Reason to believe that this would not affect Dev Kits)
- Module or chip used:  ESP32-C3
- IDF version (run ``git describe --tags`` to find it): v4.3
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt & Plain Command Prompt
- Using an IDE?: Yes - Eclipse
- Power Supply:         USB

## Problem Description

Entering Light Sleep Power mode with Optimisation set to anything other than Optimize Debug (-Og) causes a fault and reset. With Nimble and Wifi Enable

### Expected Behavior

The device should enter low power mode and wake on a given wake source.

### Actual Behavior

The ESP32-C3 resets with a core dump.

### Steps to reproduce

1. Open the SDK Config for the Light Sleep Example
2. Change the Compiler Options -> Optimization Level to "Optimize For Size (-Os)
3. Build and Program an ESP32-C3 module

[image]

### Code to reproduce this issue
Can be found under the light sleep example:
https://github.com/espressif/esp-idf/tree/master/examples/system/light_sleep

## Debug Logs

```
x<���ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382250
0x40382250: esp_restart_noos at C:/ide/esp-idf/components/esp32c3/system_api_esp32c3.c:137 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1730
load:0x403ce000,len:0x8d4
load:0x403d0000,len:0x293c
entry 0x403ce000
I (35) boot: ESP-IDF v4.3-dirty 2nd stage bootloader
I (35) boot: compile time 14:28:16
I (35) boot: chip revision: 3
I (38) boot.esp32c3: SPI Speed      : 80MHz
I (42) boot.esp32c3: SPI Mode       : DIO
I (47) boot.esp32c3: SPI Flash Size : 4MB
I (51) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (75) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (83) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  3 factory          factory app      00 00 00010000 00100000
I (97) boot:  4 ota_0            OTA app          00 10 00110000 00100000
I (105) boot:  5 ota_1            OTA app          00 11 00210000 00100000
I (112) boot: End of partition table
I (117) boot: Defaulting to factory image
I (121) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=06750h ( 26448) map
I (134) esp_image: segment 1: paddr=00016778 vaddr=3fc89200 size=01b28h (  6952) load
[... 141 lines cut ...]
3fc8e470: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e490: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x303135a5
3fc8e4b0: 0xffffffff 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e4d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e4f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e510: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e530: 0x00000243 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x3c0216d8 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e550: 0x00000013 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000001 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e570: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e590: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e5b0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e5d0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e5f0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e610: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc8e630: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x40385654
3fc8e650: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x3fc8e92c 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x40384496

ELF file SHA256: 9de9d90f0e1d637b
ELF file SHA256: 9de9d90f0e1d637b

```

## Other items if possible

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/7706611/sdkconfig.txt)

felmue, 2021-12-13:
Hello @jpmreece 

My ESP32C3 runs the light sleep example just fine (no crash) with any of the four optimization levels. My setup:
- Development Kit: M5StampC3
- Module or chip used: ESP32-C3
- IDF version (run git describe --tags to find it): v4.3
- Build System: idf.py
- Compiler version (run xtensa-esp32-elf-gcc --version to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Ubuntu 20.04
- (Windows only) environment type: CLI
- Using an IDE?: No
- Power Supply: USB

I've noticed in your log file that your ESP-IDF copy seems to have local changes, e.g.
```
I (265) cpu_start: App version:      v4.3-dirty
```
Not sure though whether that is relevant.

Build sizes for reference:
- -Og 165824
- -Os 152928
- -O2 158672
- -O0 240912

Thanks
Felix

jpmreece, 2021-12-14:
Hi @felmue,

It sounds like an oddity for my setup. I have attached a patch file for my changes which includes:

1. a modification in http_on_headers_complete to ensure that http coms continues in a certain instance (I believe there is a pull request outstanding for this)
2. SDK config for FreeRTOS for configUSE_TIMERS
3. Pin number change for WakeUp source in light_sleep_example_main

[v4.3-dirty.patch.txt](https://github.com/espressif/esp-idf/files/7710041/v4.3-dirty.patch.txt)

Thank you for looking at this,
Jon

esp-lis, 2021-12-14:
> 

Can you provide the commit id of the release/v4.3 version you are based on and the elf file when the crash occurs?

jpmreece, 2021-12-14:
Hi, 

The Commit ID is: c9646ff0beffc86d2c6d1bfbad34da16e328e0e3
The Elf is attached.

[light_sleep_example.zip](https://github.com/espressif/esp-idf/files/7710470/light_sleep_example.zip)

esp-lis, 2021-12-14:
> Hi,
> 
> The Commit ID is: [c9646ff](https://github.com/espressif/esp-idf/commit/c9646ff0beffc86d2c6d1bfbad34da16e328e0e3) The Elf is attached.
> 
> [light_sleep_example.zip](https://github.com/espressif/esp-idf/files/7710470/light_sleep_example.zip)

Configure the log level to **Info** for testing, I guess it will not crash, if it is, please use the latest version of release/v4.3

The log level configure path in menuconfig as follows:
`(Top) > Component config > Log output > Default log verbosity`

jpmreece, 2021-12-14:
Awesone - thank you @esp-lis.

I will  update to the latest version as well.

closed 2021-12-14 by jpmreece