espressif/esp-idf #8077: http_server simple

Carol-001, 2021-12-14:
接收buffer大小设置的是4096，有时候接收到的少于4096时，很容易出现报 httpd_txrx: httpd_sock_err: error in recv : 11的错误，然后文件就传输失败了，http就死掉了，wifi连接还是正常的
[image]

Alvin1Zhang, 2021-12-14:
@Carol-001 感谢反馈，您能按照 [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=) 提供包括 elf, sdk configuration, backtrace, log outputs, commit ID, hardware 等更多信息么？这些信息将有助于我们 debug；另外，您链接中的截图没有显示出来，烦请修复，谢谢。

Carol-001, 2021-12-14:
[image]
[image]

ESP-YJM, 2021-12-15:
@Carol-001 请问这个也是你们提问的吗？https://github.com/espressif/esp-idf/issues/8079 是否可以按照这个 issue 来跟进？

Carol-001, 2021-12-16:
YES

ESP-YJM, 2021-12-16:
@Carol-001 麻烦关闭这个 issue，保留 https://github.com/espressif/esp-idf/issues/8079 来跟进。

closed 2021-12-16 by Carol-001