espressif/esp-idf #8078: esp32s2 idf4.3.1 hanging there

achao1104, 2021-12-14:
Hello,

My project is in production phase now, as observed, sometimes, some devices might be hanging there, no execution and no rebooting and also no core dump, just hanging there.
1) could you please provide some clues? 
2) shall we use another independent watch dog chip? any recommendations?

Peter

Alvin1Zhang, 2021-12-14:
@achao1104 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

achao1104, 2021-12-14:
sorry, there is no more info since it is placed in a very high position.
I just wondering that if you have the similar issues (hanging).

抄送: "Peter ***@***.******@***.***&gt;;

Alvin1Zhang, 2021-12-14:
Thanks for sharing the updates, would you please help share any details about use scenario, temperature, altitude etc? Thanks.

Alvin1Zhang, 2022-03-08:
@achao1104 Thanks for reporting, will close due to short of feedback, feel free to reopen with more details, thanks.

closed 2022-03-08 by Alvin1Zhang