espressif/esp-idf #8079: An exception occurred while receiving the file through the HTTP server.

CYCHEN97, 2021-12-14:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32S2]
- IDF version (run ``git describe --tags`` to find it):
   v4.4-dev-3675-g35b20cadce
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [VSCode]
- Power Supply:  [USB]

## Problem Description

### Expected Behavior
The mobile APP connects to the AP of the ESP32S2, accesses the HTTP server of the ESP32S2, sends the file to the ESP32S2 through POST, and saves the file to the TF card when the ESP32S2 receives the file.

### Actual Behavior
The post buffer is set to 4096, but during transmission, the number of buffers received is sometimes less than 4096 bytes. If the number of consecutive packets received is less than 4096 bytes, an error of "httpd_txrxx: httpd_sock_err: error in recv: 11" is likely to occur. Then the HTTP server dies and the WiFi AP continues to work.

### Steps to reproduce

1. Open AP mode, set max_connection to 1, set static to IP192.168.8.1
2. Start the HTTP server and wait for the user to connect to the AP to send the file to ESP32S2.
3. In the process of file transfer, the received buffer is less than the configured buffer size occasionally. Once this phenomenon occurs, the HTTP server will die. At this time, WiFi is still working normally. However, this phenomenon is not inevitable every time, and it needs many tests to appear.
4. Through the test, it is found that the error"Httpd_txrxx: httpd_sock_err: error in recv: 11" is due to file timeout, so the default value of "recv_wait_timeout" in "httpd_config_timeout" has been changed to 100 instead of 5, but this problem persists even after the timeout is increased.

### Code to reproduce this issue
https://gist.github.com/CYCHEN97/ef96694d0d777925a67344fb1d3ef4f3
[image]

## Debug Logs
1.
[0;32mI (22261) example: bufsize:4096
[0m
[0;32mI (22274) SD_Write: writesize:4096
[0m
[0;32mI (22294) example: bufsize:4096
[0m
[0;32mI (22307) SD_Write: writesize:4096
[0m
[0;32mI (22328) example: bufsize:4096
[0m
[0;32mI (22339) SD_Write: writesize:4096
[0m
[0;32mI (22359) example: bufsize:4096
[0m
[0;32mI (22370) SD_Write: writesize:4096
[0m
[0;32mI (22391) example: bufsize:1696
[0m
[0;32mI (22400) SD_Write: writesize:1696
[0m
[0;32mI (23416) example: bufsize:2880
[0m
[0;32mI (23425) SD_Write: writesize:2880
[0m
[0;32mI (30555) example: bufsize:512
[0m
[0;32mI (30564) SD_Write: writesize:512
[0m
[0;32mI (38575) example: bufsize:512
[0m
[0;32mI (38584) SD_Write: writesize:512
[0m
[0;32mI (38890) example: bufsize:416
[0m
[0;32mI (38899) SD_Write: writesize:416
[0m
[0;32mI (39263) example: bufsize:512
[0m
[0;32mI (39272) SD_Write: writesize:512

2.
[0;32mI (152517) example: bufsize:1128
[0m
[0;32mI (152526) SD_Write: writesize:1128
[0m
[0;32mI (152821) example: bufsize:688
[0m
[0;32mI (152830) SD_Write: writesize:688
[0m
[0;32mI (158768) example: bufsize:188
[0m
[0;32mI (158777) SD_Write: writesize:188
[0m
[0;32mI (159580) example: bufsize:188
[0m
[0;32mI (159588) SD_Write: writesize:188
[0m
[0;32mI (167374) example: bufsize:188
[0m
[0;32mI (167382) SD_Write: writesize:188
[0m
[0;32mI (182681) example: bufsize:188
[0m
[0;32mI (182690) SD_Write: writesize:188
[0m
[0;32mI (183030) example: bufsize:188
[0m
[0;32mI (183039) SD_Write: writesize:188
[0m
[0;32mI (183438) example: bufsize:124
[0m
[0;32mI (183447) SD_Write: writesize:124
[0m
[0;32mI (183919) example: bufsize:188
[0m
[0;32mI (183927) SD_Write: writesize:188
[0m
[0;33mW (283941) httpd_txrx: httpd_sock_err: error in recv : 11[0m
[0;32mI (283941) example: HTTPD_SOCK_ERR_TIMEOUT

## Other items if possible

- [ ] [sdkconfig.txt](https://github.com/espressif/esp-idf/files/7709599/sdkconfig.txt)

ESP-YJM, 2021-12-15:
@CYCHEN97 The errno means the receive timeout. But we don’t know if the phone did not send data or the device server did not receive the data. 
[print_tcp_data.zip](https://github.com/espressif/esp-idf/files/7716074/print_tcp_data.zip)
Could you please help modify the file like the attachment patch. I want to see the TCP data flow. It would be better if you can grab the wifi air packet for comparison.

CYCHEN97, 2021-12-15:
[SaveWindows2021_12_15_12-12-58.TXT](https://github.com/espressif/esp-idf/files/7716419/SaveWindows2021_12_15_12-12-58.TXT)

CYCHEN97, 2021-12-15:
Sorry, I don't know how to grab air pack data for comparison.

CYCHEN97, 2021-12-15:
> @CYCHEN97 The errno means the receive timeout. But we don’t know if the phone did not send data or the device server did not receive the data. [...]

@ESP-YJM This is the TCP data stream log.
[SaveWindows2021_12_15_12-12-58.TXT](https://github.com/espressif/esp-idf/files/7716692/SaveWindows2021_12_15_12-12-58.TXT)
 I'm really sorry that I don't know how to get air pack data for comparison.

ESP-YJM, 2021-12-15:
@CYCHEN97 For your question 
1. Why received is sometimes less than 4096 bytes.
    Although the buffer size is 4096, the `recv` function reads data from the  LWIP data buffer, and reads as much data as there is. If the data in LWIP data buffer is less than 4096, the recv function will read the actual data and return, will not wait until the data to 4096, which is what you see that data size is less than 4096. Sometimes, poor WIFI environment or peer data sent logic will make TCP retransmission, reduce the data transmission rate，and the LWIP data buffer maybe less than 4096. I think this is a normal issue. If you want to reduce this, you can try change the `TCP_WND` and your buffer size. In LWIP, `TCP_MSS` is `1440`, and default `TCP_WND` is `1440 * 4`,you can change the buffer size to integer multiples of `TCP_MSS`, such as 4320, and enlarge the `TCP_WND` to `2*4320`,or only reduce your buffer size to `1440*2`.

2. From your log,  It seems that the log is not over yet, what about the follow-up log, such as the log when an error occurs(httpd_sock_err: error in recv: 11).

CYCHEN97, 2021-12-15:
@ESP-YJM
[image]
Next, esp32s2 has been waiting for receiving information in the file handler. After a minute or two, the mobile app prompts that sending the file failed. Later, all HTTP services of esp32s2 are unavailable and the log shows 
"[0;33mW (1068873) httpd_txrx: httpd_sock_err: error in recv : 11[0m
[0;32mI (1068873) example: HTTPD_SOCK_ERR_TIMEOUT".

ESP-YJM, 2021-12-15:
@CYCHEN97 I want to know from the log `[0;32mI (245488) TEST: @@ WiFi Tx TCP - L:40, S:811650527, A:1467562905, SP:80, DP:40326, F:10[0m` to `�[0;33mW (1068873) httpd_txrx: httpd_sock_err: error in recv : 11�[0m`, whether the mobile app send data to http server? So could you paste the whole log. From the log, i only know we ask the mobile app to send TCP sequence number 1467562905 data. I'm not sure if the phone sent.

CYCHEN97, 2021-12-15:
@ESP-YJM
The mobile APP debugging tool fails to display Post.
[image]
[image]
Esp32s2 is still waiting to receive information when canceling resending the file.
[SaveWindows2021_12_15_16-18-33.TXT](https://github.com/espressif/esp-idf/files/7717541/SaveWindows2021_12_15_16-18-33.TXT)

ESP-YJM, 2021-12-15:
@CYCHEN97 Thanks for offering the whole log. From the log, it seems the http server still want to receive the data, but now mobile app not send data, maybe like the photo you pasted above, it fails to post. From the TCP protocol itself, i think the mobile app has the problem. Because the http server request the mobile app send TCP sequence number 2273107275 data, and mobile app not send. So http server receive data timeout. Could you please try another tool to upload file to server? Such as postman, or other http client tool.

[image]

CYCHEN97, 2021-12-20:
@ESP-YJM 
Now, the file receiving timeout is added to esp32s2 to exit the current file receiving after two times, which solves the problem that the HTTP server has been waiting for the current file information and cannot respond to other requests. However, if there is an exception in receiving a file once, it is easy to see another exception in receiving a file when the app re sends a file. The following is the log information of three consecutive failures in receiving a file.
[log_202112201501.TXT](https://github.com/espressif/esp-idf/files/7743189/log_202112201501.TXT)

ESP-YJM, 2021-12-21:
@CYCHEN97 I think this problem is the same with before. The mobile app not send data and receive timeout.
[image]

CYCHEN97, 2021-12-21:
@ESP-YJM
However, in this case, if restart esp32s2, it will generally succeed.

ESP-YJM, 2021-12-21:
@CYCHEN97 Could you give me the whole log.

CYCHEN97, 2021-12-21:
@ESP-YJM
I don't think it's our app's problem. I found the HTTP tool of "网络万用表" on the Internet for testing, and there will also be sending exceptions.
[log_202112211413.TXT](https://github.com/espressif/esp-idf/files/7750088/log_202112211413.TXT)
[image]
[image]

ESP-YJM, 2021-12-21:
@CYCHEN97 Yes, i see the receive timeout  in 5s mobile app not send data until after 6s, it send the data that the server want.

[image]

And in here, also the receive timeout, but the mobile app maybe close the TCP, and the server re-send data to mobile app, it send RST to close the connection.
[image]

Could you please try close the WIFI sleep mode and try to test whether it can reduce the receive timeout.
You can call `esp_wifi_set_ps(0)` before you call `esp_wifi_start`.

CYCHEN97, 2021-12-21:
Redmi Note 10 Pro 
Mi Note 3

CYCHEN97, 2021-12-21:
@ESP-YJM
The latest test found that when sending files using "Mi note 3 mobile phone", it is easy to have file sending exceptions. When sending files using "redmi note 10 Pro mobile phone" / "Huawei yal-al 10 mobile phone", there is no file sending exception. I'd like to know whether this exception of sending file is related to the mobile phone?

ESP-YJM, 2021-12-21:
@CYCHEN97 It maybe releated to implementation of the WIFI protocol of various mobile phone manufacturers, the size of the TCP window and the sending mechanism.

CYCHEN97, 2021-12-21:
@ESP-YJM
If it is really due to the mobile phone, if the user's mobile phone is similar to mine, there will be many complaints about poor product performance in the future. Is there any way to make the file transmission more stable and reliable?

ESP-YJM, 2021-12-21:
@CYCHEN97 You can call `esp_wifi_set_ps(0)` before you call `esp_wifi_start` to test.

CYCHEN97, 2021-12-21:
> @CYCHEN97 You can call `esp_wifi_set_ps(0)` before you call `esp_wifi_start` to test.
After calling "esp_wifi_set_ps (0)", the file transfer exception still exists.

CYCHEN97, 2021-12-21:
@ESP-YJM
After calling "esp_wifi_set_ps (0)", the file transfer exception still exists.

ESP-YJM, 2021-12-27:
@CYCHEN97 I think we need to locate the problem from the TCP protocol layer, whose problem is whose problem is, and this kind of problem is not a compatibility problem. Can you grab the WIFI air packet? I need to use this to confirm the problem.

CYCHEN97, 2021-12-27:
@ESP-YJM
I haven't grabbed WiFi air bag before. How do I grab this?

ESP-YJM, 2021-12-27:
@CYCHEN97 You can use OmniPeek to capture. Before capture, you may need buy a Wireless network card.

ESP-YJM, 2021-12-27:
@CYCHEN97 You can follow this https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32/api-guides/wireshark-user-guide.html. Note, set the AP as open mode and not set password.

ESP-YJM, 2022-01-17:
@CYCHEN97 Do you have any update for this issue? If not, we will close this issue.

Alvin1Zhang, 2022-01-19:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details.

closed 2022-01-19 by Alvin1Zhang