espressif/esp-idf #8082: 使用离线包安装后，为什么还要获取esp-idf

xcmonline, 2021-12-14:
在环境搭建中有对系统生成的二个目录有疑问，我是使用离线安装包安装的。
安装在%userprofile%下，也就是 C:\Users\Administrator

安装的主目录是 C:\Users\Administrator\Desktop\esp-idf

1、但系统中会生成  C:\Users\Administrator\.espressif 这个目录，与主目录都有 tools，这二个目录有什么区别。下次如果升级，是直接修改主目录吗？

2、有了 C:\Users\Administrator\Desktop\esp-idf 目录，为什么还要从 github中下载  esp-idf 这二个目录应该是一样的吧。如果不下载可以吗？这二个目录如何升级。

在环境搭建中有对系统生成的二个目录有疑问，我是使用离线安装包安装的。
安装在%userprofile%下，也就是 C:\Users\Administrator

安装的主目录是 C:\Users\Administrator\Desktop\esp-idf

1、但系统中会生成  C:\Users\Administrator\.espressif 这个目录，与主目录都有 tools，这二个目录有什么区别。下次如果升级，是直接修改主目录吗？

2、有了 C:\Users\Administrator\Desktop\esp-idf 目录，为什么还要从 github中下载  esp-idf 这二个目录应该是一样的吧。如果不下载可以吗？这二个目录如何升级。

xiaolongba, 2021-12-14:
环境搭建的问题，你可以参考我写的教程[《基于VSCODE的ESP32C3开发环境的搭建》](https://docs.wireless-tech.cn/doc/6/)

leeebo, 2021-12-15:
@xcmonline Hi, 关于你的问题：
1. C:\Users\Administrator.espressif 的 tools 主要用于存放编译工具链（.exe），esp-idf 的 tool 主要用于存放编译脚本（.cmake），两者是分开维护的。
2. 是否可以将安装 log 添加到附件？log 默认存放你的 `C:\Users\Administrator\AppData\Local\Temp` 目录，名称是 `Setup Log 2021-12-?? #??.txt`
3. 建议使用最新版本离线安装工具（version >= 2.12） https://dl.espressif.com/dl/esp-idf/

closed 2021-12-15 by xcmonline

georgik, 2022-03-24:
Solved in https://github.com/espressif/idf-installer/issues/92
Please, use latest installer https://github.com/espressif/idf-installer/releases