espressif/esp-idf #8083: lcd - lvgl example fails to build if I run menuconfig form VSCode - Insiders

wreyford, 2021-12-14:
## Environment

- Development Kit:      [ESP32-DevKitC]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4]
- Module or chip used:  [ESP32-WROOM-32E]
- IDF version (run ``git describe --tags`` to find it):
    release/4.4 (release branch)
- Build System:         [CMake]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    outputs:
```
xtensa-esp32-elf-gcc: The term 'xtensa-esp32-elf-gcc' is not recognized as a name of a cmdlet, function, script file, or executable program.

Check the spelling of the name, or if a path was included, verify that the path is correct and try again.
```
- Operating System:     [Windows]
- (Windows only) environment type: [PowerShell].
- Using an IDE?: [VSCode-Insiders]
```
Version: 1.64.0-insider (user setup)
Commit: 745ef8b8264e12b2a704f6b2b60a2ce2cabfcda3
Date: 2021-12-14T05:16:43.978Z
Electron: 13.5.2
Chromium: 91.0.4472.164
Node.js: 14.16.0
V8: 9.1.269.39-electron.0
OS: Windows_NT x64 10.0.19044
```
- Power Supply:         [USB|]

## Problem Description

Failure to build once ESP-IDF SDK Configuration Editor (menuconfig) has been run.
See build log pasted below. 
As long as you dont run menuconfig, all is fine.
As soon as you have run menuconfig, the project will fail to build.

### Expected Behavior
Using menuconfig, should not cause the project not to build with cmake errors.

### Steps to reproduce

1. I import the LCD, LVGL example from esp-idf using F1 show example projects from within the Visual Studio Code - Insiders IDE.
2. I then set the Espressif device target to ESP32, using the ESP-IDF Set Espressif Device Target icon.

(If at this stage you ESP-IDF Build Flash and Monitor all works just fine.)

But if you proceed to step three without building the project
3. If you first open menuconfig, using ESP-IDF SDK Configuration Editor (menuconfig), save and exit or just exit.
4. Build the project ... it will fail to build.

### Code to reproduce this issue

```The example is imported as is with no modification.

## Debug Logs

```
> Executing task: cmake --build . <

[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
CMake Warning at C:/esp/esp-idf/tools/cmake/build.cmake:482 (message):
  "idf_component.yml" file was found for components:

        C:/ESP32/Projects/lvgl/main

  However, the component manager is not enabled.
Call Stack (most recent call first):
  C:/esp/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)    
  CMakeLists.txt:4 (project)

-- Project sdkconfig file C:/ESP32/Projects/lvgl/sdkconfig
Loading defaults file C:/ESP32/Projects/lvgl/sdkconfig.defaults...
-- Could NOT find Perl (missing: PERL_EXECUTABLE) 
-- App "lcd_lvgl" version: 1
-- Adding linker script C:/ESP32/Projects/lvgl/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script C:/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.eco3.ld
-- Adding linker script C:/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: C:/esp/esp-idf/components/app_trace C:/esp/esp-idf/components/app_update C:/esp/esp-idf/components/asio C:/esp/esp-idf/components/bootloader C:/esp/esp-idf/components/bootloader_support C:/esp/esp-idf/components/bt C:/esp/esp-idf/components/cbor C:/esp/esp-idf/components/cmock C:/esp/esp-idf/components/coap C:/esp/esp-idf/components/console C:/esp/esp-idf/components/cxx C:/esp/esp-idf/components/driver C:/esp/esp-idf/components/efuse C:/esp/esp-idf/components/esp-tls C:/esp/esp-idf/components/esp32 C:/esp/esp-idf/components/esp_adc_cal C:/esp/esp-idf/components/esp_common C:/esp/esp-idf/components/esp_eth C:/esp/esp-idf/components/esp_event C:/esp/esp-idf/components/esp_gdbstub C:/esp/esp-idf/components/esp_hid C:/esp/esp-idf/components/esp_http_client C:/esp/esp-idf/components/esp_http_server C:/esp/esp-idf/components/esp_https_ota C:/esp/esp-idf/components/esp_https_server C:/esp/esp-idf/components/esp_hw_support C:/esp/esp-idf/components/esp_ipc C:/esp/esp-idf/components/esp_lcd C:/esp/esp-idf/components/esp_local_ctrl C:/esp/esp-idf/components/esp_netif C:/esp/esp-idf/components/esp_phy C:/esp/esp-idf/components/esp_pm C:/esp/esp-idf/components/esp_ringbuf C:/esp/esp-idf/components/esp_rom C:/esp/esp-idf/components/esp_serial_slave_link C:/esp/esp-idf/components/esp_system C:/esp/esp-idf/components/esp_timer C:/esp/esp-idf/components/esp_websocket_client C:/esp/esp-idf/components/esp_wifi C:/esp/esp-idf/components/espcoredump C:/esp/esp-idf/components/esptool_py C:/esp/esp-idf/components/expat C:/esp/esp-idf/components/fatfs C:/esp/esp-idf/components/freemodbus C:/esp/esp-idf/components/freertos C:/esp/esp-idf/components/hal C:/esp/esp-idf/components/heap C:/esp/esp-idf/components/idf_test C:/esp/esp-idf/components/ieee802154 C:/esp/esp-idf/components/jsmn C:/esp/esp-idf/components/json C:/esp/esp-idf/components/libsodium C:/esp/esp-idf/components/log C:/esp/esp-idf/components/lwip C:/ESP32/Projects/lvgl/main C:/esp/esp-idf/components/mbedtls C:/esp/esp-idf/components/mdns C:/esp/esp-idf/components/mqtt C:/esp/esp-idf/components/newlib C:/esp/esp-idf/components/nghttp C:/esp/esp-idf/components/nvs_flash C:/esp/esp-idf/components/openssl C:/esp/esp-idf/components/openthread C:/esp/esp-idf/components/partition_table C:/esp/esp-idf/components/perfmon C:/esp/esp-idf/components/protobuf-c C:/esp/esp-idf/components/protocomm C:/esp/esp-idf/components/pthread C:/esp/esp-idf/components/sdmmc C:/esp/esp-idf/components/soc C:/esp/esp-idf/components/spi_flash C:/esp/esp-idf/components/spiffs C:/esp/esp-idf/components/tcp_transport C:/esp/esp-idf/components/tcpip_adapter C:/esp/esp-idf/components/tinyusb C:/esp/esp-idf/components/ulp C:/esp/esp-idf/components/unity C:/esp/esp-idf/components/usb C:/esp/esp-idf/components/vfs C:/esp/esp-idf/components/wear_levelling C:/esp/esp-idf/components/wifi_provisioning C:/esp/esp-idf/components/wpa_supplicant C:/esp/esp-idf/components/xtensa
CMake Error at C:/esp/esp-idf/tools/cmake/component.cmake:369 (__component_get_property):
  __component_get_property Function invoked with incorrect arguments for
  function named: __component_get_property
Call Stack (most recent call first):
  CMakeLists.txt:8 (idf_component_get_property)

CMake Error at CMakeLists.txt:9 (target_compile_options):
  Cannot specify compile options for target "CONFIG_IDF_CMAKE" which is not
  built by this project.

-- Configuring incomplete, errors occurred!
See also "C:/ESP32/Projects/lvgl/build/CMakeFiles/CMakeOutput.log".
←[31mFAILED: ←[0mbuild.ninja 
C:\esp\.espressif\tools\cmake\3.20.3\bin\cmake.exe --regenerate-during-build -SC:\ESP32\Projects\lvgl -BC:\ESP32\Projects\lvgl\build
ninja: error: rebuilding 'build.ninja': subcommand failed
The terminal process "C:\Program Files\PowerShell\7\pwsh.exe -Command cmake --build ." terminated with exit code: 1.
```

## Other items if possible
So if using menuconfig in a project with CMakeLists.txt
that includes:

```
idf_component_get_property(lvgl_lib lvgl__lvgl COMPONENT_LIB)
target_compile_options(${lvgl_lib} PRIVATE "-Wno-empty-body" "-Wno-strict-prototypes")
```
it will fail to build therafter.

brianignacio5, 2021-12-15:
You should set `"idf.enableIdfComponentManager":  true` in your `.vscode/settings.json`. This project requires to use `idf-component-manager` which is currently disabled by default in the vscode extension.

wreyford, 2021-12-15:
Thx  @brianignacio5 that was just what I was missing.
It all works as it should now.
I didn't quite pick that up from the docs, but you were very clear on what was required.

closed 2021-12-15 by wreyford