espressif/esp-idf #8084: Error while attaching large header file in SMTP

MaazSk, 2021-12-14:
## Environment

- Development Kit:      [ESP32-DevKit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it): v4.3
   
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
  
- Operating System:     [Linux]

- Using an IDE?: [VS Code]
- Power Supply:         [USB]

## Problem Description

I am trying to implement SMTP to send log files to an email address. When the log files are small at that time I am able to send the email but when the log files increase in size at that time I get the error as `Last error was: -0x2a - BASE64 - Output buffer too small` also when the log files size increases I get the error as `Unable to encode file` 

The code snippet is as follows:
```

/* text file contents*/
    ESP_LOGE(TAG, "Testing error mechanism");

    FILE *read_file = fopen("/spiffs/log_file.txt", "r");
    size_t sz;
    fseek(read_file, 0L, SEEK_END);
    sz = ftell(read_file);

    fseek(read_file, 0L, SEEK_SET);
    ESP_LOGI(TAG, "size of file is: %d", sz);
    if (read_file == NULL)
    {
        printf("Unable to open file\r\n");
    }
    //buffer to read file data
    char *buffer = NULL;
    //allocate dynamic memory
    buffer = calloc(sz + 1, sizeof(char));
    //reading data from file
    fread(buffer, sz, 1, read_file);
    //printing read data from file
    // printf("file received is: %s\r\n", buffer);
    //Closing file
    fclose(read_file);
    char *b64_buffer = calloc(strlen(buffer)+1,sizeof(char));
    size_t b64_buffer_size = sz;

    //Only for testing purpose
    if(sz)
    {
        b64_buffer_size =sz + 128;
    }
    
    ret = mbedtls_base64_encode((unsigned char *)b64_buffer, b64_buffer_size,
                                &base64_len, (unsigned char *)buffer, sz);
    if (ret != 0)
    {
        ESP_LOGE(TAG, "Error in mbedtls encode! ret = -0x%x", -ret);
        goto exit;
    }
    len = snprintf((char *)buf, BUF_SIZE, "%s\r\n", b64_buffer);
    ret = write_ssl_data(&ssl, (unsigned char *)buf, len);
```

Thank you!

vikramdattu, 2021-12-15:
Looks like you are trying to encode and send whole file at one go. That may fail for various reasons including you may not be able to allocate buffer of file_sz.

Suggestions:
Allocate a fixed size of buffer, e.g., 2KB and `read-->base64-encode-->send` these chunks one by one?
Add sanitation checks for memory allocations.

MaazSk, 2021-12-15:
@vikramdattu Thank you for your reply.

What if my file size increases by 2KB at that time how should I handle is there a way to allocate dynamic memory to the buffer?

vikramdattu, 2021-12-15:
> @vikramdattu Thank you for your reply.
> 
> What if my file size increases by 2KB at that time how should I handle is there a way to allocate dynamic memory to the buffer?

You do not need to allocate buffer = file size! Allocate just 2KB and send remaining size in next iteration.

Something like this:

```
write_header();
open_file();
while (not_complete) {
    read_next_2_kb();
    encode_base64_and_write();
    ...
}
done_sending();
```

Please elaborate if I am getting your use case wrong.

MaazSk, 2021-12-16:
I tried breaking the file into chunks and then encode it and then sending it and it is working for me now.

Thank you!

MaazSk, 2021-12-22:
Hi @vikramdattu,

I am receiving another error now as my file size has increased now.

The error is as below

> Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  0 register dump:
> PC      : 0x4008e3f2  PS      : 0x00060d33  A0      : 0x8008ecef  A1      : 0x3ffcd1c0  
> 0x4008e3f2: block_is_free at /home/maaz/esp/esp-idf/components/heap/heap_tlsf_block_functions.h:91
>  (inlined by) block_merge_prev at /home/maaz/esp/esp-idf/components/heap/heap_tlsf.c:350
>  (inlined by) tlsf_free at /home/maaz/esp/esp-idf/components/heap/heap_tlsf.c:868
> 
> A2      : 0x3ffc867c  A3      : 0x3ffcd9e0  A4      : 0x00060d20  A5      : 0x0000014e  
> A6      : 0x000000a5  A7      : 0x00000001  A8      : 0x00014e08  A9      : 0x00000000  
> A10     : 0x00000037  A11     : 0x3ffcd9d8  A12     : 0x00000002  A13     : 0x00000036  
> A14     : 0x00000001  A15     : 0x0000000d  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00014e0c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
> 
> Backtrace:0x4008e3ef:0x3ffcd1c0 0x4008ecec:0x3ffcd1e0 0x40081a46:0x3ffcd200 0x4009033d:0x3ffcd220 0x400db056:0x3ffcd240 0x400da44d:0x3ffcdb80 0x4008cbfa:0x3ffcdbd0
> 0x4008e3ef: block_merge_prev at /home/maaz/esp/esp-idf/components/heap/heap_tlsf.c:349
>  (inlined by) tlsf_free at /home/maaz/esp/esp-idf/components/heap/heap_tlsf.c:868
> 
> 0x4008ecec: multi_heap_free_impl at /home/maaz/esp/esp-idf/components/heap/multi_heap.c:220
> 
> 0x40081a46: heap_caps_free at /home/maaz/esp/esp-idf/components/heap/heap_caps.c:305
> 
> 0x4009033d: free at /home/maaz/esp/esp-idf/components/newlib/heap.c:46
> 
> 0x400db056: smtp_client_task at /home/maaz/Desktop/Git/aux-device/template-app/build/../main/mail.c:681
> 
> 0x400da44d: check_time_for_mail at /home/maaz/Desktop/Git/aux-device/template-app/build/../main/ntp_time.c:155
> 
> 0x4008cbfa: vPortTaskWrapper at /home/maaz/esp/esp-idf/components/freertos/port/xtensa/port.c:168

The code snippet is as follows

```
FILE *read_file = fopen("/spiffs/log_file.txt", "r");
buf = (char *)calloc(1, BUF_SIZE);
        if (read_file == NULL)
        {
            ESP_LOGE(TAG, "Unable to open file");
        }
        while ((bytesRead = fread(buffer, 1, sizeof(buffer), read_file)) > 0)
        {

            // process bytesRead worth of data in buffer

        ret = mbedtls_base64_encode((unsigned char *)base64_buffer, sizeof(base64_buffer),
                                        &base64_len, (unsigned char *)buffer, bytesRead);
            // ESP_LOGI(TAG,"Olen is:%d",base64_len);
            if (ret != 0)
            {
                ESP_LOGE(TAG, "Error in mbedtls encode! ret = -0x%x", -ret);
                goto exit;
            }
            // offset1 += read_bytes;
            len = snprintf((char *)buf, BUF_SIZE, "%s\r\n", base64_buffer);
            ret = write_ssl_data(&ssl, (unsigned char *)buf, len);
            memset(buffer,0,sizeof(buffer));
            // ESP_LOGI(TAG,"Bytes read are %d",bytesRead);
      }
```

Here ```BUF_SIZE is 1024```

Thanks in advance

vikramdattu, 2021-12-22:
@MaazSk looks like it's some corruption in your code.

From crash log it is wrong memory access or double free in your code.
You can debug above crash by enabling gdb_stub from `menuconfig` and do some postmortem to find out the issue.

In your code, you are allocating to `buf`
`buf = (char *)calloc(1, BUF_SIZE);`
whereas `buffer` is used to read the FILE pointer (read_file).
I hope that's taken care of.

MaazSk, 2021-12-22:
It's been resolved sorry for delay in response just need to check the output now. But it is not crashing now

closed 2021-12-25 by MaazSk