espressif/esp-idf #8086: ESP32S3 Temperature Sensor undefined reference

brunohorta82, 2021-12-14:
undefined reference to `temp_sensor_start' when I try use  
#include "driver/temp_sensor.h"
temp_sensor_start();

BOARD ESP32S3

ginkgm, 2022-02-08:
Hi @brunohorta82 ,
Sorry to say that, this feature hasn't been implemented yet. It is planned to be added in IDF v5.0.

Alvin1Zhang, 2022-07-26:
Thanks for reporting, the feature has been implemented in https://github.com/espressif/esp-idf/commit/daceb3516d4159b2b733b4facb1ebc4f4339eb14, feel free to reopen.

closed 2022-07-26 by Alvin1Zhang

benninga, 2023-02-24:
@Alvin1Zhang - hi. I'm new to ESP32 and I'm trying to compile the temperature sensor example project and having issues.

I have an ESP32-WROOM-32E chip. And I'm using ESP-IDF for VS Code with ESP-IDF version v5.0.1

I've compiled other projects for the chip successfully. But, for this project I'm seeing two errors:

During compile I get this feailure:
`/esp32/temp_sensor/main/temp_sensor_main.c:18: undefined reference to `temperature_sensor_install'`

Also - the tempereature_sensor.h file contains an unresolved reference to to `TEMPERATURE_SENSOR_CLK_SRC_DEFAULT`.

Any suggestions? (Also - let me know if I should use a different ticket).

igrr, 2023-02-24:
@benninga please note that ESP32 is not in the list of chips which the temp_sensor example is compatible with; see "Supported targets" at the top of https://github.com/espressif/esp-idf/blob/master/examples/peripherals/temp_sensor/README.md.

Initally the ESP32 chip also had a temperature sensor in the list of features, but eventually it was removed as it was difficult to get useful readings from. Newer chips (ESP32-S2 and later) do have a temperature sensor that is supported by this example.

benninga, 2023-02-24:
Thanks @igrr !