espressif/esp-idf #8087: WPS error WIFI_EVENT_STA_WPS_ER_PBC_OVERLAP when updating from ESP-IDF_v4.3 to ESP-IDF_v4.3.1

carlessole, 2021-12-14:
Module or chip used: ESP32-WROVER (With ESP_PROG)
IDF version: v4.3.1
Build System: CMake/idf.py (usingEclipse Plugin)
Compiler version: esp-2020r3-8.4.0
Operating System: Windows
(Windows only) environment type: -.
Using an IDE?: Yes. Eclipse + ESP-IDF plugin, but also thorugh CLI
Power Supply: USB

After updating the ESP-IDF from v4.3 to v4.3.1 and recompiling a project without any change on the code, WPS is not working properly. Now device is contiously reporting the "**WIFI_EVENT_STA_WPS_ER_PBC_OVERLAP**"

Any idea? With ESP-IDF v4.3 is working fine.

Thanks,

Carles

nishanth-radja, 2021-12-15:
@carlessole Yes,This issue was recently introduced,fix will be available soon.

carlessole, 2021-12-15:
Thanks @nishanth-radja ,

So it will be fixed on next version 4.3.2?

Is there an estimation about when it will be available?

Thanks

AxelLin, 2021-12-21:
> Thanks @nishanth-radja ,
> 
> So it will be fixed on next version 4.3.2?
No, v4.3.2 does not include this fix.

carlessole, 2021-12-21:
Ok. So, although today I see that v4.3.2 has been released, it is useless that I try to recompile everything with that new version and test it, right?

Thanks @AxelLin

AxelLin, 2022-01-16:
> @carlessole Yes,This issue was recently introduced,fix will be available soon.

@nishanth-radja 
Any update for this regression? BTW, I'm wondering if v4.4 also needs the fix.

kapilkedawat, 2022-01-17:
Hi All, fix is available on master https://github.com/espressif/esp-idf/commit/14b949077d2a70df710375f1e5878d82d10bd03b and will be available to other branches in some time.

AxelLin, 2022-01-25:
> Hi All, fix is available on master [14b9490](https://github.com/espressif/esp-idf/commit/14b949077d2a70df710375f1e5878d82d10bd03b) and will be available to other branches in some time.

@kapilkedawat 
This is still broken in v4.3 now.
(There are several updates on v4.3 branch recently but does not include this fix.)

AxelLin, 2022-01-26:
Fix for v4.3: 92331a8c62ac0ca657c35529c979085d904bedfa

Alvin1Zhang, 2022-01-26:
Thanks @AxelLin for the notes, the fix on release/4.3 will be part of the next bugfix release of IDF v4.3. Thanks.

closed 2022-01-26 by Alvin1Zhang

AxelLin, 2022-01-26:
@Alvin1Zhang 
Just remind that v4.4 release does not include this fix.

Alvin1Zhang, 2022-01-26:
@AxelLin Thanks for the reminder, we are back porting the fix to release branches until 4.1, will sync more if we have any progress of the fixes on other branches. Thanks.