espressif/esp-idf #8088: ESP32S2 Internal Temperature sensor and WiFi in simultaneous we get one crash

brunohorta82, 2021-12-15:
When we use the internal temperature sensor and WiFi the system call watchdog and reboots, this is normal behavior?

Alvin1Zhang, 2021-12-16:
@brunohorta82 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

arex-ebee, 2022-01-26:
@Alvin1Zhang I'm hooking up into this issue as I observed the same behaviour at that moment when WiFi is enabled. I already filed an issue on esp32.com (https://www.esp32.com/viewtopic.php?p=90332) but there I was advised to try it on esp-idf github repo. So let me follow your request for the template:

## Environment

- Development Kit:      ESP32-S2-Wrover-Kit
- Kit version: v1
- Module or chip used:  ESP32-S2-WROVER
- IDF version: v4.4-beta1-48-gddc44956bf
- Build System:         idf.py
- Compiler version: xtensa-esp32s2-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: Yes (Eclipse 21-06)
- Power Supply:         USB

## Problem Description

I'm using the internal temperature sensor on ESP32-S2 as part of the overall system health monitoring. After initialization (``temp_sensor_set_config()`` and ``temp_sensor_start()``) there is a cyclic call to ``temp_sensor_read_celsius()``. This is not a big deal and works as expected.
Additionally I needed to put the WiFi part into operation with station mode. Having this done and calling ``esp_wifi_start()`` I observed a weird behavior of the whole system suddenly printing task watchdog timeouts within different tasks, not reacting fluently any more, etc. pp.

Analyzing the root cause brought me to insight that the call to ``temp_sensor_read_celsius()`` never returns anymore after starting WiFi. Instead it enters an endless busy-wait loop within ``temp_sensor_read_raw()`` polling for HW register bit ``SENS_SAR_TSENS_CTRL_REG.SENS_TSENS_READY`` to become 1, which never happens anymore. Comparing the whole register value against the one when it was working showed that, as soon as WiFi gets enable, the temperature sensor HW is obviously powered off as the bits ``SENS_TSENS_POWER_UP`` and ``SENS_TSENS_POWER_UP_FORCE`` drop to 0 (although they were 1 prior to starting WiFi).

I could reproduce this behaviour using the simple wifi station example from IDF (using release/v4.4 branch as well as master), only adding the cyclic temperature sensor readout.

### Expected Behavior

Temperature can be read out easily while WiFi station is active and connected

### Actual Behavior

Temperature readout call ``temp_sensor_read_celsius()`` never returns and enters an unpaced busy-wait loop influencing the whole system.

### Steps to reproduce

 - optionally create a copy of the "simple_station" WiFi example of IDF
 - replace the file station_example_main.c by the one attached
 - configure WiFi SSID and WiFi Password within Example Configuration in menuconfig
 - run the application
 - the example will initially print a few temperature values prior to connecting to the AP
 - before connecting, the readout is paused
 - when connection has finished the readout is resumed after some seconds: Device is now reachable via network (ping)
 - if resumed one can see Task watchdog prints and the device will probably not reachable anymore via WiFi (ping to IP address times out)

### Code to reproduce this issue

See modified [station_example_main.c](https://gist.github.com/arex-ebee/c7b4cd53f1f4fd26e70965d8f794d06d)

## Debug Logs

(see attached file for full log)

```
[...]

I (370) wifi station: Wait 5s before connecting WiFi
I (1370) wifi station: Temp: 31.2°C
I (2370) wifi station: Temp: 31.2°C
I (3370) wifi station: Temp: 31.2°C
I (4370) wifi station: Temp: 31.2°C
I (5370) wifi station: Temp: 31.2°C
I (5380) wifi station: Pause temperature polling
I (5380) wifi station: ESP_WIFI_MODE_STA

[...]

I (9080) wifi station: connected to ap SSID:******* password:********
I (9090) wifi station: Resume temperature polling

E (14360) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (14360) task_wdt:  - IDLE (CPU 0)
E (14360) task_wdt: Tasks currently running:
E (14360) task_wdt: CPU 0: temp_read
E (14360) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4008C7CB:0x3FFC62700x40024796:0x3FFC6290 0x400884FA:0x3FFD0890 0x40088579:0x3FFD08C0 0x4008692D:0x3FFD0900 0x4002C8A9:0x3FFD0940 
0x4008c7cb: task_wdt_isr at /[...]/esp-idf-4.4/components/esp_system/task_wdt.c:183 (discriminator 3)

0x40024796: _xt_lowint1 at /[...]/esp-idf-4.4/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x400884fa: temp_sensor_read_raw at /[...]/esp-idf-4.4/components/driver/esp32s2/rtc_tempsensor.c:140 (discriminator 2)

0x40088579: temp_sensor_read_celsius at /[...]/esp-idf-4.4/components/driver/esp32s2/rtc_tempsensor.c:177

0x4008692d: temp_sensor_test_task at /[...]/esp-idf-4.4/examples/wifi/getting_started/station/build/../main/station_example_main.c:177 (discriminator 1)

0x4002c8a9: vPortTaskWrapper at /[...]/esp-idf-4.4/components/freertos/port/xtensa/port.c:130

```

## Other items if possible
[station_example_main.log](https://github.com/espressif/esp-idf/files/7940000/station_example_main.log)

Maldus512, 2022-04-07:
I'm suffering from the exact same issue

elpamyelhsa, 2022-04-09:
Not just me then. I've been seeing this problem since last year but didn't have the time to trace it. Hopfully someone will have it fixed shortly.

NoNullptr, 2022-11-10:
I had the same issue and just found this bug report.

The issue is that WiFi resets these three registers

`SENS.sar_tctrl.tsens_power_up`
`SENS.sar_tctrl.tsens_power_up_force`
`SENS.sar_tctrl.tsens_xpd_force`

to zero after starting, instead of returning them to their initial value. I assume that the WiFi code reads the temperature to calibrate, but since the WiFi stack is closed, only someone with access can solve it.

The bug is triggered by `temperature_sensor_ll_get_raw_value`, which writes 1 to `SENS.sar_tctrl.tsens_dump_out` and then loops until `SENS.sar_tctrl.tsens_ready`, which is an endless loop in the case when WiFi turned off tsense in the meantime, which then triggers the WDT.

As a temporary workaround, I replaced `temperature_sensor_ll_get_raw_value` by a function that checks if tsens is still turned on while waiting for the temperature to be ready.

bitmandu, 2022-12-07:
Just following up that this problem persists in ESP-IDF v5.1-dev-2186-g454aeb3a48. I'm using an ESP-S2-SOLO-2 development board (chip revision v1.0).

If you combine the [station][1] and [temp_sensor][2] examples, there is one temperature measurement, and then the watchdog is triggered.

    I (8511) wifi station: Enable temperature sensor
    I (8521) wifi station: Read temperature
    I (8521) wifi station: Temperature value 25.41 ℃
    E (14511) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
    E (14511) task_wdt:  - IDLE (CPU 0)
    E (14511) task_wdt: Tasks currently running:
    E (14511) task_wdt: CPU 0: main
    E (14511) task_wdt: Print CPU 0 (current core) backtrace
    
    
    Backtrace: 0x40088EB7:0x3FFC7260 0x4008900E:0x3FFC7280 0x40024815:0x3FFC72A0 0x40088824:0x3FFCED30 0x400877D6:0x3FFCED60 0x400F1F81:0x3FFCEDB0 0x4002C905:0x3FFCEDE0
    0x40088eb7: task_wdt_timeout_handling at /home/kris/local/esp-idf/components/esp_system/task_wdt/task_wdt.c:461 (discriminator 3)
    ...

[1]:https://github.com/espressif/esp-idf/tree/master/examples/wifi/getting_started/station
[2]: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/temp_sensor

softhack007, 2022-12-20:
Any news on this topic?

It seems like the temp sensor is useless once that wifi is started. Are there any workarounds, like starting/stopping the sensors directly before and after a single read? I want to read the sensor every 30 seconds.

would the following work if I repeat this for each measurement?
```C++
temp_sensor_start();
temp_sensor_read_celsius(&chip_temp);
temp_sensor_stop();
```

Samdaaman, 2022-12-20:
Any update on this?

I have tried reinitialising the temperature sensor each time but no luck when WiFi is running. Will likely take the route others have suggested and just not use temperature entirely whilst WiFi is running...

KWolfe81, 2023-01-14:
Hoping for a fix on this too.  Using ESP-IDF 4.4.

NathanJPhillips, 2023-01-25:
@NoNullptr, if your workaround is giving you success, could you possibly share it?

RilabsAutomotive, 2023-02-17:
I observe the same issue. Log is below.

I think that the issue can be coming from the implementation of temperature_sensor_ll_get_raw_value()
[image]
This blocking while loop can be interrupted by OS and this register ready may be missed.

Therefore a possible solution is to enter critical section before getting the register value and then exit: For example:
[image]

Log:
E (208759) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (208759) task_wdt:  - IDLE (CPU 0)
E (208759) task_wdt: Tasks currently running:
E (208759) task_wdt: CPU 0: APPTask
E (208759) task_wdt: CPU 1: IDLE
E (208759) task_wdt: Aborting.
E (208759) task_wdt: Print CPU 0 (current core) backtrace

Backtrace: 0x4206ccbb:0x3fcb4bb0 0x420221d9:0x3fcb4be0 0x42022258:0x3fcb4c20 0x4202234c:0x3fcb4c50 0x42021947:0x3fcb4c80 0x42129e47:0x3fcb4cb0 0x420106a9:0x3fcb4cd0 0x403867ba:0x3fcb4cf0
0x4206ccbb: temperature_sensor_ll_get_raw_value at C:/Espressif/frameworks/esp-idf-v5.0/components/hal/esp32s3/include/hal/temperature_sensor_ll.h:80
 (inlined by) temperature_sensor_get_celsius at C:/Espressif/frameworks/esp-idf-v5.0/components/driver/temperature_sensor.c:187

NoNullptr, 2023-02-26:
@NathanJPhillips, sorry for taking long to respond. The observations of @RilabsAutomotive are correct and align with what I saw. Since the temperature sensor is accessed in multiple locations (temp_sensor and the WiFi stack, at least), it needs to be protected by a critical section.

However, in order for this to work, the code needs to guarantee in all locations that it also restores the register state after reading the sensor.

I just finished making a presentation/video about ESP’s clock and its drift. For this, I needed to sample the temperature sensor every 2 seconds, while the WiFi turns on and off every 10 minutes. In case you’re interested, I uploaded it to YouTube: https://youtu.be/fZAR8WTKiSg. Long story short, it turns out that WiFi does not restore the `tsens` registers to their previous state after using it – I tested it with IDF 4.4 before IDF 4.5 came out.

Therefore: `temperature_sensor_ll_get_raw_value` needs to also ensure that the temperature sensor is powered up, that is, that all the registers are set correctly. If the sensors do need to be powered up, the task then needs be delayed for the output of temperature sensor to stabilize. Critical sections should not delay or wait, so you need to repeatedly exit and enter the section until the state is how you need it to be. Since the core issue is the WiFi code (closed-source) and it needs to properly restore the state of the temperature sensor, I didn’t bother making it work with critical sections, but used this dirty looping fix instead:

```C
static inline uint32_t temperature_sensor_ll_get_raw_value(void)
{
        if (!SENS.sar_peri_clk_gate_conf.tsens_clk_en ||
            !SENS.sar_tctrl2.tsens_xpd_force ||
            !SENS.sar_tctrl.tsens_power_up_force ||
            !SENS.sar_tctrl.tsens_power_up ||
            !SENS.sar_tctrl.tsens_dump_out
        ) {
s:              SENS.sar_peri_clk_gate_conf.tsens_clk_en = true;
                SENS.sar_tctrl2.tsens_xpd_force = true;
                SENS.sar_tctrl.tsens_power_up_force = true;
                SENS.sar_tctrl.tsens_power_up = true;
                vTaskDelay(pdMS_TO_TICKS(10));
        }

        SENS.sar_tctrl.tsens_dump_out = 1;
        while (!SENS.sar_tctrl.tsens_ready) {
                if (!SENS.sar_peri_clk_gate_conf.tsens_clk_en ||
                    !SENS.sar_tctrl2.tsens_xpd_force ||
                    !SENS.sar_tctrl.tsens_power_up_force ||
                    !SENS.sar_tctrl.tsens_power_up ||
                    !SENS.sar_tctrl.tsens_dump_out
                ) {
                        goto s;
                }
        }
        SENS.sar_tctrl.tsens_dump_out = 0;
        return SENS.sar_tctrl.tsens_out;
}
```

It worked without any issues for six weeks of continuous sampling while the WiFi regularly turned on and off. If you need to make the temperature sensor work right now, you can replace the function in “esp-idf/components/hal/esp32XX/include/hal/temperature_sensor_ll.h”.

While this works, the quality of the code is low, but you cannot fundamentally improve it without access to the closed part of the firmware. Maybe IDF 4.5 fixed the issue in WiFi already, but someone would need to test it. Most people from espressif seem to have either terrible understanding of English or dubious coding skills (or both), so this might be the only solution for a while – this bug has been open for more than a year already.

franz-ms-muc, 2023-03-25:
@igrr can you have a look into this ? seems to be a orphaned issue. Thanks !

DCSBL, 2023-06-20:
It seems this problem also exist in the ESP32S3, at least in v5.0.2. The task watchdog triggers an abort because the temperature sensor code does not return.

For us, this only happens when the device tries to connect to wifi but it cannot connect (ssid does not exists). It seems that all other tasks are waiting to be dequeued. We pin all tasks to core 0.

https://github.com/espressif/esp-idf/blob/5181de8ac5ec5e18f04f634da8ce173b7ef5ab73/components/hal/esp32s3/include/hal/temperature_sensor_ll.h#L77-L84

```
==================== THREAD 1 (TCB: 0x3fcaa404, name: 'Tmr Svc') =====================
#0  0x420649fb in temperature_sensor_ll_get_raw_value () at ~/esp/esp-idf-v5.0.2/components/hal/esp32s3/include/hal/temperature_sensor_ll.h:80
#1  temperature_sensor_get_celsius (tsens=0x3c1a7700, out_celsius=0x3fcaa1f4) at ~/esp/esp-idf-v5.0.2/components/driver/temperature_sensor.c:187
#2  0x42016fc0 in operator() (timer=..., __closure=0x3fcdfd40) at ~/project/components/app/app_temperature.cpp:40
#3  std::__invoke_impl<void, AppTemperature::init(std::shared_ptr<I2CBus>)::<lambda(Timer&)>&, Timer&> (__f=...) at ~/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/11.2.0/bits/invoke.h:61
#4  std::__invoke_r<void, AppTemperature::init(std::shared_ptr<I2CBus>)::<lambda(Timer&)>&, Timer&> (__fn=...) at ~/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/11.2.0/bits/invoke.h:111
#5  std::_Function_handler<void(Timer&), AppTemperature::init(std::shared_ptr<I2CBus>)::<lambda(Timer&)> >::_M_invoke(const std::_Any_data &, Timer &) (__functor=..., __args#0=...) at ~/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/11.2.0/bits/std_function.h:291
#6  0x4205c8c4 in std::function<void (Timer&)>::operator()(Timer&) const (__args#0=..., this=0x3fcdfd40) at ~/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/11.2.0/bits/std_function.h:560
#7  Timer::_timer_callback (timerHandle=0x3fcdfd6c) at ~/project/components/system/shared/src_cpp/timer.cpp:17
#8  0x40383300 in prvProcessExpiredTimer (xTimeNow=<optimized out>, xNextExpireTime=23798) at ~/esp/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/timers.c:572
#9  prvProcessTimerOrBlockTask (xListWasEmpty=0, xNextExpireTime=23798) at ~/esp/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/timers.c:641
#10 prvTimerTask (pvParameters=<optimized out>) at ~/esp/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/timers.c:604
#11 0x40383668 in vPortTaskWrapper (pxCode=0x4038322c <prvTimerTask>, pvParameters=0x0) at ~/esp/esp-idf-v5.0.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154
```

ddomnik, 2023-09-22:
What's the state of this issue? I see a pull request has opened months ago.

JPSaturninoOliveira, 2023-10-04:
I'm using ESP32S3 with ESP-IDF v5.1.1 and problem continues to exist (only with WIFI running).

mcoracin, 2024-03-07:
Up.
I'm using ESP32S3 with ESP-IDF v5.1, and issue is there.

diplfranzhoepfinger, 2024-03-27:
now the PR is in conflict. 

how to go on ?

mythbuster5, 2024-03-28:
@JPSaturninoOliveira @mcoracin @diplfranzhoepfinger There is already a fix since 5.1.2. And I cannot produce this issue with https://gist.github.com/arex-ebee/c7b4cd53f1f4fd26e70965d8f794d06d anymore. So, if you guys are still confused by this issue. It's very welcome that you can show your simplest reproduce code, which will help us a lot to go ahead.

If there is no issue on 5,1,2, a reply is also welcome so that we can give a note and get this issue finished,

matthew-8925, 2024-04-02:
Seems to be working for me on v5.1.2+, with v4.4 I it would lock up right away and had to use the workaround for it to work at all.

FYI, the exact fixes in [tag/v5.1.2](https://github.com/espressif/esp-idf/releases/tag/v5.1.2) are

wifi binary blob update
[Fixed Wi-Fi not working with temperature sensor on ESP32-S2](https://github.com/espressif/esp-idf/commit/c1d8013330b7d54c58f5498135f2a1127b90678b)

temperature_sensor_* fixes
[Temperature Sensor: Fixed issue that if temperature sensor driver is disabled then phy can't work properly. ](https://github.com/espressif/esp-idf/commit/584250aa2d74d69918388f6f2669df815def8d3c)

mcoracin, 2024-04-10:
No more issue on v5.2.1 too, thanks !