espressif/esp-idf #8092: win7给esp32s2烧录固件时检测不到com口

CYCHEN97, 2021-12-16:
[image]
板子上没有串口芯片，直接使用usb进行下载时，检测不到com口

CYCHEN97, 2021-12-16:
@espressif-bot 
请问这样怎么操作才能让win7可以正常检测到下载端口？
[image]

CYCHEN97, 2021-12-16:
[image]

CYCHEN97, 2021-12-16:
@ESP-YJM
请问这个检测不到下载端口的怎么解决？
[image]
[image]

ESP-YJM, 2021-12-16:
你试下 https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers 或者 https://ftdichip.com/drivers/vcp-drivers/

CYCHEN97, 2021-12-16:
@ESP-YJM
安装了CP2102驱动还是检测不到，安装后面这个驱动重启后，电脑开机一分钟左右就死机了🤣

closed 2021-12-21 by CYCHEN97