espressif/esp-idf #8094: GPIO Interuupt Strange behavior

enthusisastSR, 2021-12-16:
## Environment

- Development Kit:     Custom ESP32S2
- Operating System:    Linux
- Using an IDE?:  Visual Studio Code

static xQueueHandle gpio_evt_queue = NULL;
 
static void IRAM_ATTR gpio_isr_handler(void *arg)
{
  uint32_t gpio_num = (uint32_t)arg;
  uint32_t val = gpio_get_level(gpio_num);
  xQueueSendFromISR(gpio_evt_queue, &val, NULL);
}

static void gpio_task(void *arg)
{
  uint32_t io_num;
  for (;;)
  {
    if (xQueueReceive(gpio_evt_queue, &io_num, portMAX_DELAY))
    {
      printf("GPIO[%d] intr, val: %d\n", io_num, gpio_get_level(io_num));
    }
  }
}

void app_main()
{
//zero-initialize the config structure.
    gpio_config_t io_conf;
    //SwitchA Interrupt
    io_conf.intr_type = GPIO_INTR_ANYEDGE;
    //bit mask of the pins, use GPIO4/5 here
    io_conf.pin_bit_mask = 1UL<<SwitchA;
    //set as input mode
    io_conf.mode = GPIO_MODE_INPUT;
    //enable pull-up mode
    io_conf.pull_up_en = 1;
    io_conf.pull_down_en = 0;
    gpio_config(&io_conf);

     //SwitchB Interrupt
    io_conf.intr_type = GPIO_INTR_ANYEDGE;
    //bit mask of the pins, use GPIO4/5 here
    io_conf.pin_bit_mask = 1UL<<SwitchB;
    //set as input mode
    io_conf.mode = GPIO_MODE_INPUT;
    //enable pull-up mode
    io_conf.pull_up_en = 1;
    io_conf.pull_down_en = 0;
    gpio_config(&io_conf);

    //change gpio interrupt type for one pin
    gpio_set_intr_type(SwitchA, GPIO_INTR_ANYEDGE);

    //change gpio interrupt type for one pin
    gpio_set_intr_type(SwitchB, GPIO_INTR_ANYEDGE);

    //create a queue to handle gpio event from isr
    gpio_evt_queue = xQueueCreate(10, sizeof(uint32_t));
    //start gpio task
    xTaskCreate(gpio_task, "gpio_task", 2048, NULL, 10, NULL);

    //install gpio isr service
    gpio_install_isr_service(ESP_INTR_FLAG_DEFAULT);
    //hook isr handler for specific gpio pin

    gpio_isr_handler_add(SwitchA, gpio_isr_handler, (void*) SwitchA);

    gpio_isr_handler_add(SwitchB, gpio_isr_handler, (void*) SwitchB);

  while (1)
  {
       vTaskDelay(1000 / portTICK_RATE_MS);
  }
}

nopnop2002, 2021-12-20:
```
 BaseType_t xQueueSendFromISR
           (
               QueueHandle_t xQueue,
               const void *pvItemToQueue,
               BaseType_t *pxHigherPriorityTaskWoken
           );
```

xQueueSendFromISR(gpio_evt_queue, &val, ＆xHigherPriorityTaskWoken);

nopnop2002, 2021-12-20:
xQueueReceive retrieves gpio level, not gpio number.

Therefore, the following printf is invalid.

```
printf("GPIO[%d] intr, val: %d\n", io_num, gpio_get_level(io_num));
```

closed 2025-03-13 by Alvin1Zhang