espressif/esp-idf #8095: hardware timer wont restart with no auto_reload

chegewara, 2021-12-16:
Hi,
i am having problems with restarting hardware timer which is init with `auto_reload=false`. No matter what i am trying to do in ISR callback, and/or when i am trying to call `timer_start();` again, the timer wont fire anymore.
My tests are performed https://github.com/espressif/esp-idf/tree/master/examples/peripherals/timer_group on master branch and esp32 and S3 boards.
With `auto_reload = true` example code works fine.

Its not entirely, true. The timer will reload and fire one more time with `auto_reload=false`, when i set (only 1 timer is running in this example):
```
    if (!info->auto_reload) {
        timer_counter_value += 130; // about 130 ticks or less
        timer_group_set_alarm_value_in_isr(info->timer_group, info->timer_idx, timer_counter_value);

        timer_group_clr_intr_status_in_isr(info->timer_group, info->timer_idx);    
        timer_group_enable_alarm_in_isr(info->timer_group, info->timer_idx);
        ets_printf("%lld\n", timer_counter_value);
    }
```
logs:
```
15000145
15000304
Timer Group without auto reload
Group[0], timer[0] alarm event
------- EVENT TIME --------
Counter: 0x0000000000e4e1cf
Time   : 3.00000300 s
-------- TASK TIME --------
Counter: 0x0000000000e557e5
Time   : 3.00604900 s
Timer Group without auto reload
Group[0], timer[0] alarm event
------- EVENT TIME --------
Counter: 0x0000000000e4e26e
Time   : 3.00003480 s
-------- TASK TIME --------
Counter: 0x0000000000e6df2e
Time   : 3.02608280 s
```
with 100 ticks it finally works:
```
15000115
15000274
15000415
15000556
15000696
15000837
15000978
...
eventually crash due to ets_printf() in ISR
```

Thanks

chegewara, 2021-12-16:
Even this code borrowed from unity test is not letting to fire timer more then 1 time:
```
static bool test_timer_group_isr_cb(void *arg)
{
    bool is_awoken = false;
    timer_info_t _info = {
        .timer_group = TIMER_GROUP_0,
        .timer_idx = TIMER_0
    };
    timer_info_t *info = &_info;
    const timer_group_t timer_group = info->timer_group;
    const timer_idx_t timer_idx = info->timer_idx;
    uint64_t timer_val;
    double time;
    uint64_t alarm_value;

    if (timer_group_get_auto_reload_in_isr(timer_group, timer_idx)) { // For autoreload mode, the counter value has been cleared
        timer_group_clr_intr_status_in_isr(timer_group, timer_idx);
        esp_rom_printf("1. This is TG%d timer[%d] reload-timer alarm!\n", timer_group, timer_idx);
        timer_get_counter_value(timer_group, timer_idx, &timer_val);
        timer_get_counter_time_sec(timer_group, timer_idx, &time);
    } else {
        timer_group_clr_intr_status_in_isr(timer_group, timer_idx);
        esp_rom_printf("2. This is TG%d timer[%d] count-up-timer alarm!\n", timer_group, timer_idx);
        timer_get_counter_value(timer_group, timer_idx, &timer_val);
        timer_get_counter_time_sec(timer_group, timer_idx, &time);
        timer_get_alarm_value(timer_group, timer_idx, &alarm_value);
        timer_set_counter_value(timer_group, timer_idx, 0);
    }

    return is_awoken;
}
```
```
2. This is TG0 timer[0] count-up-timer alarm!
```

suda-morris, 2021-12-17:
Yes, it's broken by a wrong fix of https://github.com/espressif/esp-idf/issues/7001
So the alarm is disabled by accident in https://github.com/espressif/esp-idf/commit/9a14cf083f2d7b4dc9841ad7d7bcb11eec1a8594
Will have another fix for that.

chegewara, 2021-12-17:
Thanks for confirming, because i thought i am dumb and cant use this API.

closed 2021-12-24 by espressif-bot

chegewara, 2021-12-24:
It is fixed, thanks

I have one more question:
i found that accurate time is when i set alarm with this value:
```
timer_group_set_alarm_value_in_isr(group, timer_id, seconds * TIMER_BASE_CLK / TIMER_DIVIDER  - 276 / config.divider);
```

Any thoughts?

AxelLin, 2022-01-01:
v4.3 also needs this fix.

AxelLin, 2022-01-25:
@suda-morris This is still broken in v4.3.