espressif/esp-idf #8096: No Serial Output over JTAG when using `nimble/blehr`

HarryWindsor, 2021-12-16:
## Environment

- Development Kit:      ESP32-C3-DevKitM-1
- Module or chip used:  ESP32-C3-MINI-1
- IDF version (run ``git describe --tags`` to find it):
    // v4.4-dev-3703-gddc44956bf
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

I am only able to see serial prints (via ESP_LOGx) when not using Bluetooth.
I have my USB-Cable connected as described in https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/usb-serial-jtag-console.html#hardware-requirements I can see all the  outputs when running `get-started/hello_world`, but with `nimble/blehr` I do not gen any serial output via the JTAG.
Both times I do see output via the cable connected via microusb.

### Expected Behavior

See ESP_LOGx output when Bluetooth is enabled.

### Actual Behavior

No ESP_LOGx output when Bluetooth is enabled.

### Steps to reproduce

1. `cp -r esp-idf/examples/get-started/hello_world .`
2.  `idf.py set-target esp32c3` & `idf.py -p /dev/cu.usbserial-14120 flash monitor`
3.   `screen /dev/cu.usbmodem14601 115200` (output is shown via JTAG and microUSB)
4. `cp -r esp-idf/examples/bluetooth/nimble/blehr .`
5. `idf.py set-target esp32c3` & `idf.py -p /dev/cu.usbserial-14120 flash monitor`
6. `screen /dev/cu.usbmodem14601 115200` (no outpout is shown via JTA, only via microUSB)

How do I enable serial output over JTAG when Bluetooth is enabled?

mythbuster5, 2021-12-17:
Sorry, a little question, do you switch on the config named `ESP_CONSOLE_USB_SERIAL_JTAG` rather than just `ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG` ?

HarryWindsor, 2021-12-17:
Thank you, this worked for me. I did not know I had to enable `ESP_CONSOLE_USB_SERIAL_JTAG` when using Bluetooth.

When using `menuconfig` the setting is in `Component config > ESP System Settings`.

closed 2021-12-17 by HarryWindsor