espressif/esp-idf #8097: Compilation error in mbedTLS component

zinke-ct-video, 2021-12-16:
## Environment

- IDF version: v5.0-dev-758-g07bfc09d0c (same bug occurs also on release/v4.4, commit f23dcd)
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System: Windows
- Environment type: ESP Command Prompt

## Problem Description

I have a project which uses the mbedtls component of the ESP-IDF. After updating my esp-idf repository, the project cannot be compiled anymore. Following error occurs during `idf.py build`

```
C:/Users/user258/esp/projects/devel/esp-idf/components/mbedtls/port/dynamic/esp_ssl_tls.c: In function 'ssl_handshake_init':
C:/Users/user258/esp/projects/devel/esp-idf/components/mbedtls/port/dynamic/esp_ssl_tls.c:190:9: error: implicit declaration of function 'ssl_set_timer' [-Werror=implicit-function-declaration]
         ssl_set_timer( ssl, 0 );
         ^~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
[244/1001] Building C object esp-idf/bootloader_support/CMakeFiles/__idf_bootloader_support.dir/src/esp_image_format.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### Expected Behavior

Should compile without errors.

### Actual Behavior

Cannot be compiled.

### Steps to reproduce

1. Update to master-branch, commit 07bfc0
2. Compile the project using `idf.py build`

The error occurs also with release/v4.4 commit f23dcd

aguaviva, 2021-12-16:
This bug has been open for a while, and the fix is not that difficult...

mahavirj, 2021-12-17:
@zinke-ct-video We are working on fix for this issue. As quick solution, you may try disabling `CONFIG_MBEDTLS_DYNAMIC_BUFFER` that should fix the build problem.

zinke-ct-video, 2021-12-17:
@mahavirj Thank you for the quick fix. I have to check if I can afford using static memory for mbedtls because our application is very memory constrained.

LJYSP, 2021-12-20:
@zinke-ct-video Do your project need open CONFIG_MBEDTLS_SSL_PROTO_DTLS? If not, Disable CONFIG_MBEDTLS_SSL_PROTO_DTLS and open CONFIG_MBEDTLS_DYNAMIC_BUFFER will also help. The two configuration items are not supported opening at the same time and we will add the limited condition later.

zinke-ct-video, 2021-12-20:
That's a good hint, thank you!

mahavirj, 2021-12-27:
@zinke-ct-video We have added required dependency in config options per comment from @LJYSP in https://github.com/espressif/esp-idf/commit/2b1402890de0ec26fb382ca8a19dc4a9f1ff1853. Closing the issue.

closed 2021-12-27 by mahavirj