espressif/esp-idf #8098: UART0 receive wrong data at light-sleep with external 32.768kHz XTAL

Rita858, 2021-12-16:
## Environment
- module used:  ESP32-WROOM-32D
- IDF version:           release/v4.3(e493a4c30e1b2f) 
- Build System:         [CMake|idf.py]
- Refer Demo:           BLE power save demo

## Problem
module works fine in BLE power save demo until I added uart events function on(referring to [uart_events.](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/uart/uart_events))
it would receive wrong data unexpected.
it would be alleviated when calling "gpio_hold_en" to keep specific pin enable. But the problem still not be solved.

[image]

So, I wonder if there are other solution can fix it?
Looking forward to your reply, thank you.

esp-lis, 2022-03-24:
When the system is in light sleep, the uart will be suspended (the clock is gated). In the latest esp-idf version ([c1b3049](https://github.com/espressif/esp-idf/commit/c1b30491ddedd6f572dc6324eb322b639b3de029)), an example of the light sleep uart wakeup has been added. The path is as follows:
`$IDF_PATH/examples/system/light_sleep/`

jack0c, 2022-04-14:
Another problem is, compared with 32.768 KHz XTAL, 115200 baud rate is too high. Suggest send valid data after wake up from light sleep(need more wake up bytes).

jack0c, 2022-05-17:
@Rita858 any update?

jack0c, 2022-06-15:
I will close this issue for no response. Feel free to reopen if necessary.

closed 2022-06-15 by jack0c