espressif/esp-idf #8108: WiFi-mesh in chain topology seems to heavily favour low layer counts leading to network instability

MisterScience5, 2021-12-17:
## Environment

- Development Kit:      -
- Kit version (for WroverKit/PicoKit/DevKitC): v3
- Module or chip used:  ESP32-WROOM-32U
- IDF version (run ``git describe --tags`` to find it):
    v4.2-rel-490-2-g530fff6
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     Windows
- (Windows only) environment type: PowerShell.
- Using an IDE?: Yes (VSCode with Espressif IDF addon)
- Power Supply:         Battery

## Problem Description

We are using the ESP32 in a street lamp controller. We are using mesh with WiFi Long Range mode

When using WiFi-mesh in chain topology, we are experiencing a problem with our product that the mesh nodes seem to prioritize a low layer count over the best signal, leading to instability in networks larger than 10-15 devices.

The only option for this as far as I can see is reducing the minimum RSSI for connection via esp_mesh_set_switch_parent_paras(), however our devices have varying signals, and reducing this too much could leave some nodes stranded.

The documentation on the functions esp_mesh_set_rssi_threshold() and esp_mesh_set_switch_parent_paras() could also be improved.

### Expected Behavior

In chain topology, the mesh handler should select the visible node with the best signal regardless of layer. Since the chain topology can have a very high number of layers, signal stability should be preferred over latency.

### Actual Behavior

The automatic mesh handler finds all visible nodes with a signal better than the configured cutoff, and seemingly connects to the one with the lowest layer.

MisterScience5, 2021-12-19:
Quick update, I have tried to enable esp_mesh_print_scan_result(true) to try and alleviate the problem by manipulating the RSSI limits with esp_mesh_set_rssi_threshold() and esp_mesh_set_switch_parent_paras(), however the scans never seem to return RSSI for any of the available nodes, as per this example:

`I (722741) mesh: [S1]ESPM_715EA0[node, layer(cap:0/0), chain(cap:198/2), assoc(cap:7/0), rssi:-
I (722742) mesh: [S2]ESPM_A79FBC[node, layer(cap:0/0), chain(cap:198/2), assoc(cap:7/0), rssi:-
I (722749) mesh: [S3]ESPM_F8C464[node, layer(cap:0/0), chain(cap:198/2), assoc(cap:7/0), rssi:-
I (722758) mesh: [S4]ESPM_A86718[root, layer(cap:0/0), chain(cap:199/1), assoc(cap:7/4), rssi:-
I (729982) mesh: [S1]ESPM_715EA0[node, layer(cap:0/0), chain(cap:198/2), assoc(cap:7/0), rssi:-
I (729982) mesh: [S2]ESPM_F8C464[node, layer(cap:0/0), chain(cap:198/2), assoc(cap:7/0), rssi:-
I (729989) mesh: [S3]ESPM_A79FBC[node, layer(cap:0/0), chain(cap:198/2), assoc(cap:7/0), rssi:-`

MisterScience5, 2022-01-04:
Another possible relevant note is that I am using Long Range mode for the WiFi - I am currently experimenting to see if this problem exists also on B/G/N connections.

jack0c, 2022-01-07:
@MisterScience5 What is the distance of the lamps?

MisterScience5, 2022-01-07:
It varies, but approx. 25-45m.

shenjun7, 2022-01-10:
After the non-root node successfully connects to the parent node, it will scan the current channel from time to time, and then re-select a parent node to determine whether it is better than the currently connected parent node. 
Judging better parent nodes is based on layer assoc and rssi. You can check whether it is reasonable to switch the parent according to the following log.

`mesh: 989[switch]choose candidate:1c:9d:c2:40:3e:69<layer:2, rssi:-7, assoc:0>, parent:1c:9d:c2:41:30:c1<layer:2, rssi:-31 assoc:6>`

MisterScience5, 2022-01-10:
My nodes never seem to reselect a host. I have experimented a bit with the mesh_switch_parent_t.cnx_rssi and mesh_switch_parent_t.duration_ms settings, but these don't seem to make much difference.

I think I have a better understanding of the underlying mechanism now though, my specific problem seems to be based on the fact that all the nodes in the large mesh turn on at the same time.

As the mesh connection spreads, each node will rapidly connect to the first node it sees that is in MESH_NODE mode, even though it has a poor connection to it. At this point, several nodes may be closer, but they are still in MESH_IDLE mode, so the node bypasses them leading to intermittent units in the mesh that have a poor connection.
I have made a temporary fix which seems to be working where the nodes are numbered, and each node waits for 10s for each node between it and the root before starting the mesh.

I believe the reason it does not reconnect to a better host is that all the better hosts will be higher layer because the mesh spreads from the root, and it still does not seem to consider that the RSSI is below the set threshhold.

MisterScience5, 2022-08-24:
I believe this can be closed, as I think it was caused by a misconfiguration. I had set the mesh channel to 0 for all devices, which seemed to have the side effect of not switching parent nodes based on signal strength. Shouldn't be a problem, as this is not really a valid configuration as I have since learned.. :)

closed 2022-08-25 by Alvin1Zhang