espressif/esp-idf #8109: Get timestamp on every esp-log

MaazSk, 2021-12-17:
## Environment

- Development Kit:      [ESP32-DevKit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it): v4.3
   
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
  
- Operating System:     [Linux]

- Using an IDE?: [VS Code]
- Power Supply:         [USB]

## Problem Description

I want timestamp logs on every log that ESP-prints even if the logs are from the esp-idf header files I need a timestamp on it is there a method to get timestamps on each and every logs.

Also, I am using SNTP in my code so I can get the time from there but I don't know how to add that timestamp to my esp-log.

I was reading about the following function available in `esp-log.h`
```

**
 * @brief Function which returns system timestamp to be used in log output
 *
 * This function is used in expansion of ESP_LOGx macros to print
 * the system time as "HH:MM:SS.sss". The system time is initialized to
 * 0 on startup, this can be set to the correct time with an SNTP sync,
 * or manually with standard POSIX time functions.
 *
 * Currently this will not get used in logging from binary blobs
 * (i.e WiFi & Bluetooth libraries), these will still print the RTOS tick time.
 *
 * @return timestamp, in "HH:MM:SS.sss"
 */
char* esp_log_system_timestamp(void);
```
I just need some way in which I can add timestamp to each and every log

Thanks in advance

KaeLL, 2021-12-17:
https://github.com/espressif/esp-idf/blob/07bfc09d0c8837b4d149e1151c57635a0c02755a/components/log/Kconfig#L95-L119

MaazSk, 2021-12-17:
Thank you @KaeLL. 

It's working now.

closed 2021-12-17 by MaazSk