espressif/esp-idf #8110: Cannot compile ESP BLE MESH with CPP/C++ for ESP32 WROOM

nitinbhosaleemb, 2021-12-17:
Compiler : esp-idf/Code/CMakeFiles/__idf_Code.dir/Main.cpp.obj
C:\Users\Nitin.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe

IDF Version: ESP-IDF v4.2.0

Logs :

 ^
../Code/Main.cpp:209:1: note: in expansion of macro 'ESP_BLE_MESH_MODEL_PUB_DEFINE'
ESP_BLE_MESH_MODEL_PUB_DEFINE(onoff_cli_pub, 2 + 1, MSG_ROLE);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../esp32_common/esp-idf/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:433:5: warning: missing initializer for member 'esp_ble_mesh_model_pub_t::timer' [-Wmissing-field-initializers]
}
^
../Code/Main.cpp:209:1: note: in expansion of macro 'ESP_BLE_MESH_MODEL_PUB_DEFINE'
ESP_BLE_MESH_MODEL_PUB_DEFINE(onoff_cli_pub, 2 + 1, MSG_ROLE);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../esp32_common/esp-idf/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:433:5: warning: missing initializer for member 'esp_ble_mesh_model_pub_t::dev_role' [-Wmissing-field-initializers]
}
^
../Code/Main.cpp:209:1: note: in expansion of macro 'ESP_BLE_MESH_MODEL_PUB_DEFINE'
ESP_BLE_MESH_MODEL_PUB_DEFINE(onoff_cli_pub, 2 + 1, MSG_ROLE);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../esp32_common/esp-idf/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:433:5: error: designator order for field 'esp_ble_mesh_model_pub_t::msg' does not match declaration order in 'esp_ble_mesh_model_pub_t'
}
^
../Code/Main.cpp:209:1: note: in expansion of macro 'ESP_BLE_MESH_MODEL_PUB_DEFINE'
ESP_BLE_MESH_MODEL_PUB_DEFINE(onoff_cli_pub, 2 + 1, MSG_ROLE);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../Code/Main.cpp:209:31: warning: 'onoff_cli_pub' defined but not used [-Wunused-variable]
ESP_BLE_MESH_MODEL_PUB_DEFINE(onoff_cli_pub, 2 + 1, MSG_ROLE);
^~~~~~~~~~~~~
../esp32_common/esp-idf/components/bt/esp_ble_mesh/api/esp_ble_mesh_defs.h:429:37: note: in definition of macro 'ESP_BLE_MESH_MODEL_PUB_DEFINE'
static esp_ble_mesh_model_pub_t _name = {

nitinbhosaleemb, 2021-12-21:
any resolution?

nitinbhosaleemb, 2021-12-21:
@Alvin1Zhang

InfiniteYuan, 2022-06-06:
@nitinbhosaleemb I'm going to fix it.

closed 2024-01-30 by espressif-bot