espressif/esp-idf #8111: [S3] IPC stack too low

chegewara, 2021-12-17:
Hi,
here is situation i am trying to test:
1. custom event loop
2. task sending events 
3. task is running `esp_event_loop_run`
4. very important: `tick rate = 1000` and `CPU freq = 240MHz` (in all??? other cases code works fine)
5. when i stop posting events, code works fine
6. second custom event loop created with a task, so no need to call `esp_event_loop_run`
7. when i disable second event loop, code works
8. when i disable logs in event loop handler, code works
9. few more combinations of parameters are giving different outputs, either crash or works
10. esp-idf: master branch
11. on old esp32 the same code seems to work fine

```
Rebooting...
�ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42004692
0x42004692: find_desc_for_int at /home/chegewara/esp/master/components/esp_hw_support/intr_alloc.c:131

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x1724
load:0x403b6000,len:0x8c8
load:0x403ba000,len:0x2ed8
entry 0x403b61c0
I (29) boot: ESP-IDF v5.0-dev-676-g5c33570524-dirty 2nd stage bootloader
I (29) boot: compile time 20:02:32
I (29) boot: chip revision: 0
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (42) boot.esp32s3: SPI Flash Size : 2MB
I (47) boot: Enabling RNG early entropy source...
W (52) bootloader_random: RNG for ESP32-S3 not currently supported
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=08130h ( 33072) map
I (111) esp_image: segment 1: paddr=00018158 vaddr=3fc90970 size=02718h ( 10008) load
I (116) esp_image: segment 2: paddr=0001a878 vaddr=40374000 size=057a0h ( 22432) load
I (127) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=17bc0h ( 97216) map
I (148) esp_image: segment 4: paddr=00037be8 vaddr=403797a0 size=071c8h ( 29128) load
I (155) esp_image: segment 5: paddr=0003edb8 vaddr=50000000 size=00010h (    16) load
I (160) boot: Loaded app from partition at offset 0x10000
I (160) boot: Disabling RNG early entropy source...
W (166) bootloader_random: RNG for ESP32-S3 not currently supported
I (184) cpu_start: Pro cpu up.
I (184) cpu_start: Starting app cpu, entry point is 0x403750ec
0x403750ec: call_start_cpu1 at /home/chegewara/esp/master/components/esp_system/port/cpu_start.c:164

I (162) cpu_start: App cpu up.
I (198) cpu_start: Pro cpu start user code
I (198) cpu_start: cpu freq: 240000000 Hz
I (198) cpu_start: Application information:
I (201) cpu_start: Project name:     events
I (206) cpu_start: App version:      5a0dba2-dirty
I (211) cpu_start: Compile time:     Dec 17 2021 20:02:28
I (217) cpu_start: ELF file SHA256:  cf57a8a58399ff66...
I (223) cpu_start: ESP-IDF:          v5.0-dev-676-g5c33570524-dirty
I (230) heap_init: Initializing. RAM available for dynamic allocation:
I (238) heap_init: At 3FC93AE0 len 0004C520 (305 KiB): D/IRAM
I (244) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (251) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (257) heap_init: At 600FE000 len 00002000 (8 KiB): RTCRAM
I (264) spi_flash: detected chip: generic
I (268) spi_flash: flash io: dio
W (272) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (285) sleep: Configure to isolate all GPIO pins in sleep state
I (292) sleep: Enable automatic switching of GPIO sleep configuration
I (299) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.

assert failed: esp_ipc_isr_init_cpu esp_ipc_isr.c:73 (uxTaskGetStackHighWaterMark(NULL) > 128)

Backtrace:0x40375a4a:0x3fcf12500x40378f09:0x3fcf1270 0x4037ee91:0x3fcf1290 0x42001e3a:0x3fcf13b0 0x4037bb91:0x3fcf13d0 
0x40375a4a: panic_abort at /home/chegewara/esp/master/components/esp_system/panic.c:403

0x40378f09: esp_system_abort at /home/chegewara/esp/master/components/esp_system/esp_system.c:123

0x4037ee91: __assert_func at /home/chegewara/esp/master/components/newlib/assert.c:85

0x42001e3a: esp_ipc_isr_init_cpu at /home/chegewara/esp/master/components/esp_system/port/arch/xtensa/esp_ipc_isr.c:73 (discriminator 1)

0x4037bb91: vPortTaskWrapper at /home/chegewara/esp/master/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:130
```

I understand it may be edge case, and i am not sure if it require to be fixed, but i thought i will report it.

Thanks

chegewara, 2021-12-17:
This fixed issue (increase stack by 10 bytes):
```
void esp_ipc_isr_init(void)
{
    for (unsigned i = 0; i < portNUM_PROCESSORS; ++i) {
        portBASE_TYPE res = xTaskCreatePinnedToCore(esp_ipc_isr_init_cpu, "ipc_isr_init", configMINIMAL_STACK_SIZE + 10, NULL, 5, NULL, i);
        assert(res == pdTRUE);
        (void)res;
    }
}
```

0xjakob, 2021-12-21:
@chegewara Thanks for the investigation, really appreciated! We are aware of this issue already. Most likely there are some configurations still not considered [when configuring the minimal stack size](https://github.com/espressif/esp-idf/blob/master/components/freertos/esp_additions/include/freertos/FreeRTOSConfig.h#L148-L184). Or the minimal stack size is wrong altogether. We are aware of this issue and working on it, possibly making it a Kconfig option at first. If you have any new insights, please let us know!

chegewara, 2021-12-21:
Im glad that espressif is working on it, and yes, Kconfig option seems to be good start.

Thanks

closed 2022-01-18 by espressif-bot

franz-ms-muc, 2022-02-10:
will this be integrated in Release 4.4 ? 

so a 4.4.1 ? 

Thanks !

Alvin1Zhang, 2022-02-17:
@franz-ms-muc Thanks for the report, the fix doesn't make into IDF v4.4 release, currently we are back porting this to release/4.4, will update once the fix on release/4.4 is available, thanks.

KonssnoK, 2022-02-18:
aha, got into this issue today (building from v4.4 of 16Feb22), will manually increase the stack of ipc_isr_init for the time being.
Please update when fix is into v4.4

Alvin1Zhang, 2022-04-22:
Thanks for reporting and sorry for slow turnaround. The fix is included in v4.4.1, https://github.com/espressif/esp-idf/releases/tag/v4.4.1. Thanks.