espressif/esp-idf #8114: `idf_monitor.py` does not search for a serial port

HarryWindsor, 2021-12-18:
## Environment

- Development Kit:      ESP32-C3-DevKitM-1
- Module or chip used:  ESP32-C3-MINI-1
- IDF version (run ``git describe --tags`` to find it):
    // v4.4-dev-3703-gddc44956bf
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     macOS
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

When executing monitor via cmake I always have to supply the exact port, when running `idf.py monitor` it can search for the serial port.
Running `cmake --target monitor -- -j 6` which runs `python ./esp-idf/tools/idf_monitor.py --target esp32c3 --revision 3 main.elf` fails because of `FileNotFoundError: [Errno 2] No such file or directory: '/dev/ttyUSB0'`.
Running `idpy.py monitor` automatically finds the correct serial port.

### Expected Behavior

`cmake --target monitor -- -j 6` should find the correct serial port.

### Actual Behavior

`cmake --target monitor -- -j 6` sets `/dev/ttyUSB0` as the default serial port, which is wrong

dobairoland, 2021-12-19:
Hi @HarryWindsor. Yes, you are right. `idf_monitor.py` doesn't support port detection and the default `/dev/ttyUSB0` is hard-coded there I think since the initial release. The port detection is done by `esptool` and is available only when `idf.py monitor` is invoked.

If you are using `cmake` directly then you can use the `ESPPORT` environment variable. If you are working with the same device and the port doesn't change often then you can even set this variable permanently for your environment. This variable should work for flashing and monitoring as well.

HarryWindsor, 2021-12-19:
Then the documentation is wrong. In CMake, it says `${SERIAL_TOOL} will search for a serial port.`.

https://github.com/espressif/esp-idf/blob/7c7fbb1fd0762fe20cce2ea6d870101b64af04ed/components/esptool_py/run_serial_tool.cmake#L32
https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/esptool_py/get_port_args.cmake#L6

I would prefer for it to automatically search for the right port, as it does with idf.py, as I live the dongle life and have to use a USB c hub and my port changes from time to time.

closed 2023-10-19 by tomassebestik