espressif/esp-idf #8115: Weird CSI output

ENB213-xiaoh, 2021-12-19:
Hi all
I'm having this kind of issue with my collected CSI values. See below:
[30 -32 1 0 -28 -9 -28 -9 -26 -8 -25 -8 -23 -5 -20 -3 -17 0 -15 3 -12 7 -11 12 -10 16 -10 21 -11 25 -13 28 -16 30 -18 30 -22 29 -24 26 -25 22 -25 18 -24 13 -20 10 -17 7 -13 6 -10 6 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 8 -6 7 -6 6 -6 5 -6 4 -7 4 -9 4 -10 4 -13 4 -14 4 -16 5 -17 6 -19 7 -19 7 -19 7 -19 6 -18 5 -16 3 -14 1 -13 -2 -11 -6 -10 -10 -8 -13 -8 -17 -7 -20 -7 -23 -8 ]

This is weird as it shows null sub-carriers are around the central subcarrier.

Normally, we should have CSI values like this
[101 -48 5 0 0 0 0 0 0 0 5 2 23 12 25 13 27 16 28 19 27 20 24 22 22 23 20 24 19 25 18 25 20 27 20 27 18 26 16 26 16 25 16 25 14 23 12 21 12 21 12 20 14 19 15 18 14 17 16 17 18 16 18 14 10 6 20 11 20 10 22 10 22 10 23 10 25 11 25 10 24 8 25 7 27 5 27 5 26 6 26 7 27 8 27 7 28 6 29 5 27 4 25 3 25 3 26 4 26 4 26 3 26 3 25 3 24 1 5 0 0 0 0 0 0 0 0 0 ]

Please kindly advise. I'm having this issue with my two laptops. Both are running Ubuntu 18.04.

jack0c, 2022-01-07:
@ENB213-xiaoh Do you mean the abnormal part is `6 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 8`?

ENB213-xiaoh, 2022-01-08:
@jack0c Hi. Jack. Thank you for your reply. That is not a problem anymore. I just have to shift the subcarrier index. I will close it.

closed 2022-01-08 by ENB213-xiaoh