espressif/esp-idf #8118: esp_now_set_pmk does not work

FanhuaCloud, 2021-12-19:
## Environment

- Development Kit:      Ai-thinker NODEMCU-32s
- Module or chip used:  ESP32-32s
- IDF version (run ``git describe --tags`` to find it):
    // v5.0-dev-797-g279c8aeb8a-dirty
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes vscode idf plugin
- Power Supply:         USB

## Problem Description

//Detailed problem description goes here.

### Expected Behavior
When pmk is set, the sender should not be able to send to the receiver without encryption
### Actual Behavior
The receiver can receive the message from the sender
### Steps to reproduce
Look the code，enable SEND_ESP is sender,and REC_ESP is receiver

### Code to reproduce this issue
https://gist.github.com/FanhuaCloud/834486721d45a312bdfd42b4e5e614fe

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

FanhuaCloud, 2021-12-19:
Could it be a problem with the latest branch? I'm not testing on the release branch

jack0c, 2022-01-07:
@FanhuaCloud  Do you mean sender have a pmk，but receiver didn't have one，and receiver can receive message from sender？

FanhuaCloud, 2022-01-07:
> @FanhuaCloud Do you mean sender have a pmk，but receiver didn't have one，and receiver can receive message from sender？

Yes, for this I used version 2.0: https://github.com/espressif/esp-now solved the problem

mhdong, 2022-01-12:
Hi @FanhuaCloud LMK of the paired device is used to encrypt the vendor-specific action frame with the CCMP method. If the LMK of the paired device is not set, the vendor-specific action frame will not be encrypted.  You should set encrypt to be true,

    esp_now_peer_info_t peer1 = {
        .ifidx = ESP_IF_WIFI_STA,
        .encrypt = true};

closed 2022-03-04 by Alvin1Zhang