espressif/esp-idf #8120: esp32c3 i2c will cause cpu IDLE

FanhuaCloud, 2021-12-21:
## Environment

- Development Kit:     none
- Module or chip used:  esp32-c3
- IDF version (run ``git describe --tags`` to find it):
    //  ESP-IDF v5.0-dev-797-g279c8aeb8a
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // riscv32-esp-elf-gcc-8.4.0.exe (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt
- Using an IDE?: Yes  vscode esp-idf
- Power Supply:         USB

## Problem Description
When using i2c, there is a chance that the cpu will be idle
I think it might be because the i2c code took too long
//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior
E (6907) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (6907) task_wdt:  - IDLE (CPU 0)
E (6907) task_wdt: Tasks currently running:
E (6907) task_wdt: CPU 0: main
### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
        i2c_cmd_handle_t cmd = i2c_cmd_link_create();
        i2c_master_start(cmd);
        i2c_master_write_byte(cmd, AHT_I2C_ADDRESS_GND << 1 | I2C_MASTER_READ, true);
        i2c_master_read(cmd, data, 6, I2C_MASTER_ACK);
        i2c_master_stop(cmd);
        i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000 / portTICK_PERIOD_MS);
        i2c_cmd_link_delete(cmd);

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

FanhuaCloud, 2021-12-21:
I look the same ploblem #7812

mythbuster5, 2021-12-21:
[i2c_patch_1210 .txt](https://github.com/espressif/esp-idf/files/7749481/i2c_patch_1210.txt)

A patch may help here, change the *.txt to *.patch and `git apply i2c_patch_1210.patch`. Thanks

o-marshmallow, 2021-12-22:
Hi @FanhuaCloud ,
I don't expected this to fix your issue but in your code, you stated:
```
i2c_master_write_byte(cmd, AHT_I2C_ADDRESS_GND << 1 | I2C_MASTER_READ, true);
```
it should be:
```
i2c_master_write_byte(cmd, AHT_I2C_ADDRESS_GND << 1 | I2C_MASTER_WRITE, true);
```
Moreover, in a more general note, we do provide [wrappers to ease I2C transfers](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#_CPPv426i2c_master_write_to_device10i2c_port_t7uint8_tPK7uint8_t6size_t10TickType_t) and avoid such mistakes, in your case, you could use:
```
i2c_master_read_from_device(I2C_NUM_0, AHT_I2C_ADDRESS_GND, data, 6, 1000 / portTICK_PERIOD_MS)
```

Regarding the issue, can you provide a way for us to reproduce this issue? Can you provide an sdkconfig? What is you FreeRTOS tick rate and task watchdog timeout period in your sdkconfig (`FREERTOS_HZ` and `ESP_TASK_WDT_TIMEOUT_S`) ?

o-marshmallow, 2021-12-22:
Hi @FanhuaCloud ,

Can you provide a way for us to reproduce this issue? Can you provide an sdkconfig? What is you FreeRTOS tick rate and task watchdog timeout period in your sdkconfig (`FREERTOS_HZ` and `ESP_TASK_WDT_TIMEOUT_S`) ?

In a more general note, we do provide [wrappers to ease I2C transfers](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#_CPPv426i2c_master_write_to_device10i2c_port_t7uint8_tPK7uint8_t6size_t10TickType_t), in your case, you could use:
```
i2c_master_read_from_device(I2C_NUM_0, AHT_I2C_ADDRESS_GND, data, 6, 1000 / portTICK_PERIOD_MS)
```

FanhuaCloud, 2021-12-22:
> Hi @FanhuaCloud ,
> 
> ~Can you provide a way for us to reproduce this issue? Can you provide an sdkconfig? What is you FreeRTOS tick rate and task watchdog timeout period in your sdkconfig (`FREERTOS_HZ` and `ESP_TASK_WDT_TIMEOUT_S`) ?~ Thank @mythbuster5 for providing a fix
> 
> In a more general note, we do provide [wrappers to ease I2C transfers](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#_CPPv426i2c_master_write_to_device10i2c_port_t7uint8_tPK7uint8_t6size_t10TickType_t), in your case, you could use:
> 
> ```
> i2c_master_read_from_device(I2C_NUM_0, AHT_I2C_ADDRESS_GND, data, 6, 1000 / portTICK_PERIOD_MS)
> ```

https://gist.github.com/FanhuaCloud/be12c7ad155f7a75221b99bd3e64ca62
This is my code, I am using aht10 to get the temperature.But when I use i2c to operate other devices, it is normal

AxelLin, 2022-02-26:
> [i2c_patch_1210 .txt](https://github.com/espressif/esp-idf/files/7749481/i2c_patch_1210.txt)
> 
> A patch may help here, change the *.txt to *.patch and `git apply i2c_patch_1210.patch`. Thanks

@mythbuster5 
The patch does not apply to current master (and v4.4) tree.

AxelLin, 2022-03-08:
Just wondering how is the status, @mythbuster5 @o-marshmallow

AxelLin, 2022-03-28:
> Hi @FanhuaCloud ,
> 
> Thank @mythbuster5 for providing a fix

@o-marshmallow @mythbuster5 
Caould you clarify what was fix? (Above patch does not apply)

mythbuster5, 2022-03-28:
[i2c_patch_0317.txt](https://github.com/espressif/esp-idf/files/8360849/i2c_patch_0317.txt)

@AxelLin  Sorry, I saw the message just now

AxelLin, 2022-03-28:
@mythbuster5 

I test it by disconnect the SCL from my i2c sensor on pupose to simulate the TIMEOUT case.
Without the patch, I always got "task_wdt: Task watchdog got triggered.".
With the path, I no longer hit watchdog issue.
Thanks.

AxelLin, 2022-04-06:
@o-marshmallow @mythbuster5 

The fix is not available in any branch, when would you merge it?
The original patch (https://github.com/espressif/esp-idf/issues/8120#issuecomment-998456510) was provided in last year,
no idea what are you waiting for.

o-marshmallow, 2022-04-06:
@AxelLin This fix is simply a workaround, the goal is to fix the root cause first

AxelLin, 2022-04-08:
> @AxelLin This fix is simply a workaround, the goal is to fix the root cause first

The workaround fix appears in various places:
https://github.com/espressif/esp-idf/issues/8358#issuecomment-1046733687
https://github.com/espressif/esp-idf/issues/8543#issuecomment-1074125489
The feedback is the patch does not apply, however, there is no follow up.
It's a mess now.

AxelLin, 2022-04-25:
> Just wondering how is the status, @mythbuster5 @o-marshmallow
> 
> I found: In i2c_master_cmd_begin(), If portBASE_TYPE evt_res = xQueueReceive(p_i2c->cmd_evt_queue, &evt, wait_time); timeout then it calls i2c_hw_fsm_reset -> i2c_master_clear_bus(i2c_num); << This call never return
> 
@o-marshmallow 
FYI: Comment out this line fixes the "Task watchdog got triggered" issue:
https://github.com/espressif/esp-idf/blob/master/components/hal/esp32c3/include/hal/i2c_ll.h#L811

Alvin1Zhang, 2022-05-16:
Thanks for reporting and sorry for slow turnaround, fix is available https://github.com/espressif/esp-idf/commit/0b00831703ec351c4c3b1040b004b358e4396067, feel free to reopen.

closed 2022-05-16 by Alvin1Zhang