espressif/esp-idf #8122: [ESP32-S3] [USB Host] [release/v4.4] usb_host_install returns ESP_ERR_NOT_FOUND

chipweinberger, 2021-12-21:
**ESP-IDF:** release v4.4 December 8th 2021 commit: e104dd7f27d2e73ab0e9b614dd7b9295099069bf
**board:**: esp32-s3 DevKitC w 2MB PSRam & 8MB flash

USB Host is failing to install, it cannot find a suitable interrupt. 

**Note: this same code works on ESP32-S2!**

```

I (18927) : usbh_install
I (18927) intr_alloc: get_available_int: try to find existing. Cpu: 0, Source: 38
I (18927) intr_alloc: get_free_int: start looking. Current cpu: 0
I (18927) intr_alloc: Int 0 reserved 1 level 1 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 1 reserved 1 level 1 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 2 reserved 0 level 1 LEVEL hasIsr 1
I (18927) intr_alloc: ....Unusable: already in (non-shared) use.
I (18927) intr_alloc: Int 3 reserved 0 level 1 LEVEL hasIsr 1
I (18927) intr_alloc: ....Unusable: already in (non-shared) use.
I (18927) intr_alloc: Int 4 reserved 1 level 1 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 5 reserved 1 level 1 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 6 reserved 0 level 1 EDGE hasIsr 0
I (18927) intr_alloc: ....Unusable: special-purpose int
I (18927) intr_alloc: Int 7 reserved 0 level 1 EDGE hasIsr 0
I (18927) intr_alloc: ....Unusable: special-purpose int
I (18927) intr_alloc: Int 8 reserved 1 level 1 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 9 reserved 0 level 1 LEVEL hasIsr 1
I (18927) intr_alloc: ....Unusable: already in (non-shared) use.
I (18927) intr_alloc: Int 10 reserved 0 level 1 EDGE hasIsr 0
I (18927) intr_alloc: ....Unusable: incompatible trigger type
I (18927) intr_alloc: Int 11 reserved 0 level 3 EDGE hasIsr 0
I (18927) intr_alloc: ....Unusable: special-purpose int
I (18927) intr_alloc: Int 12 reserved 0 level 1 LEVEL hasIsr 1
I (18927) intr_alloc: ....Unusable: already in (non-shared) use.
I (18927) intr_alloc: Int 13 reserved 0 level 1 LEVEL hasIsr 1
I (18927) intr_alloc: ....Unusable: already in (non-shared) use.
I (18927) intr_alloc: Int 14 reserved 1 level 7 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 15 reserved 0 level 3 EDGE hasIsr 0
I (18927) intr_alloc: ....Unusable: special-purpose int
I (18927) intr_alloc: Int 16 reserved 0 level 5 EDGE hasIsr 0
I (18927) intr_alloc: ....Unusable: special-purpose int
I (18927) intr_alloc: Int 17 reserved 0 level 1 LEVEL hasIsr 1
I (18927) intr_alloc: ....Unusable: already in (non-shared) use.
I (18927) intr_alloc: Int 18 reserved 0 level 1 LEVEL hasIsr 1
I (18927) intr_alloc: ....Unusable: already in (non-shared) use.
I (18927) intr_alloc: Int 19 reserved 0 level 2 LEVEL hasIsr 1
I (18927) intr_alloc: ....Unusable: incompatible level
I (18927) intr_alloc: Int 20 reserved 0 level 2 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: incompatible level
I (18927) intr_alloc: Int 21 reserved 0 level 2 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: incompatible level
I (18927) intr_alloc: Int 22 reserved 1 level 3 EDGE hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 23 reserved 0 level 3 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: incompatible level
I (18927) intr_alloc: Int 24 reserved 1 level 4 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 25 reserved 1 level 4 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 26 reserved 0 level 5 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: incompatible level
I (18927) intr_alloc: Int 27 reserved 1 level 3 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 28 reserved 0 level 4 EDGE hasIsr 0
I (18927) intr_alloc: ....Unusable: incompatible level
I (18927) intr_alloc: Int 29 reserved 0 level 3 EDGE hasIsr 0
I (18927) intr_alloc: ....Unusable: special-purpose int
I (18927) intr_alloc: Int 30 reserved 1 level 4 EDGE hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: Int 31 reserved 1 level 5 LEVEL hasIsr 0
I (18927) intr_alloc: ....Unusable: reserved
I (18927) intr_alloc: get_available_int: using int -1
hcd_install: esp_intr_alloc failed
E (19347) usbh_install: hcd_install failed
E (19347) USB HOST: usbh_install error
```

chipweinberger, 2021-12-21:
Note: Im using:

    usb_host_config_t hostConfig = {
        // Interrupt flags for the underlying
        // ISR used by the USB Host stack
        .intr_flags = ESP_INTR_FLAG_LEVEL1,
    };

Which I found in ESP example code.

chipweinberger, 2021-12-21:
The flags: 0b100000000010, correspond to:

#define ESP_INTR_FLAG_LEVEL1        (1<<1)  ///< Accept a Level 1 interrupt vector (lowest priority)
#define ESP_INTR_FLAG_INTRDISABLED  (1<<11) ///< Return with this interrupt disabled

Dazza0, 2021-12-21:
@chipweinberger Might be because we're actually running out of interrupts. Could you try `ESP_INTR_FLAG_LEVEL2`?

chipweinberger, 2021-12-21:
I tried it, and it does succeed. But then I hit:

```
I (3277) USB HOST: usb_host_install
I (3307) pd usb host service: [host task] started...
I (3307) pd usb host service: transfer alloc 0
I (3307) pd usb host service: transfer alloc 1
I (3317) pd usb host service: transfer alloc 2
I (3327) pd usb host service: transfer alloc 3
I (3327) pd usb host service: client register
I (3337) pd usb host service: waiting usb connection....
I (3337) pd usb host service: [client task] started...
E (3627) HUB: Short device descriptor transfer failed
```

which gets called in **usb_host_lib_handle_events**.

I dont have a USB device plugged in, so it should not be attempting to get a device descriptor.

My code works perfectly with a ESP32-S2.

Dazza0, 2021-12-21:
> I tried Level 2, and it does succeed! But then I hit:

Then it probably means that the failed installation isn't a bug. The ESP32-S3 is actually running out of level 1 interrupts. Why it's running out could depend on multiple things, namely what other drivers/SW components are taking up those level 1 interrupts

> My code works perfectly with a ESP32-S2.

As for the failed enum, it's a known issue. See https://github.com/espressif/esp-idf/issues/8061

chipweinberger, 2021-12-21:
Ya that makes sense. We could close this issue. 

I'll need to open a new one for this other bug I'm now hitting.

closed 2021-12-21 by chipweinberger

chipweinberger, 2021-12-21:
Also, @Dazza0 , 

I noticed you're the author of the new host API!

I also wrote a host API on top of HCD, might be of useful inspiration. 

https://github.com/chipweinberger/xesp-usbh