espressif/esp-idf #8123: Network stack not working even though GOT_IP received from Wi-Fi connection

MartinTJDK, 2021-12-21:
If ESP32 (SDK V4.3.1) is connected via an instable Wi-Fi connection, or Wi-Fi AP is turned on/off a lot of times, the network stack sometimes get stucked, and does not work.

When it happens, the system has received the "GOT_IP" event, but any calls to network stack (like resolving host name) fails.

If the Wi-Fi AP is turned off, and then turned on again - a new "GOT_IP" event is received, and then the network functionality might work again!

When network stack is stucked, and I test the Wi-Fi AP using another device, the Wi-Fi AP does work.
A reboot of ESP32 (without changing Wi-Fi AP) also fixes the connection issue.

It is a problem, that you cannot rely on the Wi-Fi connection status - because the ESP32 application needs to know, that something is wrong - in order to reboot or whatever.

Has anyone observed issues like this?

MartinTJDK, 2021-12-21:
During one of my tests, I get the following output on console:
`
I (1999161) wifi:bcn_timout,ap_probe_send_start
[00001997.186602] Wifi : [Debug] Received unhandled system event: base=1061162584 id=21
I (2001661) wifi:ap_probe_send over, resett wifi status to disassoc
I (2001661) wifi:state: run -> init (c800)
I (2001671) wifi:pm stop, total sleep time: 325700447 us / 390902122 us
I (2001671) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
W (2009241) wifi:m f probe req l=0

W (2009491) wifi:m f probe req l=0

[00002007.711650] APP : [Error] ... socket connect failed errno=118
W (2009731) wifi:m f probe req l=0

W (2009971) wifi:m f probe req l=0

W (2010211) wifi:m f probe req l=0
`

MartinTJDK, 2021-12-21:
I have tried to implement a simple "network watchdog" task, which uses:
getaddrinfo ->
socket ->
connect ->
freeaddrinfo ->
write -> close ->
read -> close

I believe it is using low-level "lwip" for communication, and it does work, while TLS connection does not work!

The TLS is "esp_tls", and I believe it is using mbedtls as implementation...

How can I find out why TLS connections fail to work?

MartinTJDK, 2021-12-21:
My suspicion is, that the TLS has a connection, which has not been closed correctly.

The lwip "dbg_lwip_tcp_pcb_show()" shows the following output, when a new TLS connection cannot be created:

I (118883) lwip: -------------active pcbs------------
I (118883) lwip: pcb=0x3f829aec next=0x0 cb_arg=0x3f8289b8
I (118893) lwip: local ip type=0 ip=5189a8c0
I (118893) lwip: remote ip type=0 ip=22d8b85d
I (118903) lwip: so_options=0, tos=0 ttl=255
I (118903) lwip: state=7
I (118913) lwip: prio=64
I (118913) lwip: local_port=49971, remote_port=80
I (118913) lwip: flags=10
I (118923) lwip: pooltmr=0 pollinterval=2, last_tmr=151 tmr=215 rtmer=-1
I (118933) lwip: recv_nxt=582433401 recv_wnd=5744 recv_ann_wnd=5583 recv_ann_right_edge=582438984
I (118943) lwip: mss=1436
I (118943) lwip: rttest=0 rtseq=-2109679825 sa=0 sv=3
I (118943) lwip: rto=3 nrtx=0
I (118953) lwip: dupacks=0 lastack=-2109679753
I (118953) lwip: cwnd=4451 ssthreash=5744
I (118963) lwip: snd_next=-2109679753 snd_wl1=582433400 snd_wl2=-2109679753
I (118973) lwip: snd_lbb=-2109679753 snd_wnd=65535 snd_wnd_max=65535
I (118973) lwip: snd_buf=5744 snd_queuelen=0
I (118983) lwip: unsent_oversize=0
I (118983) lwip: keep_idle=7200000 keep_intvl=75000 keep_cnt=9
I (118993) lwip: persist_cnt=0 persist_backoff=0
I (118993) lwip: keep_cnt_sent=0
I (119003) lwip: unsent segments:
I (119003) lwip: unacked segments:
I (119013) lwip: ooseq segments:
I (119013) lwip: refused data=0x0
I (119013) lwip: -------------bound pcbs-------------
I (119023) lwip: -------------tw     pcbs------------
I (119023) lwip: pcb=0x3f82a548 next=0x0 cb_arg=0x0
I (119033) lwip: local ip type=0 ip=389a8c0
I (119033) lwip: remote ip type=0 ip=64e15b28
I (119043) lwip: so_options=0, tos=0 ttl=255
I (119043) lwip: state=a
I (119053) lwip: prio=64
I (119053) lwip: local_port=49951, remote_port=443
I (119063) lwip: flags=30
I (119063) lwip: pooltmr=0 pollinterval=0, last_tmr=27 tmr=8 rtmer=-1
I (119073) lwip: recv_nxt=1427062343 recv_wnd=5744 recv_ann_wnd=5048 recv_ann_right_edge=1427067391
I (119083) lwip: mss=1436
I (119083) lwip: rttest=0 rtseq=-554752305 sa=1 sv=3
I (119093) lwip: rto=3 nrtx=0
I (119093) lwip: dupacks=0 lastack=-554752304
I (119093) lwip: cwnd=5348 ssthreash=5744
I (119103) lwip: snd_next=-554752304 snd_wl1=1427062342 snd_wl2=-554752304
I (119113) lwip: snd_lbb=-554752304 snd_wnd=63653 snd_wnd_max=64620
I (119113) lwip: snd_buf=5744 snd_queuelen=0
I (119123) lwip: unsent_oversize=0
I (119123) lwip: keep_idle=7200000 keep_intvl=75000 keep_cnt=9
I (119133) lwip: persist_cnt=0 persist_backoff=0
I (119133) lwip: keep_cnt_sent=0
I (119143) lwip: unsent segments:
I (119143) lwip: unacked segments:
I (119153) lwip: ooseq segments:
I (119153) lwip: refused data=0x0

MartinTJDK, 2021-12-21:
[test_https.zip](https://github.com/espressif/esp-idf/files/7755353/test_https.zip)

I have created a full example program, which eventually will cause network stack (or wifi) to get stucked.

The example program continously tries to fetch a https web page, and it logs all events from wifi/ip as well as memory consumption etc.

When the network stack is stucked, it looks like:

_**I (352248) APP: Perform HTTPS request
I (352248) esp-tls: esp_tls_init
I (352248) esp-tls: esp_tls_conn_new_sync
I (352248) esp-tls: esp_tls_low_level_conn
I (352248) esp-tls: esp_tcp_connect
I (352258) esp-tls: resolve_host_name
E (352258) esp-tls: couldn't get hostname for :www.espressif.com:
E (352268) esp-tls: Failed to open new connection
E (352268) TRANS_SSL: Failed to open a new connection
I (352278) esp-tls: esp_tls_conn_destroy
E (352278) HTTP_CLIENT: Connection failed, sock < 0
E (352288) APP: esp_http_client_perform
I (352288) esp-tls: esp_tls_conn_destroy
I (353288) APP:
********************************************************************
Running, isConnected=1, downloadSuccess=7, downloadError=72
********************************************************************
MemoryInfo:
MALLOC_CAP_8BIT:        4251189
MALLOC_CAP_32BIT:       4307125
MALLOC_CAP_INTERNAL:     266062
MALLOC_CAP_SPIRAM:      4041063
MALLOC_CAP_DMA:          210126
MALLOC_CAP_EXEC:         180544
********************************************************************
EventLog, ticks=35169
[0] event_ticks=24566, event_base=WIFI_EVENT, event_id=5
[1] event_ticks=24576, event_base=WIFI_EVENT, event_id=4
[2] event_ticks=24655, event_base=IP_EVENT, event_id=0
[3] event_ticks=22855, event_base=IP_EVENT, event_id=0
[4] event_ticks=23243, event_base=WIFI_EVENT, event_id=5
[5] event_ticks=23449, event_base=WIFI_EVENT, event_id=5
[6] event_ticks=23655, event_base=WIFI_EVENT, event_id=5
[7] event_ticks=23661, event_base=WIFI_EVENT, event_id=4
[8] event_ticks=24154, event_base=WIFI_EVENT, event_id=5
[9] event_ticks=24360, event_base=WIFI_EVENT, event_id=5
**********************************************************************_

The event log shows, that the last received events were:
[0] event_ticks=24566, event_base=WIFI_EVENT, event_id=5     (DISCONNECTED)
[1] event_ticks=24576, event_base=WIFI_EVENT, event_id=4     (CONNECTED)
[2] event_ticks=24655, event_base=IP_EVENT, event_id=0     (GOT_IP)

But the network stack does NOT allow a new request to be processed.
It fails already during hostname resolving.

To reproduce the error enable/disable a Wi-Fi AP a lot of times (perhaps 30 times), and at irregular intervals.
When enabling the Wi-Fi AP wait to see, if a https request succeeds or not...

I used the "Mobile hotspot" functionality in Windows. As written below, the same "Mobile hotspot" does work for other devices, when the problem exists.

MartinTJDK, 2021-12-21:
I have checked, if lwip has any active pcbs, when issue is active.
It does not:

I (699628) lwip: -------------active pcbs------------
I (699628) lwip: -------------bound pcbs-------------
I (699638) lwip: -------------tw     pcbs------------

BitProgress, 2021-12-23:
@espressif
Is it possible to debug/readout any debug information from Wi-Fi handling?
The ESP32 believes it is connected to a Wi-Fi AP, but the Wi-Fi hotspot does not recognize any connected devices.

xueyunfei998, 2021-12-27:
if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        esp_wifi_connect();
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        isConnected = false;
        ESP_LOGI(TAG,"disconnect from AP. Try connect to AP again");
        esp_wifi_connect();
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_CONNECTED) {
         isConnected = true;
        ESP_LOGI(TAG,"disconnect from AP. Try connect to AP again");
        esp_wifi_connect();
    } 
else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI(TAG, "got ip:" IPSTR, IP2STR(&event->ip_info.ip));
    }
}

hi @MartinTJDK 

This time should be added to the callback function. WIFI_EVENT_STA_CONNECTED

And in the event, set isConnected to true.

MartinTJDK, 2022-01-04:
@mahavirj Can someone from Espressif confirm, that a "esp_wifi_connect();" is required after the WIFI_EVENT_STA_CONNECTED event in order to ensure reconnect to Wi-Fi network - at any time.

I am able to enter a Wi-Fi connection state to an AP (after many disconnectes/connects), where network communication does not work, but the ESP Wi-Fi stack believes, that it is connected.

Turning Wi-Fi AP off for 1 minute, and starting Wi-Fi AP again causes ESP32 to work properly again.
When ESP32 does not work properly, other devices are able to use the Wi-Fi AP.

xueyunfei998, 2022-01-05:
hi @MartinTJDK 

If you want to reconnect wifi after WIFI is disconnected, you can call `esp_wifi_connect` to reconnect WIFI after the event of WIFI_EVENT_STA_DISCONNECTED.

After reconnecting WIFI, you need to close the previous network communication connection, such as closing the socket, restarting and opening a new socket to communicate.

jack0c, 2022-05-20:
@MartinTJDK Should not call esp_wifi_connect() when WIFI_EVENT_STA_CONNECTED event get. There is a mistake on @xueyunfei998 's code.
From your log `W (2009241) wifi:m f probe req l=0`, it means ESP32 is out of memory to send some specific packets. I believe there is something wrong on Wi-Fi. I suggest update to latest IDF v4.3 and try again.

xueyunfei998, 2022-05-23:
hi @MartinTJDK 

Sorry, there is a problem with the code provided above.

`WIFI_EVENT_STA_CONNECTED`:
This connection event will be generated if the function `esp_wifi_connect ()` is called and the return value ESP_OK is received, and the station has successfully connected to the target AP.

closed 2022-07-13 by Alvin1Zhang