espressif/esp-idf #8124: OpenOCD crash with esp32s3-builtin.cfg when receives DGB connection

yntodorov, 2021-12-21:
Hello Espressif team,

The hardware that i am using is **ESP32-S3-DevKitC-1**. I am also using the **master** branch for esp-idf.
Successfully did:
1. idf.py set-target esp32s3 
2. idf.py menuconfig
3. idf.py build
4. idf.py flash

Then based on the manual in _https://docs.espressif.com/_ I tried to used OpneOCD with GDB using the "ESP32-S3 USB Port":
The way I run the OpenOCD is: "**openocd.exe -f board\esp32s3-builtin.cfg -d**" and I get:
...
_Debug: 505 2472 command.c:143 script_debug(): command - nand nand init
Debug: 507 2477 tcl.c:498 handle_nand_init_command(): Initializing NAND devices...
Debug: 508 2482 command.c:143 script_debug(): command - pld pld init
Debug: 510 2488 pld.c:206 handle_pld_init_command(): Initializing PLDs...
Debug: 511 2492 gdb_server.c:3593 gdb_target_start(): starting gdb server for esp32s3.cpu0 on 3333
Info : 512 2496 server.c:311 add_service(): Listening on port 3333 for gdb connections_

Then I run the GDB and what I get from OpenOCD is:

_Info : 513 16807 server.c:100 add_connection(): accepting 'gdb' connection on tcp/3333
Debug: 514 16813 breakpoints.c:358 breakpoint_clear_target_internal(): Delete all breakpoints for target: esp32s3.cpu0
Debug: 515 16818 breakpoints.c:358 breakpoint_clear_target_internal(): Delete all breakpoints for target: esp32s3.cpu1
Debug: 516 16826 breakpoints.c:552 watchpoint_clear_target(): Delete all watchpoints for target: esp32s3.cpu0
Debug: 517 16833 FreeRTOS.c:1197 FreeRTOS_clean(): FreeRTOS_clean
Debug: 518 16836 FreeRTOS.c:708 FreeRTOS_update_threads(): FreeRTOS_update_threads
Warn : 519 16843 FreeRTOS.c:716 FreeRTOS_update_threads(): No symbols for FreeRTOS!
Debug: 520 16847 target.c:1638 target_call_event_callbacks(): target event 19 (gdb-attach) for core esp32s3.cpu0
Debug: 521 16863 target.c:4639 target_handle_event(): target(0): esp32s3.cpu0 (esp32s3) event: 19 (gdb-attach) action:
        $_TARGETNAME_0 xtensa smpbreak BreakIn BreakOut
        # necessary to auto-probe flash bank when GDB is connected
        halt

Debug: 522 16890 command.c:143 script_debug(): command - esp32s3.cpu0 esp32s3.cpu0 xtensa smpbreak BreakIn BreakOut
Debug: 524 16898 xtensa.c:726 xtensa_smpbreak_write(): esp32s3.cpu0: write smpbreak set=0x30000 clear=0x600000
Debug: 525 16908 xtensa.c:744 xtensa_smpbreak_set(): esp32s3.cpu0: set smpbreak=30000, state=1
Debug: 526 16914 xtensa.c:726 xtensa_smpbreak_write(): esp32s3.cpu1: write smpbreak set=0x30000 clear=0x600000
Debug: 527 16918 xtensa.c:744 xtensa_smpbreak_set(): esp32s3.cpu1: set smpbreak=30000, state=1
Debug: 528 16938 command.c:143 script_debug(): command - halt halt
Debug: 530 16945 target.c:3135 handle_halt_command(): -
Debug: 531 16958 xtensa.c:1133 xtensa_halt(): xtensa_halt, target: esp32s3.cpu0
Debug: 532 16971 xtensa.c:1145 xtensa_halt(): esp32s3.cpu0: Core status 0x80000000
Debug: 533 16978 xtensa.c:890 xtensa_fetch_all_regs(): esp32s3.cpu0: start
Debug: 534 16989 xtensa.c:788 xtensa_core_status_check(): esp32s3.cpu0: DSR (80318411)
Debug: 535 17002 xtensa.c:788 xtensa_core_status_check(): esp32s3.cpu0: DSR (80318411)
Debug: 536 17005 esp32s3.c:295 esp32s3_fetch_user_regs(): esp32s3.cpu0: user regs fetching is not implememnted!
Debug: 537 17011 xtensa.c:1858 xtensa_poll(): esp32s3.cpu0: Target halted, pc=0x42017E22, debug_reason=00000000, oldstate=00000001
Debug: 538 17029 xtensa.c:1863 xtensa_poll(): esp32s3.cpu0: Halt reason=0x00000020, exc_cause=4, dsr=0x80318411
Info : 539 17036 xtensa.c:1866 xtensa_poll(): esp32s3.cpu0: Target halted, PC=0x42017E22, debug_reason=00000000
Debug: 540 17041 esp_xtensa_smp.c:257 esp_xtensa_smp_update_halt_gdb(): GDB target 'esp32s3.cpu0'
Info : 541 17045 esp_xtensa_smp.c:262 esp_xtensa_smp_update_halt_gdb(): Set GDB target to 'esp32s3.cpu0'
Debug: 542 17052 esp_xtensa_smp.c:271 esp_xtensa_smp_update_halt_gdb(): Check target 'esp32s3.cpu0'
Debug: 543 17055 esp_xtensa_smp.c:271 esp_xtensa_smp_update_halt_gdb(): Check target 'esp32s3.cpu1'
Debug: 544 17061 esp_xtensa_smp.c:285 esp_xtensa_smp_update_halt_gdb(): Poll target 'esp32s3.cpu1'
Debug: 545 17068 xtensa.c:890 xtensa_fetch_all_regs(): esp32s3.cpu1: start
Debug: 546 17079 xtensa.c:788 xtensa_core_status_check(): esp32s3.cpu1: DSR (80108411)
Debug: 547 17096 xtensa.c:788 xtensa_core_status_check(): esp32s3.cpu1: DSR (80108411)
Debug: 548 17100 esp32s3.c:295 esp32s3_fetch_user_regs(): esp32s3.cpu1: user regs fetching is not implememnted!
Debug: 549 17106 xtensa.c:1858 xtensa_poll(): esp32s3.cpu1: Target halted, pc=0x42017E22, debug_reason=00000000, oldstate=00000001
Debug: 550 17112 xtensa.c:1863 xtensa_poll(): esp32s3.cpu1: Halt reason=0x00000020, exc_cause=4, dsr=0x80108411
Info : 551 17119 xtensa.c:1866 xtensa_poll(): esp32s3.cpu1: Target halted, PC=0x42017E22, debug_reason=00000000
Debug: 552 17125 target.c:1638 target_call_event_callbacks(): target event 0 (gdb-halt) for core esp32s3.cpu1_
...
_Debug: 937 19618 target.c:1891 alloc_working_area_try_do(): allocated new working area of 9212 bytes at address 0x403b0000
Warn : 938 20128 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 192 out of 768 bytes.
Warn : 939 20635 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 768 bytes.
Debug: 940 20651 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65
Warn : 941 21162 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 832 bytes.
Debug: 942 21175 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65
[... 38 lines cut ...]
Warn : 981 31690 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 2112 bytes.
Debug: 982 31700 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65
Warn : 983 32213 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 2176 bytes.
Debug: 984 32228 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65
Warn : 985 32739 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 2240 bytes.
Debug: 986 32751 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65
Warn : 987 33263 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 2304 bytes.
Debug: 988 33275 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65
Warn : 989 33786 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 2368 bytes.
Debug: 990 33798 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65
Warn : 991 34309 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 2432 bytes.
Debug: 992 34322 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65
Warn : 993 34833 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 2496 bytes.
Debug: 994 34846 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65
Warn : 995 35358 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 2560 bytes.
Debug: 996 35373 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65
Warn : 997 35384 esp_usb_jtag.c:445 esp_usb_jtag_send_buf(): esp_usb_jtag: usb sent only 0 out of 2624 bytes.
Debug: 998 35396 libusb1_common.c:120 jtag_libusb_find_device(): USB dev found 303a:1001 @ 1:3-65_

At that point the OpenOCD exits and the board stays stuck. The GDB exits with error due to not successful connection with OpenOCD.

Anyone had similar problem and any idea how could I solv it ?

Best Regards,
Yordan Todorov

mishaturnbull, 2021-12-21:
Having a similar issue, I think, with the same hardware -- here's the output of the `openocd -f board/esp32s3-builtin.cfg`:

```
>>> openocd -f board/esp32s3-builtin.cfg
Open On-Chip Debugger  v0.10.0-esp32-20211111 (2021-11-10-21:40)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Warn : Transport "jtag" was already selected
WARNING: ESP32-S3 target may experience problems when accessing flash via builtin USB-JTAG adapter!
WARNING: It is recommended to disable flash support by running OpenOCD with `-c 'set ESP_FLASH_SIZE 0'` or use external JTAG adapter.
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : esp32s3.cpu0: Debug controller was reset.
Info : esp32s3.cpu0: Core was reset.
Info : Listening on port 3333 for gdb connections
// <------------------------------------------------------------------------   GDB started here!
Warn : esp_usb_jtag: usb sent only 0 out of 31 bytes.
Error: libusb_open() failed with LIBUSB_ERROR_NO_DEVICE
Error: esp_usb_jtag: device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: missing data from bitq interface
Warn : esp_usb_jtag: usb sent only 0 out of 50 bytes.
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Error: missing data from bitq interface
Error: Failed to exec JTAG queue!
Error: Failed to read debug stubs address location (-104)!
Polling target esp32s3.cpu0 failed, trying to reexamine
Info : esp32s3.cpu0: Debug controller was reset.
Info : esp32s3.cpu0: Core was reset.
Info : esp32s3.cpu0: Detected debug stubs @ 3fc966a8
Info : accepting 'gdb' connection on tcp/3333
Warn : No symbols for FreeRTOS!
Info : esp32s3.cpu0: Target halted, PC=0x4202D6C6, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Warn : Not full dbg stub table 1 of 3
Info : esp32s3.cpu0: Target halted, PC=0x403B2412, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c030020, 49 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 186 KB
Warn : esp_usb_jtag: usb sent only 192 out of 768 bytes.
Warn : esp_usb_jtag: usb sent only 0 out of 768 bytes.
Error: esp32s3.cpu0: DSR (8000CC13) indicates DIR instruction generated an exception!
Warn : esp32s3.cpu0: Failed reading 9212 bytes at address 0x403b0000
Error: no working area available, can't alloc space for stub code!
Error: Failed to get flash size!
Error: Failed to probe flash, size 0 KB
Error: auto_probe failed
Error: Connect failed. Consider setting up a gdb-attach event for the target to prepare target for GDB connect, or use 'gdb_memory_map disable'.
Error: attempted 'gdb' connection rejected
^Cshutdown command invoked
Info : esp32s3.cpu0: Restore debug stubs address 3fc966a8
Info : esp32s3.cpu1: Restore debug stubs address 3fc966a8
```

and the output of GDB as I'm trying to run it:
```
>> idf.py gdb
Executing action: gdb
GNU gdb (crosstool-NG esp-2021r2) 9.2.90.20200913-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32s3-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
/mnt/shared/Projects/RemoteID/Repo/air/code/esp32/build/gdbinit:2: Error in sourced command file:
Remote communication error.  Target disconnected.: Connection reset by peer.
(gdb) q
```

Has been happening for several months now, although I haven't been able to pinpoint the issue.

yntodorov, 2021-12-22:
Hello, 

With new version of openOCD v0.11.0-esp32-20211220 , debugging with GDB is working properly.

Link to the new version of openOCD: https://github.com/espressif/openocd-esp32/releases/

The ticke could be closed now.

Best Regards,
Yordan Todorov

closed 2021-12-22 by yntodorov

mishaturnbull, 2021-12-23:
Yup, new openOCD fixes it for me!  @yntodorov , do you have any estimate for how long before this version of openOCD is bundled with a new ESP-IDF release?  It was a bit of a process to figure out how to get the newer openOCD version swapped out for the current one in ESP-IDF.

Thanks!!!