espressif/esp-idf #8126: esp idf Ethernet basic Example with PHY8061 - Not Getting IP on DHCP

nikunjbambhroliya, 2021-12-21:
Working on Esp32 Ethernet Basic Example and esp idf Configure with PHY80xx Module and PHY Attached With Default EMAC Default GPIO's of ESP32.

PHY Address Set to 0(Broadcast).
Not Getting IP on DHCP as well as Also Try to static ip but not getting ping.
getting Ethernet link up and ethernet link down status during attachment and detachment of lan cable on rj45 connector.

Please See Below Debug Log and suggest solution to start DHCP.
-----------------------
Debug log :
-----------------------
hcp_fine_tmr(): request timeout
dhcp_timeout()
dhcp_timeout(): restarting discovery
dhcp_discover()
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(98f7137b)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
ip4_output_if: en1
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |       336     | (v, hl, tos, len)
+-------------------------------+
|        7      |000|       0   | (id, flags, offset)
+-------------------------------+
|  255  |   17  |    0xba96     | (ttl, proto, chksum)
+-------------------------------+
|    0  |    0  |    0  |    0  | (src)
+-------------------------------+
|  255  |  255  |  255  |  255  | (dest)
+-------------------------------+
ip4_output_if: call netif->output()
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 15000 msecs
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
d
hcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6760
load:0x40078000,len:14808
load:0x40080400,len:3792
entry 0x40080694
I (27) boot: ESP-IDF v5.0-dev-810-gb886dc6998-dirty 2nd stage bootloader
I (27) boot: compile time 12:31:28
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
[... 253 lines cut ...]
dhcp_discover(): dhcp state is DISCOVER
dhcp_select: could not allocate DHCP request
transaction id xid(17556994)
dhcp_discover: making request
dhcp_discover: sendto(DISCOVER, IP_ADDR_BROADCAST, LWIP_IANA_PORT_DHCP_SERVER)
ip4_output_if: en1
IP header:
+-------------------------------+
| 4 | 5 |  0x00 |       336     | (v, hl, tos, len)
+-------------------------------+
|        3      |000|       0   | (id, flags, offset)
+-------------------------------+
|  255  |   17  |    0xba9a     | (ttl, proto, chksum)
+-------------------------------+
|    0  |    0  |    0  |    0  | (src)
+-------------------------------+
|  255  |  255  |  255  |  255  | (dest)
+-------------------------------+
ip4_output_if: call netif->output()
dhcp_discover: deleting()ing
dhcp_discover: SELECTING
dhcp_discover(): set request timeout 4000 msecs
dhcp_coarse_tmr()
dhcp_coarse_tmr()
dhcp_coarse_tmr()

suda-morris, 2021-12-23:
Seems like the data panel is not working properly. Which GPIO is used by the RMII clock? This clock is important as it's used to synch MAC and PHY. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_eth.html#configure-mac-and-phy

nikunjbambhroliya, 2021-12-27:
Thanks @suda-morris 

I have Go Through RMII Clock Setting but it seems configuration is properly done.

Please Find Attached screenshot of Example Configuration that Implemented using menuconfig
[image]

Please Find Attached Screenshot Ethernet Configuration inside Component Config
[image]

In our Hardware Refclk pin connected with Esp32 GPIO0 with External Pullup.

please find attached kconfig file.
[kconfig file.txt](https://github.com/espressif/esp-idf/files/7778657/kconfig.file.txt)

kostaond, 2022-01-07:
@nikunjbambhroliya, do see any traffic between your ESP32 board and your host when you capture it by Wireshark (ESP should send DHCP Discover at least)?

It seems you use custom HW, please double check that all RMII connections are correctly routed and connected (Data, CLK, etc.). For example, I was recently bringing up custom Ethernet board and it worked only in one direction. It turned out that one of RMII TX data path zero resistors felt out during manufacturing/shipping processes...

nikunjbambhroliya, 2022-01-08:
Thanks @kostaond 
There is External Crystal Clock Issue. Now It's Working.

closed 2022-01-08 by nikunjbambhroliya

kostaond, 2022-01-08:
@nikunjbambhroliya I'm happy to hear you found the issue and it's work!