espressif/esp-idf #8127: FTM session to ASAP-capable FTMR crashes

crackwitz, 2021-12-21:
I have two ESP32-S2, both running in APSTA mode and initiating FTM sessions to each other. That works fine.

However, if I get them to FTMI to _another_ AP (linksys velop, FTMR capable but not advertising it, **ASAP-capable**), the wifi blob crashes. An immediately preceding FTM session to the other S2 succeeds.

I initialize the `wifi_ftm_initiator_cfg_t` structure as `{.frm_count = 16, .burst_period = 2 }`... which is reflected in the successful sessions between both S2s, but **not** in the session to this other AP. That info appears to be reset/cleared/defaulted to something else:

```
W (8255) wifi:Starting FTM session with e8:9f:80:e5:aa:1b
W (8255) wifi:Mode: ASAP, Bursts: 1, FTM's per burst: 8, Burst Period: 0mSec, Burst Duration: 0uSec
E (8265) wifi:No valid FTM Measurements found!
E (13265) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (13265) task_wdt:  - IDLE (CPU 0)
E (13265) task_wdt: Tasks currently running:
E (13265) task_wdt: CPU 0: wifi
E (13265) task_wdt: Aborting.

abort() was called at PC 0x40097040 on core 0

Backtrace:0x40023c3f:0x3ffcd2600x4002ecdd:0x3ffcd290 0x40037245:0x3ffcd2c0 0x40097040:0x3ffcd340 0x40025371:0x3ffcd370 0x400decf6:0x3ffe0f40 0x400dd4ce:0x3ffe0f60 0x4003a60b:0x3ffe0f90 0x4003aad2:0x3ffe0fb0 0x4003ac35:0x3ffe0ff0 0x40038bd2:0x3ffe1030 0x40032526:0x3ffe1060
```

A session between S2s looks like this (both sit beside each other, so the RTT is to be expected):
```
W (14194) wifi:Starting FTM session with 7c:df:a1:40:cc:3d in 0.015 Sec
W (14194) wifi:Mode: non-ASAP, Bursts: 2, FTM's per burst: 8, Burst Period: 200mSec, Burst Duration: 32000uSec
W (14444) wifi:FTM session ends with 7 valid readings out of 15, Avg raw RTT: 0.0 nSec, Avg RSSI: -41
```

## Environment

- Development Kit:      ESP32-S2-DevKitM (v1 I guess)
- Module or chip used:  ESP32-S2-MINI-1
- IDF version: `v5.0-dev-621-g4143a3f572` (from 2021-12-08, I'll retry this with a fresh copy in a moment)
- Build System:         idf.py / vscode extension
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): `xtensa-esp32-elf-gcc.exe (crosstool-NG esp-2021r2) 8.4.0`

nachiketkukade, 2021-12-30:
@crackwitz , thanks for bringing this issue to our attention.
First a question on the APSTA mode, are you seeing the same failure with another ESP32S2?
Coming back to the issue, there are 3 different ones at play here -
1. Watchdog timeout (crash you mentioned)
2. Consistent failure in ASAP mode
3. Intermittent failures in non-ASAP mode
The first 2 issues are being addressed with a fix. But the 3rd one cannot be fixed entirely from our side. The reason being that the Linksys Velop behaves erroneously and sends multiple frames with no measurement in every iteration. It also terminates the session abruptly sometimes. This is why the 'No measurements' error keeps coming. This issue is still present in ASAP mode but to a lesser extent. I've tested an Mi11X Pro Smartphone with the Velop AP using the RTT app which also uses ASAP mode, it too reports around 30% failure rate. Once the fix for the first 2 issues is merged, do let us know your observations on the failures in ASAP mode with this AP. Thanks.

closed 2022-01-04 by espressif-bot

crackwitz, 2022-01-05:
Thank you! I'll try the master branch right now.

> First a question on the APSTA mode, are you seeing the same failure with another ESP32S2?

I have several ESP32-S2-DevKitM and they behave interchangeably, when talking with the linksys velop. Among their own, all is fine.

I fully understand that Espressif isn't responsible for working around malfunctions in third party equipment like Linksys Velop. That device doesn't even advertise FTM. I don't expect it to work. What I am doing here could at best be considered "fuzz testing" or a very limited [plugfest](https://en.wikipedia.org/wiki/Plugtest). Intermittent "failures" of FTM sessions are perfectly acceptable to me, so long as the ESP side can't be crashed by a misbehaving FTMR on the other side :)

AndreiKorchunov, 2023-01-17:
Board 
esp32 c3 devkit-02 both 
ESP-ROM:esp32c3-api1-20210207 Build:Feb  7 2021
Hardware Configuration
nothing special
Arduino IDE 1.8.19 (Windows Store 1.8.57.0)
Library esp32 v 2.0.6 - softAp not work by standart example for FTM. The access point does not connect to the second esp 32, nor the Samsung Note. There are no errors in any esp32.
Library esp32 v 2.0.3 - softAp work by standart example for FTM. 
Intermittent crashes in ISAP mode FTM Error: CONF_REJECTED

nachiketkukade, 2023-03-17:
@AndreiKorchunov , please create a new issue for this if you haven't already and share the logs