espressif/esp-idf #8128: Know when there is no internet connection available on WiFi

MaazSk, 2021-12-21:
**Is your feature request related to a problem? Please describe.**

I am creating an application in which my device communicates with the server to exchange data and it is connected to the internet through a WiFi there are times when the internet services are down and at that time I don't want to share or exchange data with the server as there is no internet so can we have any feature which would help us know when the WiFI internet is down. So that we won't to tasks related to the wifi

**Describe alternatives you've considered**
The alternatives that I have considered are as below
- Using ping to know every 1 minute if the response is failure means no internet
- Using of DNS resolver if able to resolve DNS to IP means the internet is present or vice versa

**Additional context**

There was a discussion on this topic going on [here](https://esp32.com/viewtopic.php?f=13&t=5006). But nothing useful outcomes were there.

Thanks in advance

KaeLL, 2021-12-21:
I guess you could do like [Windows](https://superuser.com/a/277955/945677) or [nm](https://man.archlinux.org/man/NetworkManager.conf.5#CONNECTIVITY_SECTION) and perform a request periodically to a known server that returns a known response indicating connectivity or lack thereof.

LuggerMan, 2021-12-22:
@MaazSk DNS is the way surely

MaazSk, 2021-12-22:
@KaeLL  Thanks for the information.

@LuggerMan yes I think I should go with DNS and will be building my application based on DNS resolver.

Can you please help me with why not to use ping and the reason to use DNS?

Thanks and regards.

LuggerMan, 2021-12-22:
@MaazSk it is generally quicker to respond by design

MaazSk, 2021-12-25:
@LuggerMan, Thank you for your reply.

closed 2021-12-25 by MaazSk