espressif/esp-idf #8129: ext_flash_fatfs example not working

tbornon, 2021-12-21:
Hi,

I tried to provide all required informations. If necessary, I can add the sdkconfig (even tough I didn't modify it) and set the log level to debug / verbose

## Environment

- Development Kit:      **ESP32-Wrover-Kit**
- Kit version (for WroverKit/PicoKit/DevKitC): **v4**
- Module or chip used:  **ESP32-WROOM-32D**
- IDF version (run ``git describe --tags`` to find it): **v4.3.1**
- Build System:         **idf.py**
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): **8.4.0**
- Operating System:     **Windows**
- (Windows only) environment type: **PowerShell**
- Using an IDE?: **VS Code** but all commands are run directly from powershell (build, flash, monitor, menuconfig)
- Power Supply:        **USB**

## Problem Description

I try to add external flash to my project. In order to do so, I cloned the example provided : ext_flash_fatfs. However, I can't get it to work. The flash module I'm using is the same as in the exemple (Winbond 25Q32JVSIQ)

### Expected Behavior

Same as example 

### Actual Behavior

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6896
load:0x40078000,len:14308
ho 0 tail 12 room 4
load:0x40080400,len:3716
0x40080400: _init at ??:?

entry 0x40080680
I (29) boot: ESP-IDF v4.3.1 2nd stage bootloader
I (29) boot: compile time 15:39:08
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=08ad4h ( 35540) map
I (124) esp_image: segment 1: paddr=00018afc vaddr=3ffb0000 size=02c2ch ( 11308) load
I (129) esp_image: segment 2: paddr=0001b730 vaddr=40080000 size=048e8h ( 18664) load
I (138) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1acd4h (109780) map
I (178) esp_image: segment 4: paddr=0003acfc vaddr=400848e8 size=071b0h ( 29104) load
I (190) esp_image: segment 5: paddr=00041eb4 vaddr=50000000 size=00010h (    16) load
I (197) boot: Loaded app from partition at offset 0x10000
I (197) boot: Disabling RNG early entropy source...
I (211) cpu_start: Pro cpu up.
I (211) cpu_start: Starting app cpu, entry point is 0x400810a0
0x400810a0: call_start_cpu1 at C:/esp-idf/components/esp_system/port/cpu_start.c:141

I (197) cpu_start: App cpu up.
I (225) cpu_start: Pro cpu start user code
I (225) cpu_start: cpu freq: 160000000
I (225) cpu_start: Application information:
I (230) cpu_start: Project name:     ext_flash_fatfs
I (235) cpu_start: App version:      1
I (240) cpu_start: Compile time:     Dec 21 2021 15:38:50
I (246) cpu_start: ELF file SHA256:  2f2865b98ec5e977...
I (252) cpu_start: ESP-IDF:          v4.3.1
I (257) heap_init: Initializing. RAM available for dynamic allocation:
I (264) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (270) heap_init: At 3FFB3560 len 0002CAA0 (178 KiB): DRAM
I (276) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (283) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (289) heap_init: At 4008BA98 len 00014568 (81 KiB): IRAM
I (296) spi_flash: detected chip: generic
I (300) spi_flash: flash io: dio
W (304) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (318) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (328) example: Initializing external SPI Flash
I (338) example: Pin assignments:
I (338) example: MOSI: 23   MISO: 19   SCLK: 18   CS:  5
I (348) spi_flash: detected chip: winbond
I (348) spi_flash: flash io: dio
I (348) example: Initialized external Flash, size=4096 KB, ID=0xef4016
I (358) example: Adding external Flash as a partition, label="storage", size=4096 KB
I (368) example: Listing data partitions:
I (368) example: - partition 'nvs', subtype 2, offset 0x9000, size 24 kB
I (378) example: - partition 'phy_init', subtype 1, offset 0xf000, size 4 kB
I (388) example: - partition 'storage', subtype 129, offset 0x0, size 4096 kB
I (398) example: Mounting FAT filesystem
W (1078) vfs_fat_spiflash: f_mount failed (13)
I (1078) vfs_fat_spiflash: Formatting FATFS partition, allocation unit size=4096
I (1518) vfs_fat_spiflash: Mounting again
assertion "res == FR_OK" failed: file "../main/ext_flash_fatfs_example_main.c", line 181, function: example_get_fatfs_usage

abort() was called at PC 0x400dcc6b on core 0
0x400dcc6b: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

Backtrace:0x400856ab:0x3ffb6060 0x40085dfd:0x3ffb6080 0x4008a79e:0x3ffb60a0 0x400dcc6b:0x3ffb6110 0x400d4fe1:0x3ffb6140 0x400d5032:0x3ffb6170 0x400eab20:0x3ffb6230 0x40088529:0x3ffb6250
0x400856ab: panic_abort at C:/esp-idf/components/esp_system/panic.c:368

0x40085dfd: esp_system_abort at C:/esp-idf/components/esp_system/system_api.c:112

0x4008a79e: abort at C:/esp-idf/components/newlib/abort.c:46

0x400dcc6b: __assert_func at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/assert.c:62 (discriminator 8)

0x400d4fe1: example_get_fatfs_usage at c:\code\esp32_testfatfs\ext_flash_fatfs\build/../main/ext_flash_fatfs_example_main.c:181 (discriminator 1)

0x400d5032: app_main at c:\code\esp32_testfatfs\ext_flash_fatfs\build/../main/ext_flash_fatfs_example_main.c:59

0x400eab20: main_task at C:/esp-idf/components/freertos/port/port_common.c:133 (discriminator 2)

0x40088529: vPortTaskWrapper at C:/esp-idf/components/freertos/port/xtensa/port.c:168
```

### Steps to reproduce

1. Clone ext_flash_fatfs example project
2. idf.py flash monitor

### Code to reproduce this issue

https://github.com/espressif/esp-idf/tree/v4.3.1/examples/storage/ext_flash_fatfs

## Other items if possible

igrr, 2021-12-21:
Thank you for the report @tbornon! Could you please see what is the value of `res` in `example_get_fatfs_usage` before the assertion fails?

tbornon, 2021-12-21:
Hi @igrr, thank you for responding that fast.
Before failing, res value is 13