espressif/esp-idf #8130: BLE advertising scanning not reporting after a period of time

cmsedore, 2021-12-21:
I have an application that listens for BLE advertisements running on an ESP32C3.  After a period of time, the scan stops reporting new results.  I have verified that I'm not running out of memory and that other tasks continue to run.

I noticed that v4.3.2 has this entry:
```
Fixed no ADV report after scan for a while
```
I updated to v4.3.2 and I see the same issues as I did before the upgrade.  It looks like that fix may not have carried through.

This code (hacked up from ble50_security_client example) will demonstrate the problem--after a period of time, callbacks to esp_gap_cb stop:

```
    // Initialize NVS.
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK( ret );

    ESP_ERROR_CHECK(esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT));

    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    ret = esp_bt_controller_init(&bt_cfg);
    if (ret) {
        ESP_LOGE(TAG, "%s initialize controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bt_controller_enable(ESP_BT_MODE_BLE);
    if (ret) {
        ESP_LOGE(TAG, "%s enable controller failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bluedroid_init();
    if (ret) {
        ESP_LOGE(TAG, "%s init bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    ret = esp_bluedroid_enable();
    if (ret) {
        ESP_LOGE(TAG, "%s enable bluetooth failed: %s\n", __func__, esp_err_to_name(ret));
        return;
    }

    //register the  callback function to the gap module
    ret = esp_ble_gap_register_callback(esp_gap_cb);
    if (ret){
        ESP_LOGE(TAG, "%s gap register error, error code = %x\n", __func__, ret);
        return;
    }

    esp_ble_gap_set_ext_scan_params(&ext_scan_params);
    ESP_ERROR_CHECK(esp_ble_gap_start_ext_scan(UINT32_MAX,0));
```

Thanks.

cmsedore, 2021-12-23:
Note: using 0 (zero) instead of UINT32_MAX for duration appears to solve for this problem.  It is still likely a bug as the failure reliably occurs in less than 10 minutes in my tests.

xiewenxiang, 2021-12-31:
@cmsedore 

If the Duration parameter is non-zero and the period parameter is zero, it shall continue scanning until the duration has expired.
The time range of duration is 10ms~655.35s