espressif/esp-idf #8131: Starting wifi will cause esp32-c3 USB_SERIAL_JTAG to fail

FanhuaCloud, 2021-12-22:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-C3
- IDF version (run ``git describe --tags`` to find it):
    // ESP-IDF v5.0-dev-797-g279c8aeb8a
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // riscv32-esp-elf-gcc-8.4.0.exe (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: ESP Command Prompt.
- Using an IDE?: Yes  vscode esp-idf
- Power Supply:      USB

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior
The system prompts that the usb device is abnormal
### Steps to reproduce

1. start wifi
2. use usb serial jatg

When I set ESP System Settings->Channel for console output->usb serial/jtag controller,This problem can be solved.
But I think the problem of invalidation by default should be solved

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

mythbuster5, 2021-12-22:
Good advice!

dnstzzx, 2021-12-22:
The same problem, thanks for your solution

closed 2021-12-30 by espressif-bot