espressif/esp-idf #8135: Watchdog triggered when formatting external flash with SPIFFS

tbornon, 2021-12-22:
## Environment

- Development Kit:      ESP32-DevKitC
- Kit version (for WroverKit/PicoKit/DevKitC): v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it): v4.3.2
- Build System:         idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): 8.4.0
- Operating System:     Windows
- (Windows only) environment type: PowerShell
- Using an IDE?: VSCode but I use powershell with idf.py building, flashing, monitoring, ...
- Power Supply:        USB

## Problem Description

When the formatting of my external flash module (W25Q32) starts, I got a lot of errors / warning regarding the triggering of the watchdog. However, the process ends successfuly so maybe this isn't an issue and is the normal behavior

### Expected Behavior

No warnings / no watchdog triggered ?

### Actual Behavior

Watchdog gets triggered every 5s during the formatting process.

### Steps to reproduce

1. Clone my code (https://github.com/tbornon/spiffs_example)
2. idf.py flash monitor

### Code to reproduce this issue
 
https://github.com/tbornon/spiffs_example/blob/master/main/spiffs_example_main.c

## Debug Logs

```
I (27) boot: ESP-IDF v4.3.2-dirty 2nd stage bootloader
I (27) boot: compile time 10:33:02
I (27) boot: chip revision: 1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (37) boot.esp32: SPI Speed      : 40MHz
I (42) boot.esp32: SPI Mode       : DIO
I (46) boot.esp32: SPI Flash Size : 2MB
I (51) boot: Enabling RNG early entropy source...
I (56) boot: Partition Table:
I (60) boot: ## Label            Usage          Type ST Offset   Length
I (67) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (75) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (82) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot:  3 storage          Unknown data     01 82 00110000 000f0000
I (97) boot: End of partition table
I (101) boot_comm: chip revision: 1, min. application chip revision: 0
I (108) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=08770h ( 34672) map
I (130) esp_image: segment 1: paddr=00018798 vaddr=3ffb0000 size=02980h ( 10624) load
I (134) esp_image: segment 2: paddr=0001b120 vaddr=40080000 size=04ef8h ( 20216) load
I (144) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=1b264h (111204) map
I (185) esp_image: segment 4: paddr=0003b28c vaddr=40084ef8 size=071a8h ( 29096) load
I (197) esp_image: segment 5: paddr=0004243c vaddr=50000000 size=00010h (    16) load
I (203) boot: Loaded app from partition at offset 0x10000
I (204) boot: Disabling RNG early entropy source...
I (217) cpu_start: Pro cpu up.
I (217) cpu_start: Starting app cpu, entry point is 0x400810d8
0x400810d8: call_start_cpu1 at C:/esp-idf/components/esp_system/port/cpu_start.c:150

I (0) cpu_start: App cpu up.
I (232) cpu_start: Pro cpu start user code
I (232) cpu_start: cpu freq: 160000000
I (232) cpu_start: Application information:
I (236) cpu_start: Project name:     spiffs
[... 141 lines cut ...]

0x400dd124: esp_task_wdt_reset at C:/esp-idf/components/esp_common/src/task_wdt.c:336

0x400dd12f: idle_hook_cb at C:/esp-idf/components/esp_common/src/task_wdt.c:88

0x400d4725: esp_vApplicationIdleHook at C:/esp-idf/components/esp_common/src/freertos_hooks.c:51 (discriminator 1)

0x40087372: prvIdleTask at C:/esp-idf/components/freertos/tasks.c:3846 (discriminator 1)

0x400889b9: vPortTaskWrapper at C:/esp-idf/components/freertos/port/xtensa/port.c:168

I (60974) example: Partition size: total: 3844316, used: 0
I (60974) example: Opening file
I (61124) example: File written
I (61204) example: Renaming file
I (61274) example: Reading file
I (61274) example: Read from file: 'Hello World!'
I (61274) example: SPIFFS unmounted
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

0xjakob, 2021-12-28:
@tbornon The watchdog which triggers here is the task watchdog. This is indeed the "default" behavior of the [task watchdog](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/wdts.html?highlight=watchdog#task-watchdog-timer). This watchdog watches the idle task. If your task occupies the CPU for a long time, the idle task may never be scheduled during that time and hence not be able to reset the watchdog on time.

You have several possibilities to resolve the issue:
* Increase the default [timeout of the task watchdog via Kconfig](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/kconfig.html#config-esp-task-wdt-timeout-s)
* Deactivate the watchdog as described in the [watchdog's programming guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/wdts.html?highlight=watchdog#task-watchdog-timer) during the cpu-intensive task
* feed the watchdog periodically during inside the code running on the CPU (in this case SPIFFS code, so probably less feasible).

Keep in mind that the watchdog is there for a reason. When deactivated, it can't protect your task from stalling indefinitely.

There might also be some details involved about the dual-core nature which I currently don't recall.

0xjakob, 2022-01-24:
@tbornon Did you try the suggestions? Please let us know how it went.

closed 2022-01-25 by tbornon