espressif/esp-idf #8136: NimBLE init and deinit memory leak

kadosch, 2021-12-22:
## Environment

- Development Kit:     Olimex ESP32-DevKit-LiPo
- Kit version (for WroverKit/PicoKit/DevKitC): rev.C
- Module or chip used:  ESP32-WROOM-32E
- IDF version: v4.3.1-480-g233dc30fb1
- Build System:         idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2) 8.4.0
- Operating System:     Linux
- Using an IDE?: No
- Power Supply:         USB

## Problem Description

I'm working on a project that uses NimBLE and enters light sleep mode periodically, so it has to disable the bluetooth controller before going to sleep and reenable it after waking up.

There seems to be a memory leak in the init/deinit code from the ESP-IDF NimBLE port, as my device eventually runs out of heap memory.

The problem can be reproduced with the provided minimal example.

### Expected Behavior
NimBLE bluetooth controller is initialized and deinitlialized without memory leaks.

### Actual Behavior
NimBLE bluetooth controller is initialized and deinitlialized with memory leaks.

### Steps to reproduce
Simply execute the provided code.

### Code to reproduce this issue

```cpp
#include <esp_log.h>
#include <nvs_flash.h>
#include <esp_bt.h>
#include <esp_nimble_hci.h>
#include <nimble/nimble_port.h>
#include <nimble/nimble_port_freertos.h>
#include <esp_heap_trace.h>

#define NUM_RECORDS 100
static heap_trace_record_t trace_record[NUM_RECORDS];
static const char *TAG = "nimble_leak_test";

static void ble_manager_host_task(void *param)
{
    (void)param;

    ESP_LOGI(TAG, "BLE Host Task Started");
    /* This function will return only when nimble_port_stop() is executed */
    nimble_port_run();

    nimble_port_freertos_deinit();
}

void start_ble_func()
{
    esp_err_t err = esp_nimble_hci_and_controller_init();
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Error at nimble controller init: %s", esp_err_to_name(err));
        return;
    }

    nimble_port_init();

    nimble_port_freertos_init(ble_manager_host_task);
}

void stop_ble_func()
[... 170 lines cut ...]

0x400de164: npl_freertos_callout_init at /home/javi/esp/esp-idf/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:361

0x400dba5e: ble_hs_flow_startup at /home/javi/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_flow.c:275

0x400d9c8e: ble_hs_startup_go at /home/javi/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_startup.c:367

0x400d70c7: ble_hs_sync at /home/javi/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:345

0x400d7286: ble_hs_start at /home/javi/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:668

0x400d728e: ble_hs_event_start_stage2 at /home/javi/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:569

0x400d5fb1: nimble_port_run at /home/javi/esp/esp-idf/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:79

0x400d5d8d: ble_manager_host_task at /tmp/nimble_leak_test/main/nimble_leak_test.c:21

0x40093b50: vPortTaskWrapper at /home/javi/esp/esp-idf/components/freertos/port/xtensa/port.c:170

4504 bytes 'leaked' in trace (4 allocations)
total allocations 38 total frees 41
```

## Other items if possible
[nimble_leak_test.zip](https://github.com/espressif/esp-idf/files/7762880/nimble_leak_test.zip)

rahult-github, 2021-12-23:
Hi @kadosch ,
  Very recently, during fixing one of the issue https://github.com/espressif/esp-idf/issues/7653 , we have fixed memory leak during init / deinit.  Can you please confirm the commit https://github.com/espressif/esp-idf/commit/0f96ebce139ff9fc63069819e0e568adac8a6e3f exists at your end ? 

Thanks,
 Rahul

kadosch, 2021-12-23:
Hi @rahult-github,
yeah, I saw that issue before opening this one, but they are different. I'm not using power management at all, and the commit that fixes that issue is already included in the branch that I'm working with:

```console
[javi@LEGION esp-idf]$ git branch --show-current
release/v4.3
[javi@LEGION esp-idf]$ git describe --tags
v4.3.1-480-g233dc30fb1
[javi@LEGION esp-idf]$ git branch --contains 0f96ebc
* release/v4.3
```

Regards.

kadosch, 2021-12-24:
Hi again,
recent commit 7a4ed65769bd4c4dfafdb4c85d655bab70a13c18 fixes the problem. Funnily enough, that commit was merged to branch release/v4.3 barely 20 minutes after creating this issue.

Anyway, closing the issue as resolved.

closed 2021-12-24 by kadosch