espressif/esp-idf #8140: else misplaced compile error when BT_MULTI_CONNECTION_ENBALE != FALSE in smp_act.c

norventa, 2021-12-24:
## Problem Description
When BT_MULTI_CONNECTION_ENBALE != FALSE and SMP_SLAVE_CON_PARAMS_UPD_ENABLE == TRUE the following code produces an else misplaced compile error.
A possible solution is shown below.

Also could correct spelling error for BT_MULTI_CONNECTION_ENBALE ?

## Environment
- framework-espidf 3.40301.0 (4.3.1)

### Steps to reproduce
Set BT_MULTI_CONNECTION_ENBALE to TRUE and SMP_SLAVE_CON_PARAMS_UPD_ENABLE to TRUE in menu config and compile any BLE example.

### Code to reproduce this issue
```cpp
void smp_fast_conn_param(tSMP_CB *p_cb, tSMP_INT_DATA *p_data)
{
#if (BT_MULTI_CONNECTION_ENBALE == FALSE)
    if(p_cb->role == BTM_ROLE_MASTER) {
        L2CA_EnableUpdateBleConnParams(p_cb->pairing_bda, FALSE);
    }
#endif
#if (SMP_SLAVE_CON_PARAMS_UPD_ENABLE == TRUE)
    else {
        tBTM_SEC_DEV_REC    *p_rec = btm_find_dev (p_cb->pairing_bda);
        if(p_rec && p_rec->ble.skip_update_conn_param) {
            //do nothing
            return;
        }
        /* Disable L2CAP connection parameter updates while bonding since
        some peripherals are not able to revert to fast connection parameters
        during the start of service discovery. Connection paramter updates
        get enabled again once service discovery completes. */
        #if (BT_MULTI_CONNECTION_ENBALE == FALSE)
        L2CA_EnableUpdateBleConnParams(p_cb->pairing_bda, FALSE);
        #endif
    }
#endif
}
`

Possible fix shown below:
`
void smp_fast_conn_param(tSMP_CB *p_cb, tSMP_INT_DATA *p_data)
{
#if (BT_MULTI_CONNECTION_ENBALE == FALSE)
    if(p_cb->role == BTM_ROLE_MASTER) {
        L2CA_EnableUpdateBleConnParams(p_cb->pairing_bda, FALSE);
    }
#if (SMP_SLAVE_CON_PARAMS_UPD_ENABLE == TRUE)
    else {
        tBTM_SEC_DEV_REC    *p_rec = btm_find_dev (p_cb->pairing_bda);
        if(p_rec && p_rec->ble.skip_update_conn_param) {
            //do nothing
            return;
        }
        /* Disable L2CAP connection parameter updates while bonding since
        some peripherals are not able to revert to fast connection parameters
        during the start of service discovery. Connection paramter updates
        get enabled again once service discovery completes. */
        L2CA_EnableUpdateBleConnParams(p_cb->pairing_bda, FALSE);
    }
#endif
#endif
}
`

egi1965, 2021-12-27:
Encountered the same problem today. I recommend a file search throughout all source code for "Enbale" because such error appears in multiple files. If one uses idf.py menuconfig that seems to write "enable" but the sources contain several times "enbale" what results in compile errors.

norventa, 2021-12-28:
When I change BT_MULTI_CONNECTION_ENBALE using menuconfig everything compiles and sdkconfig.h contains "#define CONFIG_BT_MULTI_CONNECTION_ENBALE 1". For me, it is just a spelling mistake that has propogated throughout the various bluetooth libraries.

My issue is with the "else" in the first code block above. By refactoring that code, as shown in the second code block, everything works fine.

zikalino, 2021-12-29:
@norventa would you like to create a pull request or should we do it?

norventa, 2021-12-29:
I really don't know git at all so could you do it please.

xiewenxiang, 2021-12-30:
@norventa 

Thank you very much for your report issue and we will fix it as soon as possible.

Alvin1Zhang, 2022-01-05:
Thanks for reporting, fix on master is available https://github.com/espressif/esp-idf/commit/6cb6087dd0170cd534d8ad68ee2f8740a33368f9, we are back porting the fix to release/4.4 and release/4.3, will update once they are available, thanks.

AxelLin, 2022-02-17:
v4.4: 2fe53759c968e2467d39c6ae24f370719c1b9ce0
v4.3: abacbb1d12a91bf6b9afdb9a358701d82dce78f6
This can be closed.

Alvin1Zhang, 2022-02-17:
Thanks for reporting and sharing the updates, feel free to reopen if the issue still happens, thanks.

closed 2022-02-17 by Alvin1Zhang