espressif/esp-idf #8141: [Crash][release/v4.4][ESP32-S2] Start wifi crashes if NVS is not initialized

chipweinberger, 2021-12-24:
**commit** a79dc75f0a91f4b558dcefb7ae5096c71c63e2e9 (HEAD -> release/v4.4, origin/release/v4.4)
**Date:**   Thu Dec 16 10:17:56 2021 +0000

**Board:** ESP32-S2

**Expected behavior:** does full calibration without crash

```

I (5463) wifi_init: rx ba win: 6
I (5463) wifi_init: tcpip mbox: 32
I (5463) wifi_init: udp mbox: 6
I (5463) wifi_init: tcp mbox: 6
I (5473) wifi_init: tcp tx win: 5744
I (5473) wifi_init: tcp rx win: 5744
I (5483) wifi_init: tcp mss: 1440
I (5483) wifi_init: WiFi IRAM OP enabled
I (5493) wifi_init: WiFi RX IRAM OP enabled
I (5493) pd wifi: switching to 'Create Wifi AP Hotspot' mode
I (7873) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
E (7883) phy_init: esp_phy_load_cal_data_from_nvs: NVS has not been initialized. Call nvs_flash_init before starting WiFi/BT.
W (7883) phy_init: failed to load RF calibration data (0x1101), falling back to full calibration
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400efa3f  PS      : 0x00060730  A0      : 0x800f0e6e  A1      : 0x3ffe40a0  
0x400efa3f: ieee80211_hostap_attach at ??:?

A2      : 0x00000000  A3      : 0x3ffd8734  A4      : 0x3ffda388  A5      : 0x00000000  
A6      : 0x3ffd871c  A7      : 0x00000000  A8      : 0x3ffe6cff  A9      : 0x3ffe4030  
A10     : 0x3f01e8d1  A11     : 0x40039d10  A12     : 0x00000002  A13     : 0x3f0250a9  
0x40039d10: esp_log_write at /esp/esp-idf/components/log/log.c:207

A14     : 0x40039d10  A15     : 0x00000057  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
0x40039d10: esp_log_write at /esp/esp-idf/components/log/log.c:207

EXCVADDR: 0x0000002c  LBEG    : 0x00000002  LEND    : 0x3f0250a9  LCOUNT  : 0x400279a9  
0x400279a9: _xt_user_exc at /esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:633

Backtrace:0x400efa3c:0x3ffe40a00x400f0e6b:0x3ffe4100 0x400f4947:0x3ffe4140 0x400f55b9:0x3ffe4160 0x400f451a:0x3ffe4180 0x4003cd19:0x3ffe41a0 0x4003547e:0x3ffe41d0 
0x400efa3c: ieee80211_hostap_attach at ??:?

0x400f0e6b: wifi_softap_start at ??:?

0x400f4947: _do_wifi_start at ??:?

0x400f55b9: wifi_start_process at ??:?

0x400f451a: ieee80211_ioctl_process at ??:?

0x4003cd19: ppTask at ??:?

0x4003547e: vPortTaskWrapper at /esp/esp-idf/components/freertos/port/xtensa/port.c:130
```

chipweinberger, 2021-12-24:
I tried unchecking "Store phy calibration data in NVS" (CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE) and it stil crashes at the same place.

chipweinberger, 2021-12-24:
[image]

Interestingly, I fixed the crash by lowering this from 16384 to 512.

I have abort on memory allocation enabled. But it does not trigger.

jack0c, 2022-01-06:
@chipweinberger If you want to disable NVS, please also make wifi_init_config_t->nvs_enable false. But I still suggest not disable NVS.

chipweinberger, 2022-01-06:
Why disable NVS?  I dont want any state to be remembered. I want every boot to be the same. 

I could instead *clear* the nvs partition on during boot. Would you recommend that?

xueyunfei998, 2022-01-17:
hi @chipweinberger 
If you think nvs has an impact on the module restart, you can erase nvs, so that WIFI will be configured with the default configuration next time, but it is not recommended to turn off nvs.

AxelLin, 2022-01-17:
> hi @chipweinberger If you think nvs has an impact on the module restart, you can erase nvs, so that WIFI will be configured with the default configuration next time, but it is not recommended to turn off nvs.

Could you explain the impact of turning off nvs for wifi?

chipweinberger, 2022-01-26:
Just wanted to report back that the workaround fixed the crash. i.e.:

```
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
cfg.nvs_enable = false;
```

xueyunfei998, 2022-01-26:
HI @chipweinberger 

You can turn off nvs of wifi through menuconfig.

 Component config → Wi-Fi-->WiFi NVS flash

Because nvs saved the last connection parameters of wifi, such as channel, bandwidth, password, ssid, pmk and so on.

There are also some verification parameters of the physical layer that will remain in it.

When wifi is connected, the last connection parameters from nvs will be used for quick connection, which takes a short time.

AxelLin, 2022-03-01:
> **commit** [a79dc75](https://github.com/espressif/esp-idf/commit/a79dc75f0a91f4b558dcefb7ae5096c71c63e2e9) (HEAD -> release/v4.4, origin/release/v4.4) **Date:** Thu Dec 16 10:17:56 2021 +0000
> 
> **Board:** ESP32-S2
> 
> **Expected behavior:** does full calibration without crash
> 
> ```
> 
> I (5493) pd wifi: switching to 'Create Wifi AP Hotspot' mode
> I (7873) phy_init: phy_version 1800,e7ef680,Apr 13 2021,11:45:08
> E (7883) phy_init: esp_phy_load_cal_data_from_nvs: NVS has not been initialized. Call nvs_flash_init before starting WiFi/BT.
> ```

@chipweinberger 
Maybe the problem is you didn't call nvs_flash_init before starting WiFi/BT? (see above error message)

closed 2022-03-16 by Alvin1Zhang