espressif/esp-idf #8143: Memory leak in ICMP

MaazSk, 2021-12-25:
## Environment

- Development Kit:      [ESP32-DevKit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-WROOM-32|]
- IDF version (run ``git describe --tags`` to find it): v4.3
   
- Compiler version : xtensa-esp32-elf-gcc (crosstool-NG esp-2021r1) 8.4.0
  
- Operating System:     [Linux]

- Using an IDE?: [VS Code]
- Power Supply:         [USB]

## Problem Description

I was trying the ICMP echo example and there is a possible memory leak in the code.

The code which I am trying is as below:

/* ICMP echo example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

```
#include <stdio.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/FreeRTOSConfig.h"
#include "sdkconfig.h"
#include "lwip/inet.h"
#include "lwip/netdb.h"
#include "lwip/sockets.h"
#include "esp_console.h"
#include "esp_event.h"
#include "nvs_flash.h"
#include "argtable3/argtable3.h"
#include "protocol_examples_common.h"
#include "ping/ping_sock.h"
#include "esp_heap_trace.h"

#define NUM_RECORDS 200
static heap_trace_record_t trace_record[200];

#define portTICK_PERIOD_MS 1

static void ping_app(void *params);

static void cmd_ping_on_ping_success(esp_ping_handle_t hdl, void *args)
{
    uint8_t ttl;
    uint16_t seqno;
    uint32_t elapsed_time, recv_len;
    ip_addr_t target_addr;
    esp_ping_get_profile(hdl, ESP_PING_PROF_SEQNO, &seqno, sizeof(seqno));
    esp_ping_get_profile(hdl, ESP_PING_PROF_TTL, &ttl, sizeof(ttl));
    esp_ping_get_profile(hdl, ESP_PING_PROF_IPADDR, &target_addr, sizeof(target_addr));
    esp_ping_get_profile(hdl, ESP_PING_PROF_SIZE, &recv_len, sizeof(recv_len));
    esp_ping_get_profile(hdl, ESP_PING_PROF_TIMEGAP, &elapsed_time, sizeof(elapsed_time));
    printf("%d bytes from %s icmp_seq=%d ttl=%d time=%d ms\n",
           recv_len, ipaddr_ntoa((ip_addr_t *)&target_addr), seqno, ttl, elapsed_time);
}

[... 198 lines cut ...]
> 4 bytes (@ 0x3ffc8bc4) allocated CPU 1 ccount 0x31669190 caller 0x400d302f:0x400f376d
> 0x400d302f: pthread_setspecific at /home/maaz/esp/esp-idf/components/pthread/pthread_local_storage.c:222
> 
> 0x400f376d: sys_thread_sem_init at /home/maaz/esp/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:539
> 
> 12 bytes (@ 0x3ffc8bd4) allocated CPU 1 ccount 0x316699e4 caller 0x400d307f:0x400f376d
> 0x400d307f: pthread_setspecific at /home/maaz/esp/esp-idf/components/pthread/pthread_local_storage.c:248
> 
> 0x400f376d: sys_thread_sem_init at /home/maaz/esp/esp-idf/components/lwip/port/esp32/freertos/sys_arch.c:539
> 
> 84 bytes (@ 0x3ffc8abc) allocated CPU 0 ccount 0x3506ac10 caller 0x40089312:0x400895af
> 0x40089312: xQueueGenericCreate at /home/maaz/esp/esp-idf/components/freertos/queue.c:413
> 
> 0x400895af: xQueueCreateMutex at /home/maaz/esp/esp-idf/components/freertos/queue.c:525
> 
> 84 bytes (@ 0x3ffc8b14) allocated CPU 0 ccount 0x3506bf28 caller 0x40089312:0x400895af
> 0x40089312: xQueueGenericCreate at /home/maaz/esp/esp-idf/components/freertos/queue.c:413
> 
> 0x400895af: xQueueCreateMutex at /home/maaz/esp/esp-idf/components/freertos/queue.c:525
> 
> 3408 bytes 'leaked' in trace (20 allocations)
> total allocations 60 total frees 41
> (NB: New entries were traced while dumping, so trace dump may have duplicate entries.)

I can see from these logs that 3408 bytes are been leaked and my code calls it every minute.

ESP-YJM, 2021-12-29:
@MaazSk I think your approach is not very correct, because `esp_ping` runs as a task. Although you have executed `esp_ping_stop(ping)` and `esp_ping_delete_session(ping)`, the task of `esp_ping` will not exit immediately, and the resources will not be recovered immediately, so there will be the memory leaks you see.

MaazSk, 2021-12-29:
@ESP-YJM Thank you for your reply, I discovered the same yesterday only and I have already made the changes you have suggested. But still, I don't know why when my device is running for a long time I am getting this error.

```
***ERROR*** A stack overflow in task Ping has been detected.

Backtrace:0x40088f4b:0x3ffcae80 0x40089749:0x3ffcaea0 0x4008d0e1:0x3ffcaec0 0x4008b622:0x3ffcaf40 0x40089840:0x3ffcaf60 0x400897f2:0x00000000 |<-CORRUPTED

```

Can you please help me with this? The stack overflow happened after 4-5 hours from the time the device first booted

ESP-YJM, 2021-12-29:
@MaazSk You can simply en-large the ping task size by `config.task_stack_size = 4096`.

MaazSk, 2021-12-29:
Ok, I will try and let you know the output.

Thank you! @ESP-YJM

MaazSk, 2021-12-29:
@ESP-YJM I checked just now changing the stack size to 4096 but still I am getting the stack overflow error as below

> ***ERROR*** A stack overflow in task ping has been detected.
> 
> Backtrace:0x40088f4b:0x3ffcee30 0x40089749:0x3ffcee50 0x4008d0e1:0x3ffcee70 0x4008b622:0x3ffceef0 0x40089840:0x3ffcef10 0x400897f2:0x58575655 |<-CORRUPTED
> 0x40088f4b: panic_abort at /home/maaz/esp/esp-idf/components/esp_system/panic.c:368
> 
> 0x40089749: esp_system_abort at /home/maaz/esp/esp-idf/components/esp_system/system_api.c:112
> 
> 0x4008d0e1: vApplicationStackOverflowHook at /home/maaz/esp/esp-idf/components/freertos/port/xtensa/port.c:490
> 
> 0x4008b622: vTaskSwitchContext at /home/maaz/esp/esp-idf/components/freertos/tasks.c:3289
> 
> 0x40089840: _frxt_dispatch at /home/maaz/esp/esp-idf/components/freertos/port/xtensa/portasm.S:432
> 
> 0x400897f2: _frxt_int_exit at /home/maaz/esp/esp-idf/components/freertos/port/xtensa/portasm.S:231

ESP-YJM, 2021-12-30:
@MaazSk Could you please offer the code that i can reproduce locally.

MaazSk, 2021-12-30:
@ESP-YJM, I think it is a bug in my code let me confirm it first and then I will share the snippet because the standalone code of ping is working fine. Just give me a day I will keep you updated.

Thank you!

MaazSk, 2022-01-03:
There was a bug in my code sorry.

closed 2022-01-03 by MaazSk