espressif/esp-idf #8144: SPI0_HOST and SPI3_HOST for ESP32-C3

arktrin, 2021-12-26:
As of [ESP32-C3 datasheet v1.1](https://www.espressif.com/sites/default/files/documentation/esp32-c3_datasheet_en.pdf#page=20) (page 20): 
```
ESP32-C3 features three SPI interfaces (SPI0, SPI1, and SPI2). 
SPI0 and SPI1 can only be configured to operate in SPI memory mode,
while SPI2 can be configured to operate in both SPI memory and general-purpose SPI modes.
```
As of [online documentation on ESP32-C3](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/peripherals/spi_master.html):
```
ESP32-C3 integrates 3 SPI peripherals.
SPI0 and SPI1 are used internally to access the ESP32-C3’s attached flash memory. 
Both controllers share the same SPI bus signals, and there is an arbiter to determine 
which can access the bus. Currently, SPI Master driver does not support SPI1 bus.
```
So at least there is no such thing as SPI3 and `SPI3_HOST` for ESP32-C3, though online docs have this somehow:
```
enum spi_host_device_t
Enum with the three SPI peripherals that are software-accessible in it.
Values:
SPI1_HOST = 0
SPI1.
SPI2_HOST = 1
SPI2.
SPI3_HOST = 2
SPI3.
```
As of [spi_types.h from esp-idf](https://github.com/espressif/esp-idf/blob/master/components/hal/include/hal/spi_types.h) there is no such thing as SPI0 or `SPI0_HOST`.

Is this just a documentation issue?

mythbuster5, 2022-01-04:
Yes, what you pointed out is correct. Because this is the way to operate on ESP32 but not ESP32C3, so, it's a historical issue. On ESP32C3, User can only operate SPI2 (general spi) and spi1/spi0 are only using for memory.

arktrin, 2022-01-04:
@mythbuster5 Is anybody going to fix docs ?